ToniA/arduino-heatpumpir #5: typo in "FujitsuHeatpumpIR.cpp"

kalleka, 2015-09-14:
Thanks for a sharing such a nice library. Shouldn’t there be “==” on the line 79 in a FujitsuHeatpumpIR.cpp instead of “!=” ?

Some more Fujitsu Nocria codes (all in hex):
“ECO mode on” (limits compressor RPM to 70% max, could be useful for a low temperature “hold heat”):
14,63,00,10,10,FE,09,30,00,04,01,00,00,00,00,CB  (byte 14  0x00, normal mode 0x20)
Hi-Power:
14,63,00,10,10,39,C6
Filter cleaning:
14,63,00,10,10,3F,C0
Super quiet:
14,63,00,10,10,2D,D2
Test run:
14,63,00,10,10,03,FC

Kalle

ToniA, 2015-09-14:
Yes, I think you're right. I pushed fix to this.

I left out almost all 'advanced' features, as I wanted to keep the interface the same for all heatpumps. But now that you mention it, there's nothing stopping me (or anybody, pull requests are welcome) adding model-specific commands... Would you be kind enough to test out the Fujitsu enhancements I made?

kalleka, 2015-09-14:
Thank you for a reply. I would be happy to test the enhancements. I have access to the heatpump mostly on the weekends though, so threre may be some delay.
Kalle

closed 2015-11-03 by ToniA