ToniA/arduino-heatpumpir #6: Quiet and powerfull modes on Panasonic aircons

ptoump, 2015-12-11:
Could you please implement the quiet and powerfull modes of Panasonic Aircons? 
They should be on byte 13 or 14 depending on where you start counting : 
http://www.instructables.com/id/Reverse-engineering-of-an-Air-Conditioning-control/?ALLSTEPS
http://www.analysir.com/blog/2014/12/27/reverse-engineering-panasonic-ac-infrared-protocol/

ToniA, 2015-12-11:
Sure, it should be pretty simple to implement, Fujitsu already has a special argument 'ecomode':

    void send(IRSender& IR, uint8_t powerModeCmd, uint8_t operatingModeCmd, uint8_t fanSpeedCmd, uint8_t temperatureCmd, uint8_t swingVCmd, uint8_t swingHCmd);
    void send(IRSender& IR, uint8_t powerModeCmd, uint8_t operatingModeCmd, uint8_t fanSpeedCmd, uint8_t temperatureCmd, uint8_t swingVCmd, uint8_t swingHCmd, bool ecoModeCmd);

... so Panasonic could very well have two boolean arguments, 'powerful' and 'quiet'. I'll take a look at this next week.

ptoump, 2015-12-14:
Or another int argument with three acceptable values for "normal, quiet, powerfull". 

One way or another thanks a bunch!

ToniA, 2016-01-07:
Hi, would you try out the 'dev' branch and see if this works? Thanks!

At least in DKE-model heatpump the 'powerful' and 'quiet' seem to be toggles, sent as special 'short messages'.

ptoump, 2016-01-25:
Sorry I've been up to my neck and didn't check in for a while.
I'll check it as soon as possible.
Just FYI all I have available are basic split units that seem to work with JKE/NKE profile so far.

ptoump, 2016-02-12:
ok so finally I got around to test it.. It seems to work PERFECTLY !!  Kudos !!

here's my mysensors sketch on a domoticz controler for reference :

https://github.com/ptoump/MyHomeSketches/blob/master/Basement_Room_01/AC_n_TEMP_1.6/AC_n_TEMP_1.6.ino

ptoump, 2016-02-13:
Sorry , in the end it didn't work properly, I was to happy to notice the difference...
anyways, after loosing a few (5-6) hours of sleep I made the following changes (I'm too sleepy to make any pulls / commits so i'm putting them here) :
Keep in mind these work for MKE/MKR models (A75C3755 remote) and maybe others..

1. the powerful , quiet codes are reversed.. they should be: 
`#define PANASONIC_AIRCON2_QUIET      0x20 // Quiet setting
#define PANASONIC_AIRCON2_POWERFUL   0x01 // Powerful setting`

1. the template in PanasonicHeatpumpIR.cpp for the MKE is
`{ 0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0x80, 0x00, 0x06, 0x00 },`

I just altered the DKE template and removed the hswing check...

ir codes taken from here : http://www.instructables.com/id/Reverse-engineering-of-an-Air-Conditioning-control/?ALLSTEPS

If you make any updates, please reply here, in case I miss it... again...

advfr, 2018-11-16:
Hello guys !
First Toni big thanks for that great project. I was able to "burn" your ESPeasy fork on an ESP01 (ESP8266 with 1M rom), as I've never been able to compile ESPeasy with your project, I had each time error durring the compilation. I would like to come back on this QUIET and POWERFULL mode on Panasonic heatpump. I have a recent TKE which repplies correctly to JKE orders. The problem is it is quite noisy even at level1 and I always use QUIET mode, problem... I don't really understand what did @ptoump to make it working, in which file should I modify what you didi in step 1, and what should I do at step 2 ? Will try to compile the project in dev branch on ESP but I'm a bit afraid not to have the knowledge for that... Thank you guys !

ptoump, 2018-11-18:
Hello @advfr 

Coincidentally I forked this project  on Friday morning and made some changes to make it work with MKE . Haven't tried it yet, but everything should work EXCEPT the inverted quiet / powerfull modes. Since these are definitions the cannot be changed on the fly, and they either need to be made variables or insert multiple checks in the code which is bad programming. Perhaps @ToniA could take a look if I missed anything. I also tried to edit the espeasy plugin to accommodate for the changes..

Either way it's easier to just label your home automation software buttons the other way around than make more changes to the code.

My nodemcus will be shipped next week so it will take some time before I'll be able to check for errors myself.

Sorry it took so long to answer!