ToniA/arduino-heatpumpir #7: Able to support Mitsubishi P/N KM09G 0116667?

bilogic, 2015-12-29:
Hi,

I tried this code, but no reaction from the AC. 
I have confirmed that the IR LED is indeed blinking (using smartphone camera)

Below are the RAW codes for my on and off. Any tips on the best way to get started? Thank you.

```
//aircon  ON Raw: (583) 3436, -1724, 440, -1284, 408, -1316, 444, -424, 440, -424, 440, -428, 436, -1288, 436, -428, 440, -424, 412, -1316, 440, -1280, 436, -436, 440, -1284, 440, -428, 436, -424, 412, -1316, 444, -1280, 436, -428, 440, -1280, 416, -1308, 440, -428, 408, -460, 412, -1316, 436, -424, 412, -456, 440, -1288, 412, -456, 412, -456, 412, -456, 436, -432, 412, -460, 412, -456, 440, -428, 440, -428, 440, -424, 436, -432, 412, -460, 436, -424, 440, -424, 440, -436, 436, -428, 440, -432, 436, -428, 440, -424, 440, -432, 436, -428, 440, -1280, 412, -456, 440, -428, 436, -432, 412, -452, 440, -428, 440, -1284, 440, -1276, 468, -412, 436, -424, 436, -444, 408, -460, 440, -1288, 412, -452, 440, -1284, 440, -436, 412, -460, 440, -424, 412, -456, 436, -428, 468, -1264, 412, -1316, 440, -424, 468, -1264, 412, -1312, 408, -456, 440, -432, 412, -460, 436, -424, 440, -428, 444, -420, 412, -456, 440, -428, 436, -1288, 440, -424, 440, -1280, 440, -1280, 440, -432, 408, -460, 440, -428, 440, -424, 412, -456, 412, -456, 412, -456, 412, -456, 412, -456, 412, -456, 412, -460, 432, -440, 408, -460, 436, -432, 436, -428, 440, -424, 440, -428, 412, -460, 440, -424, 440, -420, 440, -440, 436, -424, 444, -428, 444, -420, 440, -424, 444, -428, 412, -452, 412, -456, 436, -440, 436, -432, 440, -428, 412, -460, 436, -428, 412, -452, 440, -428, 440, -1284, 412, -456, 412, -456, 440, -424, 440, -428, 440, -420, 408, -456, 412, -460, 436, -424, 412, -456, 440, -432, 436, -432, 440, -424, 412, -456, 440, -424, 412, -456, 412, -456, 408, -456, 440, -424, 412, -456, 440, -428, 440, -424, 412, -456, 412, -1312, 440, -1292, 416, -1308, 412, -1316, 440, -17200, 3512, -1744, 444, -1280, 412, -1312, 440, -428, 436, -428, 444, -424, 440, -1280, 440, -424, 444, -424, 408, -1316, 440, -1280, 440, -424, 436, -1288, 440, -424, 408, -464, 408, -1312, 436, -1288, 440, -432, 408, -1316, 440, -1288, 440, -432, 440, -428, 440, -1280, 436, -428, 440, -428, 436, -1284, 440, -424, 440, -424, 440, -428, 440, -428, 440, -424, 440, -424, 440, -424, 476, -404, 440, -424, 412, -464, 436, -424, 440, -424, 412, -456, 412, -456, 436, -428, 436, -432, 440, -424, 436, -428, 436, -428, 436, -432, 440, -1288, 412, -448, 436, -436, 440, -424, 408, -460, 436, -432, 436, -1292, 412, -1316, 412, -452, 408, -456, 424, -444, 436, -432, 440, -1284, 440, -420, 412, -1320, 436, -424, 412, -456, 436, -428, 412, -452, 436, -436, 440, -1284, 440, -1280, 412, -460, 440, -1280, 412, -1316, 412, -456, 436, -428, 440, -428, 440, -428, 440, -424, 440, -428, 412, -460, 436, -428, 412, -1316, 436, -432, 436, -1280, 436, -1284, 444, -424, 408, -460, 412, -484, 412, -428, 436, -428, 444, -420, 444, -420, 412, -456, 412, -456, 412, -456, 412, -456, 412, -464, 412, -456, 440, -428, 440, -424, 440, -424, 440, -432, 408, -460, 440, -424, 440, -420, 444, -436, 440, -424, 440, -428, 440, -424, 440, -428, 440, -428, 412, -456, 408, -460, 436, -436, 440, -428, 440, -428, 412, -460, 440, -424, 412, -452, 416, -452, 440, -1284, 436, -428, 436, -428, 440, -428, 412, -456, 440, -428, 412, -460, 440, -424, 440, -428, 436, -428, 440, -424, 436, -432, 408, -464, 440, -424, 412, -460, 436, -432, 468, -408, 436, -428, 412, -464, 444, -424, 412, -452, 440, -428, 440, -428, 440, -1276, 412, -1316, 440, -1284, 444, -1276, 440, 

//aircon OFF Raw: (583) 3412, -1748, 416, -1312, 436, -1284, 408, -460, 436, -428, 408, -464, 412, -1308, 412, -456, 412, -456, 436, -1284, 412, -1312, 440, -424, 408, -1316, 440, -424, 408, -460, 412, -1312, 412, -1316, 440, -428, 412, -1312, 412, -1312, 412, -456, 444, -424, 440, -1276, 416, -464, 440, -424, 440, -1284, 440, -424, 408, -460, 408, -460, 436, -428, 412, -460, 412, -452, 408, -464, 408, -456, 412, -456, 412, -448, 412, -456, 408, -456, 412, -456, 440, -424, 408, -460, 408, -456, 412, -456, 440, -428, 440, -420, 412, -456, 476, -404, 440, -424, 408, -468, 412, -456, 440, -424, 440, -428, 440, -1280, 440, -1284, 412, -456, 440, -428, 440, -428, 436, -432, 412, -1312, 412, -456, 440, -1284, 412, -456, 412, -452, 412, -456, 412, -456, 444, -420, 440, -1284, 408, -1320, 412, -452, 412, -1312, 440, -1284, 412, -452, 436, -428, 436, -436, 412, -452, 436, -436, 436, -428, 440, -424, 408, -460, 440, -1280, 412, -460, 412, -1316, 412, -1316, 408, -456, 412, -452, 412, -456, 408, -460, 408, -456, 440, -428, 412, -456, 440, -424, 412, -456, 412, -460, 408, -460, 472, -404, 436, -428, 408, -468, 412, -460, 436, -428, 440, -424, 440, -424, 412, -456, 412, -456, 408, -456, 412, -452, 412, -460, 420, -444, 412, -452, 412, -456, 432, -444, 412, -460, 440, -424, 408, -456, 444, -424, 412, -452, 436, -432, 440, -424, 412, -460, 412, -1312, 412, -456, 412, -456, 444, -420, 436, -432, 412, -460, 440, -424, 412, -456, 408, -460, 472, -408, 412, -452, 412, -456, 444, -420, 412, -452, 436, -432, 412, -456, 444, -424, 408, -456, 412, -456, 436, -428, 408, -460, 408, -460, 412, -456, 432, -1292, 412, -456, 468, -1264, 412, -1312, 412, -17096, 3536, -1724, 472, -1264, 412, -1312, 412, -456, 408, -456, 408, -460, 412, -1312, 412, -452, 412, -456, 436, -1284, 412, -1312, 408, -456, 412, -1308, 440, -432, 412, -452, 444, -1280, 412, -1316, 440, -428, 436, -1284, 416, -1312, 412, -456, 412, -456, 412, -1316, 440, -428, 440, -428, 408, -1312, 412, -456, 440, -424, 440, -428, 440, -424, 440, -424, 408, -468, 440, -424, 436, -432, 440, -424, 440, -428, 440, -428, 412, -456, 408, -456, 436, -440, 440, -428, 436, -432, 412, -460, 440, -424, 408, -460, 436, -428, 412, -452, 436, -432, 436, -428, 440, -424, 408, -456, 412, -456, 436, -1284, 412, -1312, 412, -456, 440, -424, 412, -456, 440, -424, 436, -1288, 412, -456, 412, -1312, 408, -460, 408, -456, 440, -424, 440, -428, 412, -460, 412, -1312, 440, -1284, 412, -456, 436, -1284, 412, -1312, 440, -428, 408, -460, 440, -420, 440, -428, 436, -432, 412, -452, 412, -456, 408, -456, 408, -1316, 436, -424, 412, -1316, 436, -1284, 440, -420, 440, -428, 436, -424, 412, -452, 412, -460, 412, -456, 440, -424, 440, -424, 440, -428, 440, -428, 440, -424, 440, -424, 412, -456, 440, -428, 412, -452, 412, -464, 436, -424, 412, -456, 440, -428, 436, -424, 440, -428, 440, -424, 440, -424, 412, -456, 408, -456, 440, -424, 440, -432, 412, -460, 440, -424, 412, -452, 412, -456, 464, -416, 436, -436, 412, -452, 440, -428, 408, -1316, 440, -432, 440, -428, 436, -428, 440, -428, 408, -460, 440, -424, 436, -432, 440, -424, 440, -424, 440, -428, 440, -420, 468, -408, 436, -436, 408, -456, 436, -436, 412, -452, 440, -428, 436, -432, 436, -428, 436, -428, 436, -428, 416, -452, 412, -1312, 412, -456, 436, -1288, 440, -1284, 412, 
```

ToniA, 2015-12-29:
Is this a Mitsubishi MSZ GA-series air conditioner? It might have totally different codes.

Is there any chance you could try to decode the original remote signals using https://github.com/ToniA/Raw-IR-decoder-for-Arduino ?

Also the range of the IR transmission is quite short, maybe max. 2 meters if the LED is directly connected into the Arduino.

bilogic, 2015-12-29:
It's the MSY-GE10VA series air conditioner. I'm hoping to adapt your code for it. The IR LED is about 30 cm from the aircon. I managed to run https://github.com/ToniA/Raw-IR-decoder-for-Arduino with the following output:

```
Ready to decode IR!

Number of symbols: 384
Symbols:
Hh11000010000101010001010110000100100000000000000000000000000000000000000001010000010010000000001100101000000000100111000010000000000000000000000000000000100000000000000000000011011010WHh110000000100010100010000101010100010000100000000000000000000000000000000011000000100100000000101000110000000000100110100010000000000000000000000000000000000000100000000000000
åÏÏ0
```

ToniA, 2015-12-29:
Let's take the discussion to email, look at my profile for the address. Seems like the decoder crashed...

bilogic, 2015-12-30:
Ok, have sent you an email. Thank you.

closed 2015-12-30 by bilogic