ToniA/arduino-heatpumpir #19: Gree ac

funivan, 2016-08-11:
Is there any way to control Gree AC. I tried Hyundai (Remote Control P/N Y512F2) on arduino Uno. But it doesn't work.

ToniA, 2016-08-11:
There is no Gree support at the moment, but there was some discussion on this subject in the MySensors forum recently: https://forum.mysensors.org/topic/4253/ac-ir-code-decrypting

'dpressle' chose to make his own library, so once he publishes it, it will be a standalone version for Gree only.

closed 2016-08-11 by ToniA

ToniA, 2016-08-15:
The Gree module has been published here: https://github.com/dpressle/MySensors/tree/Development/AcControl

It's a totally separate implementation, but it also integrates with MySensors.

funivan, 2016-08-15:
Thanks for help. I will try next week. 
Is it possible to add free support to the arduino-heatpumpir library?

ToniA, 2016-08-15:
Yes, I will certainly appreciate pull requests :)

adrianmihalko, 2016-09-15:
Hi ToniA,

I am waiting for my first ESP8266 module and I would like to control my AC unit with ESP EASY. I have here an Arduino Uno so I tried your app: IR Raw decoder for Arduino.

The result:

```
Number of symbols: 115
Symbols:
Hh1100010011010011011001001000000000000001001001001000000011000000101000100000000000000000000000000000000001000000
Bytes:
23,CB,26,01,80,24,01,03,45,00,00,00,00,02
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  3820
HEADER SPACE: 1560
BIT MARK:     457
ZERO SPACE:   318
ONE SPACE:    1321
Decoding known protocols...
Looks like a Fuego etc. protocol
Checksum matches
POWER ON
MODE HEAT
Temperature: 28
FAN: 3
Vertical air direction: UP
```

It looks fine, it is successfully recognized my remote. I tried in all mode 1-2-3-4 and my remote is recognized in every mode. I don't know it's good or not?

Interesting, that sometimes I see signals like:

```
Number of symbols: 1
Symbols:

Bytes:

Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  0
HEADER SPACE: 0
BIT MARK:     80
ZERO SPACE:   0
ONE SPACE:    0
Decoding known protocols...
Unknown protocol

```

...when no remote is used.

Back to my real question(s):

- arduino-heatpumpir: my AC unit is not an unit listed in the code (Pana, Samsung, Sharp, etc.) - is there any way to control my AC based on the code recognized in ir-raw decoder?

- I am happy to see that you implemented your AC IR Library to ESP8266 - this is my final goal, do you think that it will possible to control my AC with your plugin? If yes, how?

Best regards,
Adrian

funivan, 2016-09-15:
@adrianmihalko hello. I find some code how to switch on my Gree AC. 
At this time I try to implement control of the temperature, fan mode, etc...

And also I have similar behaviour. Sometimes I get null packages (I don`t know how to call them) when I just grab signals.

P.S. My current platform is Arduino UNO + IRLib