ToniA/arduino-heatpumpir #20: Additional Fujitsu codes

Chalmes, 2016-09-19:
Not an issue as such, but some additional data points using a Fujitsu AR-RAH2E remote to fill in the blanks.  Data from your Raw IR Decode project.

Test mode button:
	Number of symbols: 59
	Symbols:
	Hh00101000110001100000000000001000000010001100000000111111
	Bytes:
	14,63,00,10,10,03,FC
	Timings (in us): 
	PAUSE SPACE:  0
	HEADER MARK:  3440
	HEADER SPACE: 1560
	BIT MARK:     420
	ZERO SPACE:   314
	ONE SPACE:    1154

Economy Mode button:
	Number of symbols: 59
	Symbols:
	Hh00101000110001100000000000001000000010001001000001101111
	Bytes:
	14,63,00,10,10,09,F6
	Timings (in us): 
	PAUSE SPACE:  0
	HEADER MARK:  3420
	HEADER SPACE: 1560
	BIT MARK:     420
	ZERO SPACE:   317
	ONE SPACE:    1160

10degrees C Heat button: (Heat, 10degrees, no fan information on remote, changes operating mode byte (9) to 0B)
	Number of symbols: 131
	Symbols:
	Hh00101000110001100000000000001000000010000111111110010000000011000000010011010000001000000000000000000000000000000000010010000001
	Bytes:
	14,63,00,10,10,FE,09,30,20,0B,04,00,00,00,20,81
	Timings (in us): 
	PAUSE SPACE:  0
	HEADER MARK:  3440
	HEADER SPACE: 1560
	BIT MARK:     420
	ZERO SPACE:   300
	ONE SPACE:    1143

Auto swing off:
14,63,00,10,10,FE,09,30,20,20,04,E0,0A,00,20,82

Vertical auto swing:
14,63,00,10,10,FE,09,30,20,20,14,DF,0A,00,20,73

Horizontal auto swing:
14,63,00,10,10,FE,09,30,20,20,24,DE,0A,00,20,64

Both Auto
14,63,00,10,10,FE,09,30,20,20,34,DD,0A,00,20,55

Can also provide information regarding the timer if necessary, though given the nature of the library, that's probably not required.  At a glance though, i think the remote sends the timer on/off as offsets from the current time, which is then absolute time independant, ie no actual clock information is ever sent from remote to main unit.  The code i get when setting the On Timer to 2:30 AM was different from one 5-minute period to the next. (Initially EA in byte 11, and then E4 5 minutes later)

ToniA, 2016-09-19:
Hi,

the reason why I haven't added these is that I have wanted to keep this as generic as possible. With all of these model-specific things IMHO it just grows unnecessarily.

But let's do so that I'll add these as comments into the code, so if anybody would like to utilize these, at least the data would be there?

Toni

Chalmes, 2016-09-20:
No worries.

In that case its just the test mode codes that can be updated as TestRun, FilterClean and SuperQuiet are all using the Off message.

ToniA, 2016-09-24:
I just pushed changes to implement these, please let me know if these work... For example:

`FujitsuHeatpumpIR f;`
`f.sendFujitsuVerticalSwingOn(irSender);`

closed 2016-10-25 by ToniA