ToniA/arduino-heatpumpir #23: How to add more models?

bserem, 2016-12-11:
I just found this repo, and the idea looks amazing.

I have two Toshiba AC split type Units (RAS-10G2KVP-E RAS-10G2AVP-E and RAS-13G2KVP-E RAS-13G2AVP-E) that I would like to control.
With the simple.ino sketch provided by this library they do not seem to react, so I guess I'll probably need to add support for them.

I have successfully managed to decode other remotes (amplifiers, tvs) with https://github.com/z3t0/Arduino-IRremote library. It reads the codes sent out by the Toshiba remote two (WH-TA01EE, first one here http://www.toshiba-aircon.co.uk/assets/uploads/product_assets/EU_Wireless_Remote_controller_20150120.pdf) but when I try to send them the AC does nothing.

How can I proceed so that I improve this library to support the afforementioned models?
I have an IR receiver and transmitter, I do not have the adafruit shield.
Maybe I could help decode the remote and/or unit.

Thanks

PS: Here is what I get when I press the on/off button (first for on, second for off):
```
Encoding  : SAMSUNG
Code      : F20D05FA (32 bits)
Timing[179]: 
     +4300, -4350     + 550, -1550     + 600, -1550     + 600, -1550
     + 550, -1600     + 550, - 550     + 550, - 500     + 550, -1600
     + 600, - 450     + 600, - 500     + 550, - 500     + 600, - 450
     + 600, - 500     + 550, -1600     + 550, -1600     + 550, - 500
     + 600, -1550     + 600, - 450     + 600, - 500     + 550, - 500
     + 600, - 500     + 550, - 500     + 600, -1550     + 600, - 450
     + 600, -1550     + 600, -1550     + 600, -1550     + 600, -1550
     + 550, -1550     + 600, -1550     + 600, - 500     + 600, -1550
     + 550, - 500     + 600, - 450     + 600, - 500     + 600, - 450
     + 600, - 500     + 600, - 450     + 600, - 500     + 550, - 500
     + 600, -1550     + 600, - 450     + 600, -1550     + 600, -1550
     + 600, -1550     + 600, - 450     + 600, - 500     + 550, - 500
     + 600, - 500     + 550, - 500     + 600, - 450     + 600, - 500
     + 600, - 450     + 600, - 500     + 550, - 500     + 600, - 500
     + 550, - 500     + 600, - 450     + 600, - 500     + 600, - 450
     + 600, - 500     + 550, - 500     + 600, - 500     + 550, - 500
     + 600, - 500     + 550, - 500     + 600, - 450     + 600, -1550
     + 600, - 500     + 550, - 500     + 600, - 500     + 550, - 500
     + 600, -1550     + 600, - 450     + 600, - 500     + 550, - 500
     + 600, -1550     + 600, -1550     + 550, - 500     + 600, -1550
     + 600, - 450     + 600, - 500     + 600, -1550     + 550, - 500
     + 600, - 500     + 550, -1550     + 600, - 500     + 600, -1550
     + 550, - 500     + 550
unsigned int  rawData[179] = {4300,4350, 550,1550, 600,1550, 600,1550, 550,1600, 550,550, 550,500, 550,1600, 600,450, 600,500, 550,500, 600,450, 600,500, 550,1600, 550,1600, 550,500, 600,1550, 600,450, 600,500, 550,500, 600,500, 550,500, 600,1550, 600,450, 600,1550, 600,1550, 600,1550, 600,1550, 550,1550, 600,1550, 600,500, 600,1550, 550,500, 600,450, 600,500, 600,450, 600,500, 600,450, 600,500, 550,500, 600,1550, 600,450, 600,1550, 600,1550, 600,1550, 600,450, 600,500, 550,500, 600,500, 550,500, 600,450, 600,500, 600,450, 600,500, 550,500, 600,500, 550,500, 600,450, 600,500, 600,450, 600,500, 550,500, 600,500, 550,500, 600,500, 550,500, 600,450, 600,1550, 600,500, 550,500, 600,500, 550,500, 600,1550, 600,450, 600,500, 550,500, 600,1550, 600,1550, 550,500, 600,1550, 600,450, 600,500, 600,1550, 550,500, 600,500, 550,1550, 600,500, 600,1550, 550,500, 550};  // SAMSUNG F20D05FA
unsigned int  data = 0xF20D05FA;

Encoding  : SAMSUNG
Code      : F20D05FA (32 bits)
Timing[179]: 
     +4350, -4300     + 600, -1550     + 550, -1600     + 550, -1600
     + 550, -1500     + 700, - 450     + 600, - 500     + 550, -1500
     + 650, - 500     + 600, - 450     + 600, - 500     + 550, - 500
     + 600, - 500     + 550, -1550     + 600, -1550     + 600, - 500
     + 550, -1600     + 550, - 500     + 600, - 450     + 600, - 500
     + 600, - 450     + 600, - 500     + 550, -1600     + 550, - 500
     + 600, -1550     + 600, -1550     + 550, -1550     + 600, -1550
     + 600, -1550     + 600, -1550     + 600, - 450     + 600, -1550
     + 600, - 500     + 550, - 500     + 600, - 500     + 550, - 500
     + 600, - 450     + 600, - 500     + 600, - 450     + 600, - 500
     + 550, -1600     + 550, - 500     + 600, -1550     + 600, -1550
     + 550, -1600     + 550, - 500     + 600, - 450     + 600, - 500
     + 600, - 450     + 600, - 500     + 550, - 500     + 600, - 500
     + 550, - 500     + 600, - 450     + 600, -1550     + 600, -1550
     + 600, -1550     + 600, - 450     + 600, - 500     + 600, - 450
     + 600, - 500     + 550, - 500     + 600, - 500     + 550, - 500
     + 600, - 450     + 600, - 500     + 600, - 450     + 600, -1550
     + 600, - 450     + 600, - 500     + 600, - 450     + 600, - 500
     + 600, -1550     + 550, - 500     + 600, - 450     + 600, - 500
     + 600, -1550     + 550, -1600     + 550, - 500     + 600, -1550
     + 600, - 450     + 600, - 500     + 550, -1550     + 600, - 500
     + 600, - 450     + 600, -1550     + 600, -1550     + 600, - 500
     + 550, -1550     + 600
unsigned int  rawData[179] = {4350,4300, 600,1550, 550,1600, 550,1600, 550,1500, 700,450, 600,500, 550,1500, 650,500, 600,450, 600,500, 550,500, 600,500, 550,1550, 600,1550, 600,500, 550,1600, 550,500, 600,450, 600,500, 600,450, 600,500, 550,1600, 550,500, 600,1550, 600,1550, 550,1550, 600,1550, 600,1550, 600,1550, 600,450, 600,1550, 600,500, 550,500, 600,500, 550,500, 600,450, 600,500, 600,450, 600,500, 550,1600, 550,500, 600,1550, 600,1550, 550,1600, 550,500, 600,450, 600,500, 600,450, 600,500, 550,500, 600,500, 550,500, 600,450, 600,1550, 600,1550, 600,1550, 600,450, 600,500, 600,450, 600,500, 550,500, 600,500, 550,500, 600,450, 600,500, 600,450, 600,1550, 600,450, 600,500, 600,450, 600,500, 600,1550, 550,500, 600,450, 600,500, 600,1550, 550,1600, 550,500, 600,1550, 600,450, 600,500, 550,1550, 600,500, 600,450, 600,1550, 600,1550, 600,500, 550,1550, 600};  // SAMSUNG F20D05FA
unsigned int  data = 0xF20D05FA;
```

MRuniversee, 2016-12-11:
If you read how he made the libraries you will notice he uses the raw decoder to analyze the signal,do that and you can make your own library.

bserem, 2016-12-11:
Thanks for the reply!
I do not really get what you mean by "how he made the libraries", I still can't decrypt how he made them :D

What is different in the case of my remote compared to the remote used in this library, is that the array consists of 179 elements instead of 144. Also the transmission code is F20D05FA instead of F20D03FC, but I do not get where this is set either.

in the meanwhile I just managed to power off my AC unit using a RAW command:
```
  int khz = 38; // 38kHz carrier frequency for the NEC protocol

  unsigned int  irSignal[] = {4350,4300, 600,1550, 550,1600, 550,1600, 550,1500, 700,450, 600,500, 550,1500, 650,500, 600,450, 600,500, 550,500, 600,500, 550,1550, 600,1550, 600,500, 550,1600, 550,500, 600,450, 600,500, 600,450, 600,500, 550,1600, 550,500, 600,1550, 600,1550, 550,1550, 600,1550, 600,1550, 600,1550, 600,450, 600,1550, 600,500, 550,500, 600,500, 550,500, 600,450, 600,500, 600,450, 600,500, 550,1600, 550,500, 600,1550, 600,1550, 550,1600, 550,500, 600,450, 600,500, 600,450, 600,500, 550,500, 600,500, 550,500, 600,450, 600,1550, 600,1550, 600,1550, 600,450, 600,500, 600,450, 600,500, 550,500, 600,500, 550,500, 600,450, 600,500, 600,450, 600,1550, 600,450, 600,500, 600,450, 600,500, 600,1550, 550,500, 600,450, 600,500, 600,1550, 550,1600, 550,500, 600,1550, 600,450, 600,500, 550,1550, 600,500, 600,450, 600,1550, 600,1550, 600,500, 550,1550, 600};  // SAMSUNG F20D05FA  Serial.println("irSignal");
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz);
```

irSignal array is from the first post, a simple copy paste.
My second arduino, which works as a receiver for debugging purposes, doesn't get the same numbers, but the command works. Some sort of error correction maybe?

ToniA, 2016-12-11:
Can you try this one out: https://github.com/ToniA/Raw-IR-decoder-for-Arduino

I have done everything by reverse-engineering. First by writing a receiver module for this receiver sketch, and then by writing the transmitter library.

bserem, 2016-12-11:
Hurray, I got a responce!

This code https://github.com/ToniA/Raw-IR-decoder-for-Arduino was not working on the Arduino Mega. It took me some time to realize this.

On the UNO I get proper replies (you can see on/off below). Modes 1 and 4 seem to give correct results.

I know I'm still far away from actually decrypting all these, but its a start.
I'm gonna take a break and continue later.

```
Number of symbols: 182
Symbols:
Hh1111001000001101000001011111101000000001011100000000000000000000001000010001101001001010hHh1111001000001101000001011111101000000001011100000000000000000000001000010001101001001010
Bytes:
4F,B0,A0,5F,80,0E,00,00,84,58,52,4F,B0,A0,5F,80,0E,00,00,84,58,52
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  4290
HEADER SPACE: 4533
BIT MARK:     520
ZERO SPACE:   419
ONE SPACE:    1480
Decoding known protocols...
Unknown protocol

Number of symbols: 182
Symbols:
Hh1111001000001101000001011111101000000001011100000000011100000000001000010001101001001101hHh1111001000001101000001011111101000000001011100000000011100000000001000010001101001001101
Bytes:
4F,B0,A0,5F,80,0E,E0,00,84,58,B2,4F,B0,A0,5F,80,0E,E0,00,84,58,B2
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  4280
HEADER SPACE: 4533
BIT MARK:     520
ZERO SPACE:   420
ONE SPACE:    1478
Decoding known protocols...
Unknown protocol
```

ToniA, 2016-12-11:
Strange that it's not working on the Mega... That's what I'm using all the time.

bserem, 2016-12-11:
It runs, but after I select a brand it goes on and on even with no data from the IR Receiver.
I might have some time next weekend to check whats going on with it.

bserem, 2016-12-12:
So, after pressing some buttons on the remote to get it close to what the decoder expects, I got this reply:

```
Number of symbols: 150
Symbols:
Hh111100100000110100000011111111000000000101100000000000110000000001100010hHh111100100000110100000011111111000000000101100000000000110000000001100010
Bytes:
4F,B0,C0,3F,80,06,C0,00,46,4F,B0,C0,3F,80,06,C0,00,46
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  4300
HEADER SPACE: 4520
BIT MARK:     520
ZERO SPACE:   431
ONE SPACE:    1480
Decoding known protocols...
Looks like a Carrier protocol #1
FAN: AUTO
MODE: HEAT
Temperature: 23
ModeFan  11000000
Checksum 01000110 matches
```

Airflow direction (when set) doesn't resolve to a known protocol, but it looks like for normal operation I could go with Carrier. I know that Carrier and Toshiba work together (I think), so maybe the IR codes are indeed close.

Moving on, I adapted the Hisense demo file to send CarrierNQV commands and my AC unit responded.
My means of verifying it got the correct temperature is the "double beep" it does when I set the temperature on 24 degrees (single beep on anything else).

It took a day, but it was worth it.
Shall I prepare a PR with documendation improvements about Toshiba Daiseikai units (RAS-10G2KVP-E etc) and a sample sketch?

PS: on a side note, my IR led doesn't seem to be as powerful as the one found on the remotes. I need to get it about 1m for the unit for it to respond. @ToniA do you have any experience on that issue?

ToniA, 2016-12-12:
Thanks for the PR, I'll take a look at that when I have some spare time.

Yes, the IR led is not very powerful, I also get 1-2m range. You'd need an IR blaster for longer range.

bserem, 2016-12-12:
I'll try to find this one https://www.adafruit.com/products/387
in my market, and then search into the irblaster thing.

Thanks for the support. I guess we can close the issue?

MRuniversee, 2016-12-12:
I had the same issue so i connected the IR led through an 2N2222 NPN Transistor,it acts like a boost.

bserem, 2016-12-12:
For people that might read this issue:
http://www.riyas.org/2014/01/raspberry-pi-ir-blaster-control-your.html
http://forum.arduino.cc/index.php?topic=73732.0
https://forum.arduino.cc/index.php?topic=423788.0

@MRuniversee thanks for the info. I'll try it as soon as I get my hands on the Transistor

closed 2017-01-09 by bserem