ToniA/arduino-heatpumpir #25: I made a simple WebUI

bserem, 2016-12-16:
I made a simple WebUI for this library:
https://github.com/bserem/arduino-home-automation

At the moment it doesn't really provide the means to control the temperature from the UI, but
it is a start. I will work more on this, as I'd like to add Lamp controls and an LCD screen.

Maybe you'd like to link to this repo.
I could even make a simpler version of it and add it to this repo if you like the idea.

MRuniversee, 2016-12-16:
i have made something similar but i did it using the esp8266.

bserem, 2016-12-17:
I do not have an esp8266 board currently (I think I need to get one, or a nodemcu)
Is your code online/opensource? I still need to improve mine so that I can choose fan speed and temperature, instead of just on-auto/on-heat/on-cool/off which I do now.

ToniA, 2016-12-18:
This is an example of a simple web gui for Panasonic heatpumps. This doesn't have the HeatpumpIR integrated, as this was actually the grandfather of the current HeatpumpIR library. But I hope it gives you some ideas: https://github.com/ToniA/arduino-panasonic-heatpump-controller

bserem, 2016-12-18:
I'm pretty sure I can get the temperature/mode controls from your sketch and intergrate it into mine, with HeatpumpIR support.

Thanks, I'll keep you posted as I progress.

@ToniA do you know if there is any way to get the current AC status from these split units?
Do they emit some signal (IR or other) or are they capable only to receive?

ToniA, 2016-12-18:
All AC units I've seen have had one-way infrared communication. The only ways to get feedback is to either listen for the 'beep' the give on receive, or hook a wire into the status leds. I'm actually powering my MySensors heatpump nodes directly from the Panasonic's indoor unit, by stealing +5V and GND from the infrared eye. Planning to switch to ESP8266 (Wemos D1 mini) soon.

bserem, 2016-12-18:
The question was more about me getting the current settings of the AC unit on a WebUI/LCD by clicking a button "get status".
It isn't much of a problem, because you can always issue the command you want if you are unsure about the current status.

The Wemos D1 mini looks absolutely fantastic for what I want to do. Thanks again for pointing me to the right direction.

ToniA, 2016-12-18:
You may also want to take a look at this: https://github.com/ToniA/ESPEasy -> https://github.com/ToniA/ESPEasy/blob/master/ESPEasy/_P115_HeatpumpIR.ino

It's not a GUI for the heatpump, but rather a sensor for a home automation system like Domoticz or openHAB.

bserem, 2016-12-18:
I'll have a look, thought with a quick glance the code you use there is beyond my understanding.

Here is a working version of your WebUI combined with HeatpumpIR:
https://github.com/bserem/arduino-home-automation/blob/webui/arduino_home_automation.ino
I'll make it even better in the upcoming days (documentation, better UI etc).
You opened my eyes with "Webserver.h", and given that in reality I am frontend web developer I suppose I can make it look really nice (soon!).

bserem, 2016-12-19:
[image]
[image]

I do not know how you feel about the styling in the attached images.
I seperated on/off with two different buttons. The form currently works (see link in my previous comment) and is very easy to use on a mobile phone.
I use bootstrap (but the form will work without it too). I am missing a couple of "row" classes, but it is not visible to the human eye on such a simple design, so it is intentional.
I will prepare a stripped down version of it, that I will save in flash so that no internet will be required for the form to appear superb. If I get the time I'll also try to redo everything without boostrap, but without loosing the good looks.

Any feedback is most welcome.

MRuniversee, 2016-12-19:
What you have done looks nice! I will upload my code soon with the website code also. I did mine with the mqtt protocol because its a MUST on thngs like this.