ToniA/arduino-heatpumpir #26: IVT (SHARP) ON command seems wrong, was 0x31 and I must use 0x11

bjacobse, 2016-12-28:
Hi, I have an IVT heatpump, and use Sharp codes, however I can't switch ON my heatpump with existing code. But if I edit SharpHeatpumpIR.h and change 0x31 to 0x11 as below codesnippet then it is working:
//define SHARP_AIRCON1_MODE_ON    0x31 // Power ON
#define SHARP_AIRCON1_MODE_ON    0x11 // Power ON IVT

I have attached below power off and power on from my IVT remote control CRMC-A673JBEZ, that shows to use 0x11 for switch ON command
I'm not sure if this is for IVT only? or also Sharp, and then switch on command was a type when originally created by ToniA? does anyon have a real SHARP heatpump and can confirm it's a typo, or shall there be cloning of SHARP and then create a separate IVT?

Select model to decode (this affects the IR signal timings detection):
* '1' for Panasonic DKE>, Mitsubishi Electric, Fujitsu etc. codes
* '2' for Panasonic CKP, Midea etc. codes
* '3' for Mitsubishi Heavy etc. codes
* '4' for Hyyndai etc. codes
* '9' for entering the bit sequence on the serial monitor (instead of the IR receiver)

Enter choice: 1

Ready to decode IR for choice '1'

Number of symbols: 107
Symbols:
Hh01010101010110101111001100001000011000001000010010001100010010000001000000000001000000000000111110001110
Bytes:
AA,5A,CF,10,06,21,31,12,08,80,00,F0,71
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  3660
HEADER SPACE: 1780
BIT MARK:     388
ZERO SPACE:   419
ONE SPACE:    1315
Decoding known protocols...
Looks like a Sharp protocol
POWER OFF
MODE HEAT
Temperature: 23
FAN: 1
Checksum '0x7' matches

Number of symbols: 107
Symbols:
Hh01010101010110101111001100001000011000001000100010001100010010000001000000000001000000000000111110000010
Bytes:
AA,5A,CF,10,06,11,31,12,08,80,00,F0,41
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  3660
HEADER SPACE: 1780
BIT MARK:     394
ZERO SPACE:   415
ONE SPACE:    1314
Decoding known protocols...
Looks like a Sharp protocol
MODE HEAT
Temperature: 23
FAN: 1
Checksum '0x4' matches

ToniA, 2016-12-28:
All I ever had was the Sharp remote controller CRMC-A788JBEZ... 0x31 vs. 0x11 is just one bit difference. I think we'd just need to make another variant of the SharpHeatpumpIR class, very much the same way as the Carrier and Panasonic heatpumps have different implementations of the common base class .

bjacobse, 2016-12-29:
yes it's best to make a separate IVT model - I'm not sure how I can do this? or can I ask you to clone and split this?
My IVT indoor unit model: AY-XP12FR-N
Remote control CRMC-A673JBEZ

ToniA, 2016-12-29:
Hi, would you try out the latest version, and let me know if it works? I just quickly coded it, without running in hardware.

bjacobse, 2016-12-29:
Verified to work for my IVT, kiitos
And here you can see how I have added the controller to my IVT heatpump
https://forum.mysensors.org/topic/3383/heatpump-airconditioner-controller/3

closed 2016-12-29 by bjacobse

ToniA, 2016-12-30:
I made some final touches, to to further simplify the IVTHeatpumpIR class. After all, this is OO programming, so I don't need to repeat myself that much :)

bjacobse, 2016-12-30:
do want me to download and verify after your final touch? I can try this tonight

ToniA, 2016-12-30:
It would be nice if you would do this. Thanks in advance :)

bjacobse, 2016-12-31:
When  I assembled my Arduino after uploading the final touch IVT sketch, I managed to put VSS to pin3, so I killed the Ardduino. today I soldered pinheaders on a new Arduino. and uploaded the sketch. it's verified to be working good can switch on and off my IVT. Paljon Kiitos. OFF-topic item, do you know Kitta Virtavuo, she was my girlfriend when I lived in Turku, and she is originally from Kaarina