ToniA/arduino-heatpumpir #27: Any luck with Demos D1 mini or other ESP8266?

bserem, 2017-01-16:
Has anybody any working samples of a HeatpumpIR interface running on ESP8266?

I've had great success with arduino+ethernet (using the webduino library), but I can't say the same when I try to port my code on  Wemos/ESP8266.
I tried to simplify things a lot (just on/off for the AC unit), but I get an exception when I try to send an IR command.

Code:
```
//This example will set up a static IP - in this case 192.168.2.3

#include <ESP8266WiFi.h>

const char* ssid = "";
const char* password = "";

WiFiServer server(80);
IPAddress ip(192, 168, 2, 3); // where xx is the desired IP Address
IPAddress gateway(192, 168, 2, 1); // set gateway to match your network

// Daiseikai IR
#include <Arduino.h>
#include <ToshibaDaiseikaiHeatpumpIR.h>
// Support for WemosD1
#ifndef ESP8266
IRSenderPWM irSender(9);       // IR led on Arduino digital pin 9, using Arduino PWM
//IRSenderBlaster irSender(3); // IR led on Arduino digital pin 3, using IR Blaster (generates the 38 kHz carrier)
#else
IRSenderBitBang irSender(1);   // IR led on ESP8266 GPIO pin 1
#endif

ToshibaDaiseikaiHeatpumpIR *heatpumpIR;

void setup() {
  Serial.begin(115200);
  delay(10);

  Serial.print(F("Setting static ip to : "));
  Serial.println(ip);

  // Connect to WiFi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your network
  WiFi.config(ip, gateway, subnet);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");

  // Start the server
  server.begin();
  Serial.println("Server started");

  // Print the IP address
  Serial.print("Use this URL : ");
  Serial.print("http://");
  Serial.print(WiFi.localIP());
  Serial.println("/");

}

void loop() {
  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }

  // Wait until the client sends some data
  Serial.println("new client");
  while (!client.available()) {
    delay(1);
  }

  // Read the first line of the request
  String request = client.readStringUntil('\r');
  Serial.println(request);
  client.flush();

  // Match the request

  int value = LOW;
  if (request.indexOf("/LED=ON") != -1) {
    heatpumpIR->send(irSender, POWER_ON, MODE_AUTO, FAN_AUTO, 24, VDIR_AUTO, HDIR_AUTO);
    value = HIGH;
  }
  if (request.indexOf("/LED=OFF") != -1) {
    heatpumpIR->send(irSender, POWER_OFF, MODE_AUTO, FAN_AUTO, 24, VDIR_AUTO, HDIR_AUTO);
    value = LOW;
  }

  // Return the response
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: text/html");
  client.println(""); //  do not forget this one
  client.println("<!DOCTYPE HTML>");
  client.println("<html>");

  client.print("Last command: ");

  if (value == HIGH) {
    client.print("On");
  } else {
    client.print("Off");
  }
  client.println("<br><br>");
  client.println("Click <a href=\"/LED=ON\">here</a> to turn ON<br>");
  client.println("Click <a href=\"/LED=OFF\">here</a> to turn OFF<br>");
  client.println("</html>");

  delay(1);
  Serial.println("Client disconnected");
  Serial.println("");

}
```

Exception:
```
WiFi connected
Server started
Use this URL : http://192.168.2.3/
new client
GET /LED=OFF HTTP/1.1

Exception (28):
epc1=0x40202170 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffefc40 end: 3ffefe50 offset: 01a0

>>>stack>>>
3ffefde0:  3fffdad0 3ffeedfc 00000000 40202164  
3ffefdf0:  3ffe8df2 3ffeec0c 3ffeedfc 4020200c  
3ffefe00:  3ffe8c50 00000000 000003e8 000077f8  
3ffefe10:  00000000 3fff1104 00000000 00000000  
3ffefe20:  00000000 3fff11dc 0000001f 00000015  
3ffefe30:  3fffdad0 00000000 3ffeee20 40203a28  
3ffefe40:  feefeffe feefeffe 3ffeee30 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v0291a6e3
~ld
```

MRuniversee, 2017-01-16:
You have to use MQTT Broker for this kind of stuff, you cant keep asking the server for changed values(spamming it),also if you want to use the send command for the esp8266 i think you have to use heatpumpIR->sendPWM  or something like this,its in the comments you  cant miss it.

bserem, 2017-01-16:
I'll check sendPWM. I do not currently have the time to learn about MQTT, hopefully soon enough.

MRuniversee, 2017-01-16:
Να δεις σιγουρα ομως το MQTT  αλλιως δεν εχει νοημα!

ToniA, 2017-01-16:
Yes, take a look at this: https://github.com/ToniA/ESPEasy/blob/master/ESPEasy/_P115_HeatpumpIR.ino

I have already integrated this into the ESPEasy firmware, it probably has everything you need.

bserem, 2017-01-17:
Oh boy... This adventure is full of new things to learn!
I did a small research on ESPEasy (new term!), it now runs on my Wemos. Still haven't connected it to Domoticz (another new term), but I'll get there.
At the moment I can succesfully control the unit with http requests:
`http://192.168.2.3/control?cmd=heatpumpir,toshiba_daiseikai,1,2,0,24,0,0`
so, I am ahead of where I was yesterday.

Lessons learned and thoughts:
1) I was wrong to think of Wemos+NodeMCU as "Arduinos with Wifi". They are way more capable.
Of course, the complexity increases dramatically, but then again many things are way easier to implement now (htauth, web UI, change settings without flashing everything all over).

2) I suppose I'll also have to create a PR against ESPEasy, to add support for the new AC units (Daiseikai) just added in heatpump-ir. Maybe there is a way to automatically load all new models? So that we don't need to update ESPEasy code everytime this library gets updated?

I am asking a lot of questions, thanks for bearing my ignorance :)

ToniA, 2017-01-17:
I added some comments into the readme of this repository, I hope that helps.

The ESPEasy does not include this library, it only uses it in the build, so if the HeatpumpIR is updated, and ESPEasy is rebuilt with the updated HeatpumpIR, it will pick up the newest changes.

bserem, 2017-01-23:
Wow! That wiki entry has some really nice info.

I am still thinking that all these are an overkill to what I need though, and I prefer to not have a domoticz server running, but the maintainance of ESPEasy is easier than my custom systems.

MRuniversee, 2017-02-02:
@bjacobse hello ! do you maybe know where i can find the html code for the interface they have created ?

bjacobse, 2017-02-02:
@MRuniversee do you maybe know where i can find the html code for the interface they have created ?

I'm not sure I understand your question, the HTML code used in the Wemos D11 Mini Pro is from Espeasy, and the HTML code used on the Domoticz is created by Domoticz, those are created when you add a "dummy switch"
Maybe you can re-phrase and ask your question again?

MRuniversee, 2017-02-02:
@bjacobse oh im sorry if my question was wrong , i would like to use these "https://www.domoticz.com/wiki/images/0/02/Domo_AC.png" on my custom website , but for that i have to find the html/css code he is using. i hope my question is a bit more clear now.

bjacobse, 2017-02-02:
yes much better, then the files are stored here in this DIR:
https://github.com/domoticz/domoticz/blob/master/www/images/Fan.png

andrewerrington, 2017-11-23:
I built this:
https://github.com/krzynio/esp8266-ir-ac-remote

I am using a Wemos D1 Mini Pro with an IR LED and a 150R resistor directly to pin 5 (#D1) to control my Fujitsu heat pump (AR-RAH1E remote control).

I'm not convinced the Fujitsu protocol is quite correct so I am using the Arduino IR decoder to check a few things. It does turn the unit on and off though, via the HTML (REST) interface.

bjacobse, 2017-11-24:
I think your Fujitsu is a different model, and requires different codes.
The code below specifies:
Fujitsu Nocria (AWYZ14) heatpump control (remote control P/N AR-PZ2)

https://github.com/ToniA/arduino-heatpumpir/blob/master/FujitsuHeatpumpIR.h

I remember that Toni have created an Arduino sketch that can read the IR to get timing and values.
You need an IR reciever to your Arduiino to recive the remote control IR codes. Check the link in below text to get raw codes.

All these heatpump / A/C models have been reverse-engineered, by decoding the IR signal from the remote control. See the sketch https://github.com/ToniA/Raw-IR-decoder-for-Arduino for more information.

andrewerrington, 2017-11-24:
Hi there. No, the codes are the same for my model, and probably all Fujitsu heat pumps. I was not clear in my message. I meant to say there are some things missing from the Fujitsu code implementation, and I am working to discover what they are. I will post new code and documentation shortly.