ToniA/arduino-heatpumpir #30: Support for Mitsubishi SEZKDVAL heatpump model

markcame, 2017-05-09:
I've reversed the protocol of SEZ series of mitsubishi and i've write the code if you want i can share the info to integrate in the existing code

ToniA, 2017-05-10:
Thanks, I'll take a closer look at that over the next weekend. The code looked nice to me.

markcame, 2017-05-10:
ok just a clarification, the timings for MARK, SPACE etc.. was adapted to match the timings decoded with the original remote and the raw-decoder you provide. Actually i don't know why i put a value and this was not respected by the arduino so the value you see in the file is not the real, i can provide the grab from my original remote . I think that is something reletad to the timings managed by my arduino. Do you have an idea why ?

ToniA, 2017-05-13:
I merged the code, many thanks for the contribution :)

The timing constants in the code are not quite the same as the ones given by the decoder. The way I adjust these is to capture the real remote timings with the decoder, and then program one Arduino with the HeatpumpIR sender, and capture the codes again with the decoder. Then I adjust the timing constansts until they are close enough. None of the A/C models I have tried out have been too picky about the timings.

closed 2017-05-13 by ToniA