ToniA/arduino-heatpumpir #40: Timer led turned on

mortenmathiasen, 2017-10-18:
On my Panasonic JKE model the library turns on blinking the Timer led. Why?

ToniA, 2017-10-19:
It should not work that way... The problem is that I don't have access to HExJKE-series, but just assumed it works based on the feedback from others. Is there any chance you could build in IR receiver using an IR receiver module + any Arduino board? This way we could record samples from the real remote, and compare against what the library sends.

mortenmathiasen, 2017-10-20:
Yes, I can record the remote signals next time I travel to my vacation house where the heatpump and its remote are located. I could record the signals by [http://zmote.io/](http://zmote.io/) but I suspect that you prefer the signals recorded by [https://github.com/ToniA/Raw-IR-decoder-for-Arduino](https://github.com/ToniA/Raw-IR-decoder-for-Arduino) ?

mortenmathiasen, 2018-01-13:
Now I got recorded the remotes signals. The resulting report is available [here](http://tranebaerkaeret.dk/owncloud/index.php/s/dm5JJMWQEPi2waH). How do we proceed from here?

mortenmathiasen, 2018-02-03:
Any progress on this one?

mortenmathiasen, 2018-02-16:
I have now adjusted the library and made the Arduino send IR-signal similar to the original Heatpump remote. For a test I have recorded the signals sent from both two devices.

My setup is MySensors based and you find my Arduino sketch at [MySensors.ino](https://github.com/mortenmathiasen/arduino-heatpumpir/blob/master/examples/MySensors/MySensors.ino)

Arduino sends:
> Number of symbols: 222
> Symbols:
> Hh0100000000000100000001110010000000000000000000000000000001100000WHh01000000000001000000011100100000000000001001000001010100000000011111010110110000000000000111000000000111000000000000000010000001000000000000000000100111
> Bytes:
> 02,20,E0,04,00,00,00,06,02,20,E0,04,00,09,2A,80,AF,0D,00,0E,E0,00,00,81,00,00,E4
> Timings (in us): 
> PAUSE SPACE:  10500
> HEADER MARK:  3630
> HEADER SPACE: 1720
> BIT MARK:     435
> ZERO SPACE:   340
> ONE SPACE:    1260

Original remote sends:
> Number of symbols: 222
> Symbols:
> Hh0100000000000100000001110010000000000000000000000000000001100000WHh01000000000001000000011100100000000000001001000001010100000000011111010110110000000000000111000000000111000000000000000010000001000000000000000000100111
> Bytes:
> 02,20,E0,04,00,00,00,06,02,20,E0,04,00,09,2A,80,AF,0D,00,0E,E0,00,00,81,00,00,E4
> Timings (in us): 
> PAUSE SPACE:  10540
> HEADER MARK:  3620
> HEADER SPACE: 1700
> BIT MARK:     440
> ZERO SPACE:   340
> ONE SPACE:    1250

Signals from the two devices are equal, but the heatpump starts blinking with error H11 when signal comes from the Arduino.
Any suggestions on how to solve this problem?

ToniA, 2018-02-18:
I'm quite puzzled... The Panasonic protocol ha snot changed much since the DKE series, and I haven't heard of issues like this. Like you state, the signals look identical, the timings are very much the same, and still you get a different result.

H11 (Indoor/Outdoor Abnormal Communication) sounds very strange, I mean as a result of receiving an infrared command.

mortenmathiasen, 2018-02-18:
I agree that the error is strange. In addition, signals from the Arduino have a short recognition distance. Right now, my best guess is that the problem is a timing issue. Hence, I consider to check the timings, e.g. by using [IRLib2](https://github.com/cyborg5/IRLib2) - would that be your suggestion too?

ToniA, 2018-02-18:
The short distance could be because of timings, or if the frequency is a bit off. At least on ESP8266 I understood using a bit more accurate send method gives a better range.

mortenmathiasen, 2018-03-28:
Now, I have used [IRLib2](https://github.com/cyborg5/IRLib2) to compare the timings of the IR signals from original remote and arduino:

Signal from original remote:
`
[org_remote_timings.txt](https://github.com/ToniA/arduino-heatpumpir/files/1854971/org_remote_timings.txt)
`

Signal from Arduino:
`
[arduino_remote_timings.txt](https://github.com/ToniA/arduino-heatpumpir/files/1854975/arduino_remote_timings.txt)
` 

My Arduino sketch is using these timing settings:
`
#define PANASONIC_AIRCON2_HDR_MARK   3400 
#define PANASONIC_AIRCON2_HDR_SPACE  1700 
#define PANASONIC_AIRCON2_BIT_MARK   500 
#define PANASONIC_AIRCON2_ONE_SPACE  1260
#define PANASONIC_AIRCON2_ZERO_SPACE 397 
#define PANASONIC_AIRCON2_MSG_SPACE  10220
`
 
Any suggestions on how to solve my problem?