ToniA/arduino-heatpumpir #41: Gree module not working as expected

zombiehunter1, 2017-11-01:
I have a Delonghi 5kw Aircon  using rawdecode  I get the following output from the original remote pressing the power button 

```
Number of symbols: 71
Symbols:
Hh10010010011010111101010010100000000100000000000101100100000001011010
Bytes:
49,D6,2B,05,08,80,26,A0
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  9240
HEADER SPACE: 4560
BIT MARK:     567
ZERO SPACE:   498
ONE SPACE:    1652
Decoding known protocols...
Looks like a Gree protocol
Checksum matches
POWER ON
MODE COOL
Temperature: 22
FAN: AUTO
SLEEP: OFF
SWING: ON
```

So I tried the Gree protocol in a test application but the air con does not respond and raw decode shows the following output 

```
Number of symbols: 71
Symbols:
Hh00100000000100000000000000000000010W00000000000001000000000000000000
Bytes:
04,08,00,00,00,20,00,00
Timings (in us): 
PAUSE SPACE:  19480
HEADER MARK:  9340
HEADER SPACE: 4000
BIT MARK:     655
ZERO SPACE:   448
ONE SPACE:    1539
Decoding known protocols...
Looks like a Gree protocol
Checksum matches
POWER OFF
MODE HEAT
Temperature: 24
FAN: AUTO
SLEEP: OFF
SWING: OFF
```

I'm not really sure where to go from here

zombiehunter1, 2017-11-02:
I've used z3t0/Arduino-IRremote app to get some data dumps I really have little clue on how to interrupt
the data other than seeing the bits change but I don't know how to apply it to a module  

Power On rawdump
```

Encoding  : NEC
Code      : 5A929C20 (32 bits)
Timing[139]: 
     +8900, -4400     + 650, - 550     + 550, -1700     + 550, - 550
     + 650, -1600     + 600, -1650     + 650, - 500     + 650, -1600
     + 650, - 500     + 600, -1650     + 600, - 550     + 600, - 550
     + 600, -1600     + 650, - 500     + 600, - 550     + 600, -1650
     + 600, - 550     + 600, -1650     + 600, - 550     + 550, - 550
     + 600, -1650     + 650, -1600     + 650, -1600     + 650, - 500
     + 600, - 550     + 600, - 550     + 550, - 550     + 650, -1600
     + 650, - 500     + 600, - 550     + 550, - 600     + 600, - 500
     + 650, - 500     + 650, - 500     + 550, - 600     + 600, - 500
     + 650, -1600     + 600, - 550     + 650, - 500     + 600, - 550
     + 600, - 550     + 550, - 550     + 650, - 500     + 600, - 550
     + 600, - 550     + 550, - 550     + 600, - 550     + 600, - 550
     + 600, -1650     + 550, - 550     + 650, -1650     + 600, -1650
     + 600, - 500     + 650, - 500     + 600, -1650     + 600, - 550
     + 600, - 550     + 600, - 500     + 600, - 550     + 650, - 500
     + 600, - 550     + 600, - 500     + 600, -1650     + 650, - 500
     + 650, -1600     + 600, -1650     + 600, - 550     + 550, -1700
     + 600, - 550     + 600
unsigned int  rawData[139] = {8900,4400, 650,550, 550,1700, 550,550, 650,1600, 600,1650, 650,500, 650,1600, 650,500, 600,1650, 600,550, 600,550, 600,1600, 650,500, 600,550, 600,1650, 600,550, 600,1650, 600,550, 550,550, 600,1650, 650,1600, 650,1600, 650,500, 600,550, 600,550, 550,550, 650,1600, 650,500, 600,550, 550,600, 600,500, 650,500, 650,500, 550,600, 600,500, 650,1600, 600,550, 650,500, 600,550, 600,550, 550,550, 650,500, 600,550, 600,550, 550,550, 600,550, 600,550, 600,1650, 550,550, 650,1650, 600,1650, 600,500, 650,500, 600,1650, 600,550, 600,550, 600,500, 600,550, 650,500, 600,550, 600,500, 600,1650, 650,500, 650,1600, 600,1650, 600,550, 550,1700, 600,550, 600};  // NEC 5A929C20
unsigned int  data = 0x5A929C20;
```

Power off raw dump

```
Encoding  : NEC
Code      : 4A905C20 (32 bits)
Timing[139]: 
     +8900, -4450     + 600, - 550     + 550, -1700     + 600, - 550
     + 600, - 500     + 650, -1600     + 650, - 500     + 550, -1700
     + 550, - 600     + 600, -1650     + 600, - 500     + 650, - 500
     + 650, -1600     + 650, - 500     + 600, - 550     + 600, - 550
     + 600, - 500     + 650, - 500     + 650, -1600     + 650, - 500
     + 600, -1600     + 650, -1650     + 600, -1650     + 600, - 550
     + 550, - 600     + 600, - 500     + 650, - 500     + 600, -1650
     + 600, - 550     + 600, - 550     + 600, - 500     + 650, - 500
     + 600, - 500     + 650, - 500     + 650, - 500     + 650, - 500
     + 600, -1650     + 600, - 550     + 600, - 550     + 600, - 500
     + 650, - 500     + 650, - 500     + 600, - 550     + 600, - 500
     + 650, - 500     + 650, - 500     + 600, - 500     + 650, - 550
     + 600, -1650     + 600, - 500     + 600, -1650     + 600, -1650
     + 600, - 550     + 550, - 600     + 600, -1650     + 600, - 550
     + 600, - 500     + 650, - 500     + 600, - 550     + 600, - 550
     + 600, - 500     + 650, - 500     + 600, -1650     + 600, - 550
     + 600, -1650     + 600, -1600     + 600, - 600     + 600, -1650
     + 550, - 550     + 650
unsigned int  rawData[139] = {8900,4450, 600,550, 550,1700, 600,550, 600,500, 650,1600, 650,500, 550,1700, 550,600, 600,1650, 600,500, 650,500, 650,1600, 650,500, 600,550, 600,550, 600,500, 650,500, 650,1600, 650,500, 600,1600, 650,1650, 600,1650, 600,550, 550,600, 600,500, 650,500, 600,1650, 600,550, 600,550, 600,500, 650,500, 600,500, 650,500, 650,500, 650,500, 600,1650, 600,550, 600,550, 600,500, 650,500, 650,500, 600,550, 600,500, 650,500, 650,500, 600,500, 650,550, 600,1650, 600,500, 600,1650, 600,1650, 600,550, 550,600, 600,1650, 600,550, 600,500, 650,500, 600,550, 600,550, 600,500, 650,500, 600,1650, 600,550, 600,1650, 600,1600, 600,600, 600,1650, 550,550, 650};  // NEC 4A905C20
unsigned int  data = 0x4A905C20;
```

ToniA, 2017-11-02:
You have used my 'Raw-IR-decoder-for-Arduino' sketch, and I think this is the way to go. The protocol selection in the decoder is not very selective, so it accepted the DeLonghi code as Gree code just based on the fact that both have 71 bits of data.

The first thing to do would be to duplicate the file https://github.com/ToniA/Raw-IR-decoder-for-Arduino/blob/master/Gree.cpp, write a better detection (for example if the first bytes are always the same) to tell Gree and DeLonghi apart. And then just start looking at the code, and start writing the decoder. Once the decoder is working, writing a transmitter into this library is easy, as you have something to verify against.

zombiehunter1, 2017-11-04:
okay I've decoded up to the time, timer on, timer off bits not sure if I need to decode these as well 

Hh10011010000010101110100001010000001111110000001111110000000001011010

I've get 
Power 
```
ON    Hh100[1]
OFF   Hh100[0]

```
Mode 
```
Cool   Hh[100]1
Dry    Hh[010]1
Fan    Hh[110]1
Heat   Hh[001]1
Auto   Hh[000]1
```

Fan Speed
```
Auto   Hh1001[00]
1      Hh1001[10]
2      Hh1001[01]
3      Hh1001[11]
```

Vertical Swing

```
On    Hh100110[1]
Off   Hh100110[0]
```
Sleep 
```
On    Hh1001101[1]
Off   Hh1001101[0]
```

Temperature 
This is a little tricky as bits need reversing 

```
16c  Hh10011010[00001]    00001  = 10000
17c  Hh10011010[10001]    
18c  Hh10011010[01001]     01001 = 10010
```

The rest of the binary string looks like time and timer settings also special features like display,, boost 
So to summarize
           [mode] [power] [fan] [vertical] [sleep] [temp] [timers and stuff]
Hh[100][1][10][1][0][00001][0101110100001010000001111110000001111110000000001011010]

I have looked at the gree  .cpp and h files but I'm completely lost I really don't understand binary so well any pointers on creating a new module will be most helpful  Thanks

zombiehunter1, 2017-11-05:
So I managed to modify the gree module over to work with my Delonghi. I'm stuck on calculating the check sum, currently I'm manually setting it to hex 5A  which I decoded from the end of the string.

What is the correct way to calculate this?

ToniA, 2017-11-06:
The stock checksum calculation is not working? https://github.com/ToniA/arduino-heatpumpir/blob/master/GreeHeatpumpIR.cpp#L166

Does the decoder always say the checksum is correct (even though it does not understand the protocol)?

N-Storm, 2019-04-01:
@abdulmannan this issue was opened long time before Gree YAA support was added.

abdulmannan, 2019-04-01:
> @abdulmannan this issue was opened long time before Gree YAA support was added.

I don't get your point. I think it is irrelevant when this issue was open, if code is not working good.
This is the corrected code: https://github.com/ToniA/arduino-heatpumpir/issues/69
PS: this issue was open on Nov 1, 2017 while Gree YAA support was added on Sept 15, 2016.

N-Storm, 2019-04-02:
> PS: this issue was open on Nov 1, 2017 while Gree YAA support was added on Sept 15, 2016.

@abdulmannan, Gree YAA support was added by me on 7 Aug 2018. Please check pull request https://github.com/ToniA/arduino-heatpumpir/pull/60
I've wrote checksum calculation routine based on the PDF document describing protocol which you can find on the pull request link. Here is a screenshot from it showing checksum calculation routing:
[image]
You've fixed it with your commit based on IRremoteESP8266 library routine which adds a few more steps. I'm not sure which routine are correct, because IRremoteESP8266 lib claims that Gree support are untested too (Status: ALPHA / Untested). Have you tested your commit?

abdulmannan, 2019-04-19:
> > PS: this issue was open on Nov 1, 2017 while Gree YAA support was added on Sept 15, 2016.
> 
> @abdulmannan, Gree YAA support was added on 7 Aug 2018. Please check pull request #60
> I've added template routine based on the PDF document describing protocol which you can find on the pull request link. Here is a screenshot from it showing checksum calculation routing:
> [image]
> It has the same checksum calculation method as it used to be before in this library for generic Gree models (including YAA models, noted that and edited my comment).
> You've changed it with your commit based on IRremoteESP8266 library routine which adds a few more steps. I'm not sure which routine are correct, because IRremoteESP8266 lib claims that Gree support are untested too (Status: ALPHA / Untested). Have you tested your change? Could you confirm it works for YAA remote?
> I'm asking because there might be different checksum algorithms on different Gree units.

I have this Gree remote and old checksum code was not working with it. Modified checksum code (with help from  IRremoteESP8266 lib) is working fine with this remote.
[image]

I am now also confused how is it possible that there are different checksum calculations for different Gree YAA remotes.

Please let me know if anything is needed from my side to resolve this issue.

N-Storm, 2019-04-24:
Ok then I suppose if you tested this and it works when the document I was referring to was incorrect. Probably there are more other Gree remotes revisions with other checksum codes. If someone finds that new checksum routine does NOT works for them they will need to open a new issue then.