ToniA/arduino-heatpumpir #42: Daikin ARC480A6 - FTXN18NMVJU Decode

dancwilliams, 2017-11-03:
Hello,

First I would like to let you know that my FTXN18NMVJU, which uses the ARC480A6 remote, works great with the DaikinHeatpumpIR library.  All the modes, temp, and fan are working fine for me.  Thanks for putting this together!

I would like to get vswing working.  My model does not have an hswing.  I only have a single swing button on my remote to start or stop swing.

I am new to this IR decoding stuff so let me know if more information is needed.

I have built your decoder as described, but do not know how to decode the data and was hoping to get some help.  My remote is currently set to Temp: 76F, Fan: Auto, Mode: Cool.

When I hit the swing button to start swing I get the following information:

```
Number of symbols: 161
Symbols:
00000WHh10001000010110111110010000000000000000001000110010001100000000001111010100000000000000000000000000000000000000000000000010100011000000000000000000010111
Bytes:
11,DA,27,00,00,31,31,00,AF,00,00,00,00,00,00,C5,00,00,E8
Timings (in us): 
PAUSE SPACE:  25900
HEADER MARK:  3500
HEADER SPACE: 1780
BIT MARK:     360
ZERO SPACE:   408
ONE SPACE:    1310
Decoding known protocols...
Unknown protocol
```

And when I hit the button again to turn it off I get the following:

```
Number of symbols: 161
Symbols:
00000WHh10001000010110111110010000000000000000001000110010001100000000000000010100000000000000000000000000000000000000000000000010100011000000000000000010011011
Bytes:
11,DA,27,00,00,31,31,00,A0,00,00,00,00,00,00,C5,00,00,D9
Timings (in us): 
PAUSE SPACE:  25820
HEADER MARK:  3580
HEADER SPACE: 1700
BIT MARK:     440
ZERO SPACE:   340
ONE SPACE:    1241
Decoding known protocols...
Unknown protocol
```

Hope this helps!  If there is anything else you need (or if I am totally off base) just let me know.

Dan

ToniA, 2017-11-03:
Thanks Dan :)

I got this particular Daikin module as a contribution, so I can't test it out myself. But looking at your sample, I'd think the swing is in byte 8, the lower part of the byte is either 0 or F. Could you try if changing the byte 8 from 0xDA to 0xD0 or 0xDF on this row affects the swing:

https://github.com/ToniA/arduino-heatpumpir/blob/master/DaikinHeatpumpARC417IR.cpp#L92

If so, adding the swing command would be easy.

dancwilliams, 2017-11-03:
I will test tomorrow for sure. 

I see that you referenced the DaikinHeatpumpARC417IR module, but I am using the DaikinHeatpump module personally. Should I change libraries for my project and test that? I have not fully looked through the code so if the reference one another I apologize. 

Thanks,

Dan

closed 2020-05-09 by dancwilliams