ToniA/arduino-heatpumpir #44: PanasonicHeatpumpIR.cpp

flashtel, 2018-01-06:
Hi,

I'm Paul, after some research I found this repository. I have had some success with it today for a Panasonic HeatPump.

I did have to make one modification to the code to enable power off functionality.

I added this into  PanasonicHeatpumpIR.cpp

case POWER_OFF:
      operatingMode |= PANASONIC_AIRCON2_MODE_OFF;
      break;

Into the code at this location.

switch (powerModeCmd)
  {
    case POWER_ON:
      operatingMode |= PANASONIC_AIRCON2_MODE_ON;
      break;
   case POWER_OFF:
      operatingMode |= PANASONIC_AIRCON2_MODE_OFF;
      break;
  }

I've not made any repositories of my own and I am very new to the programming environment, my main interest is to integrate this library into tasmota

https://github.com/arendst/Sonoff-Tasmota

This is becoming a very versatile mqtt firmware for the sonoff device D1 mini and many more.

Any advice on who and how to ask nicely would be appreciated.

Thanks
Paul

mortenmathiasen, 2018-01-13:
[Sonoff-Tasmota](https://github.com/arendst/Sonoff-Tasmota) is something different. It is primaryly used for turning on and off power. To control a Heat pump you should consider to use a home automation system instead. I am controlling Panasonic heat pump with [Home assistant](https://home-assistant.io/hassio/) using heatpumpir. To turn on and off power for a simple fan I use [Sonoff-Tasmota](https://github.com/arendst/Sonoff-Tasmota) connected to the same [Home assistant](https://home-assistant.io/hassio/) instance.

flashtel, 2018-01-13:
Hi Morten,

Tasmota is well integrated with Home Assistant and the library’s are increasing IR/Counters (Power)/Temperature, Humidity, I2C, Dimmers, RF Bridge, button platform that can publish directly to mqtt without hass.

Here is the full command list.

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

The native platform support has extended far beyond sonoff.

I’m open to use anything, the only reason I want use tasmota is because I have one firmware to maintain for all devices.

Multi language support (I’m an native english speaker) it is cool they are acomadating 
I’ve started on the heatpumpir integration and it appears to have drv.ino that you can add and enable at compile time, which is useful to hook what ever library you choose.

I’ve copied xdrv_irremote.ino to xdrv_heatpumpir.ino, this is my first coding attemtp in anger as well.

It’s good having a template and a requirement.

If you have something you can suggest for mqtt/heatpumpir/hass let me know.

Thanks
Paul

mortenmathiasen, 2018-01-13:
I use wireless integration betweeen heatpump to MQTT by using [MySensors](https://www.mysensors.org/build/heatpump). It suggests to use Domoticz as home automation system. However, I choose to use Home Assistant because it allows me to create a simpler user interface and do customized settings/programming.

brucysNZ, 2019-06-15:
Hi @mortenmathiasen 
Is your heat pump and Mqtt still working?
I would like to know more as this is my next project at home.
Using home assistant and familiar with Tasmota. 
Not sure about the board or flashing your using though
Cheers

mortenmathiasen, 2019-06-15:
MQTT and Home Assistant are working just fine on the Raspberry PI board. However, I have dropped the Arduino board because it has a hard time sending a precise signal to the heat pump. Heat pump IR signals are very long and complicated strings which means that the timing has to be very precise. Checking my signal from the arduino and the original remote control showed that the content from the arduino was correct, but the timing was very poor which meant that the heat pump was not able to interpret the signal from the arduino correctly.

N-Storm, 2019-06-15:
@mortenmathiasen which IRSender class you've used with Arduino? If you've used IRSenderPWM it should be pretty accurate.

mortenmathiasen, 2019-06-15:
If used [MySensors](https://www.mysensors.org/build/heatpump) which does this:
`
#ifndef ESP8266
IRSenderPWM irSender(9);       // IR led on Arduino digital pin 9, using Arduino PWM
//IRSenderBlaster irSender(3); // IR led on Arduino digital pin 3, using IR Blaster (generates the 38 kHz carrier)
#else
IRSenderBitBang irSender(D2);  // IR led on Wemos D1 mini, connect between D2 and G
#endif
`
As I used Arduino micro I suspect that I didn't use IRSenderPWM. I think I e.g. would need to use [NodeMCU](https://www.nodemcu.com/index_cn.html) instead to use IRSenderPWM?

brucysNZ, 2019-06-15:
@mortenmathiasen 
So being very new to this type of setup and programming. I would very much like to setup one of these units for our Heat Pump. Not sure if they model is noted in the available files thought, as we have a Panasonic E18PKR model in NZ. Hope it does
In terms of software and hardware
1. You use the Sensebender + nRF24L01. Could I use a D1 Mini or a NodeMCU instead?
2. Software. I am guessing I would use Panasonic DKE version. However how do I get this onto the hardware board and what other software is required

Sorry for all the questions and my lack of understanding, however I am keen to learn and build it for our home

Thanks

ToniA, 2019-06-15:
I would suggest using the espeasy firmware. I have built some binaries a while ago: https://github.com/ToniA/ESPEasy/releases

For instructions, take a look at this: https://github.com/ToniA/ESPEasy/blob/HeatpumpIR/src/_P115_HeatpumpIR.ino

brucysNZ, 2019-06-15:
@ToniA Thanks heaps for info. So flash .bin onto what device? D1 Mini or NodeMCU? Which is best device.
Thanks again

N-Storm, 2019-06-16:
> As I used Arduino micro I suspect that I didn't use IRSenderPWM. I think I e.g. would need to use [NodeMCU](https://www.nodemcu.com/index_cn.html) instead to use IRSenderPWM?

Nope, exactly the opposite. Code you've quoted does uses IRSenderPWM if ESP8266 is NOT defined. I.e. it's any AVR based Arduino, including micro. Weird you didn't had good timings with it. Probably it wasn't a timing issue but a LED power insted? How did you know that the timings were inaccurate? Did you measured them in some way? Please describe. Connection schematics would be useful too. I connect via 2n3904 to get 100 mA power to IR LED.

@brucysNZ, both D1 Mini and NodeMCU are based on the ESP8266. They are exactly the same in terms of hardware. The only difference are the number of available pins. D1 Mini does exposes only some of them. So if you don't need more pins it should be fine.
But the ESPEasy uses older heatpumpir library which has only IRSenderBitBang class available for ESP8266. Which was pretty inaccurate and you won't get a good distance from it. You can try it, but for longer distance it should be updated and recompilled with new library. I've added pair of new IRSender classes in https://github.com/ToniA/arduino-heatpumpir/issues/32 which improves carrier frequency accuracy.