ToniA/arduino-heatpumpir #50: Checksum for SHARP Air Conditioner

HoangDuong, 2018-01-26:
Hi, i am Hoang

I google sharp AC infrared protocol and found this repository.
this is the data i get from my Sharp remote:

`85 90 243 8 211 140 74 0 80 1 32 7 134 
85 90 243 8 222 140 74 0 80 1 32 7 139 
85 90 243 8 51 140 74 0 80 1 32 7 136 
85 90 243 8 62 140 74 0 80 1 32 7 133 
85 90 243 8 179 140 74 0 80 1 32 7 128 `

i used checksum method of Sharp like SharpHeatpumpIR.cpp but can not calculate the checksum byte.

`// Calculate the checksum
  for (int i=0; i<12; i++) {
    checksum ^= SharpTemplate[i];
  }

  checksum ^= SharpTemplate[12] & 0x0F;
  checksum ^= (checksum >> 4);
  checksum &= 0x0F;

  SharpTemplate[12] |= (checksum << 4);`

how can i calculate the checksum byte?? 

(sorry for my bad english)

ToniA, 2018-01-26:
Take a look at the IR decoder I have written: https://github.com/ToniA/Raw-IR-decoder-for-Arduino

Check out if it recognizes the signal from your remote.

closed 2018-01-26 by ToniA

HoangDuong, 2018-02-01:
Hi ToniA, 
I use STM8 MCU instead Arduino, but i use algorithm of IR lib of arduino. I use CRC method of your code with Toshiba and Panasonic AC and was success.
 
I do not know, i write post hear is wrong or righ, this is the first time i post in gitbub.

i checked the first post and it look stupid frame code, so i was edited and repost here. I really need calculate the last byte CRC. This is my Ir frame in hex value for temperature from 16.5 - 17 - 17.5 and 18.

0x55 0x5A 0xF3 0x08 0x8E 0x8C 0x42  0x00 0xB0 0x20 0x07 0x88

0x55 0x5A 0xF3 0x08 0x43 0x8C 0x42 0x00 0xB0 0x20 0x07 0x89 

0x55 0x5A 0xF3 0x08 0x4E 0x8C 0x42 0x00 0xB0 0x20 0x07 0x84

0x55 0x5A 0xF3 0x08 0xC3 0x8C 0x42 0x00 0xB0 0x20 0x07 0x81

for my observe:
Byte 1 to 4 alway the same.
Byte 5 contain data for temperature 
byte 6: ON/OFF or continue mode
Byte 7: for fan speed
Byte 8: alway 0x00
Byte 9: for swing value 
Byte 10: alway 0x01
Byte 11: unclear but i guess for confirm ON/OFF and continue 
Byte 12:  alway 0x07
Byte 13: CRC

Do you have any suggestion?.