ToniA/arduino-heatpumpir #52: Support for ESP32

mgknauth, 2018-06-27:
Is it possible to get support for ESP32 modules? Attempts to compile the examples fail with multiple errors (below). Using this library on an ESP32 allows for other libraries and modules to be integrated on the same MCU (i.e. support for more IO and resources).

Even though the following line was used `IRSenderBitBang irSender(9);`, there are still clear references to PWM and Arduino-specific (AVR MCU) registers.

`Arduino: 1.8.5 (Linux), Board: "WEMOS LOLIN32, 80MHz, Default, 921600"`

`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp: In member function 'virtual void IRSenderPWM::setFrequency(int)':`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:60:7: error: 'TCCR2A' was not declared in this scope`
`       TCCR2A = _BV(WGM20);`
`       ^`
`In file included from ./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:1:0:`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:60:20: error: 'WGM20' was not declared in this scope`
`       TCCR2A = _BV(WGM20);`
`                    ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:61:7: error: 'TCCR2B' was not declared in this scope`
`       TCCR2B = _BV(WGM22) | _BV(CS20);`
`       ^`
`In file included from ./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:1:0:`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:61:20: error: 'WGM22' was not declared in this scope`
`       TCCR2B = _BV(WGM22) | _BV(CS20);`
`                    ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:61:33: error: 'CS20' was not declared in this scope`
`       TCCR2B = _BV(WGM22) | _BV(CS20);`
`                                 ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:62:7: error: 'OCR2A' was not declared in this scope`
`       OCR2A = pwmval8;`
`       ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:63:7: error: 'OCR2B' was not declared in this scope`
`       OCR2B = pwmval8 / 3;`
`       ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:68:7: error: 'TCCR1A' was not declared in this scope`
`       TCCR1A = _BV(WGM11);`
`       ^`
`In file included from ./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:1:0:`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:68:20: error: 'WGM11' was not declared in this scope`
`       TCCR1A = _BV(WGM11);`
`                    ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:69:7: error: 'TCCR1B' was not declared in this scope`
`       TCCR1B = _BV(WGM13) | _BV(CS10);`
`       ^`
`In file included from ./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:1:0:`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:69:20: error: 'WGM13' was not declared in this scope`
`       TCCR1B = _BV(WGM13) | _BV(CS10);`
`                    ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:69:33: error: 'CS10' was not declared in this scope`
`       TCCR1B = _BV(WGM13) | _BV(CS10);`
`                                 ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:70:7: error: 'ICR1' was not declared in this scope`
`       ICR1 = pwmval16;`
`       ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:71:7: error: 'OCR1A' was not declared in this scope`
`       OCR1A = pwmval16 / 3;`
[... 51 lines cut ...]
`In file included from ./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:1:0:`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:159:24: error: 'COM1A1' was not declared in this scope`
`       (TCCR1A &= ~(_BV(COM1A1))); // Disable pin 9 PWM output`
`                        ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:162:24: error: 'COM1B1' was not declared in this scope`
`       (TCCR1A &= ~(_BV(COM1B1))); // Disable pin 10 PWM output`
`                        ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`./Arduino/libraries/arduino-heatpumpir-master/IRSenderPWM.cpp:165:24: error: 'COM2A1' was not declared in this scope`
`       (TCCR2A &= ~(_BV(COM2A1))); // Disable pin 11 PWM output`
`                        ^`
`./Arduino/hardware/espressif/esp32/cores/esp32/Arduino.h:106:25: note: in definition of macro '_BV'`
` #define _BV(b) (1UL << (b))`
`                         ^`
`exit status 1`
`Error compiling for board WEMOS LOLIN32.`

`This report would have more information with`
`"Show verbose output during compilation"`
`option enabled in File -> Preferences.`

ToniA, 2018-06-27:
ESP32 support is probably fairly simple. Right now I have conditional compilation which excludes certain things on ESP8266, just add the ESP32 flag there (whatever it happens to be called), and try again:

`
// ESP8266 does not have the Arduino PWM control registers
#ifndef ESP8266 
`

aenertia, 2019-06-03:
The ESP32 has a built in Frequency wave generator. 

There are some examples of using it here : https://github.com/kimeckert/ESP32-RMT-Rx-raw

and 

Here : https://github.com/kimeckert/ESP32-RMT-server

Likely this will need a new define and a datastructure to translate between it's interface and what HeatpumpIR spits out.

maxx-ukoo, 2019-08-06:
I guess this issue can be closed

closed 2019-08-06 by ToniA

szebenyib, 2020-02-24:
I was on version .3, upgrading to .12 solved this for me. Thank you for your work. The library is awesome :)