ToniA/arduino-heatpumpir #59: Mitsubishi MSZ DM-25

fhenryco, 2018-08-05:
How to check that Mitsubishi MSZ DM-25 is supported . Given that you report Mitsubishi MSZ FD-25 to be supported may be there is a chance but ... how to know ?
thanks

ToniA, 2018-08-05:
Try if it responds as MitsubishiFD or MitsubishiFE model. Take a look at https://github.com/ToniA/arduino-heatpumpir/blob/master/examples/simple/simple.ino, and remove everything else but MitsubishiFDHeatpumpIR from the heatpumpIR array (starting from line 35).

fhenryco, 2018-08-05:
OK thank you, but actually i don't have the heatpump yet. My seller quote is for a DM-25 ... may be i should ask for a  FD-25 to be sure i can control it in domoticZ ...

ToniA, 2018-08-05:
I really don't know... It might very well work, or it might not. I have very limited experience with Mitsubishi units, all of the code and IR samples is contributed.

fhenryco, 2018-08-06:
I'm buying the DM. If the control does not work (buyer told me the telecommand changes often ) i'll be interested in the retro engineering method used to decode the IR transmission protocol ... and for that may be i need to contact the people who did the job for the Mitsubishi FD ... (?)

rkgladney, 2023-03-13:
Is there any follow-up on whether Mitsubishi MSZ units are supported by this code?  I am interested in developing a controller that will automatically shut down the AC after approximately 9 PM and then turn it back on every morning.  The Mitsubishi remote requires you to manually enable on/off control every 24 hours or it defaults back to the last mode (on or off) based on the mode the AC was in when the 24 hour period ends.  I understand that I have to send a long string of commands, but it looks like the HeatpumpIR library will handle the string, I just have to add the ability to enter my desired operating temperature .  Also if there is a simpler method to automatic daily on/off operation please point me in that direction. Thanks in advance for your feedback and this library that you have provided.