ToniA/arduino-heatpumpir #61: Panasoic DKE (A75C2616) e-ion mode

ripkrizbi, 2018-08-26:
Hi, the remote A75C2616 i have has an extra button to switch on/off the e-ion mode of my HVAC. By looking into the raw dump of the protocol, it seems the byte 22 is responsible for controlling it, ie the value of 0x00 means e-ion OFF while the value of 0x01 means e-ion ON.
Would appreciate a kind soul implementing it into the already excellent library!
Cheers

ToniA, 2018-08-27:
I pushed a new branch called panasonic_ion. Let me know if it works the way you want, and I'll merge it into master. It's also time to tag a new release, I think.

ripkrizbi, 2019-01-27:
Hi, thanks a million and sorry for the delay in reporting. It took some time before I was able to get back to this one...

The branch works perfectly with one fix:

https://github.com/ToniA/arduino-heatpumpir/blob/5814f293d7f797bba9950dd981d05162e8a9953f/PanasonicHeatpumpIR.cpp#L251

This should reference byte 22 instead of byte 21 as is now. When I change it, everything else works!

Thanks again for the library and all of the great work you put into it!