ToniA/arduino-heatpumpir #63: Telefunken airconditioner - Remote control GZ01-BEJ0-000

grebar, 2018-09-16:
Hi, I'm trying to get the library working with my Air Conditioner.
Exploring the heatpumpir code I notice that the Fuego Heat Pump has a remote control similar to my HVAC

> https://github.com/ToniA/arduino-heatpumpir/blob/master/FuegoHeatpumpIR.h#L13
> 
> Fuego etc. (Vivax, Classe, NEO, Galanz, Simbio, Beko, Sintech) heatpump control (remote control P/N GZ-1002B-E3)
> 

My HVAC remote control has the code GZ01-BEJ0-000 (on some internet store they sell a remote compatible with both codes so I suppose there is some reasons to be compatible with Fuego. e.g. https://www.amazon.co.uk/Secure-Vending-Sweden-Westinghouse-GZ01-BEJ0-000/dp/B01JGA54LO)

I tried the library on my sonoff device where I installed the ToniA / ESPEasy fork (https://github.com/ToniA/ESPEasy) that include the heatpumpir library but unfortunately the FuegoHeatpumpIR does not work for me.

I started to investigate on my remote and I try to decode the raw data of my IR, I haven't an Arduino board so I used my Sonoff device with a modified tasmota firmware to debug the IR codes. 
The Tasmota firmware uses the lib IRremoteESP8266 to receive/send ir commands so I just made a little modification in the Tasmota firmware to print out the rawbuf of IRrecv::decode method.
Simply I used the sprintf to format the rawbuf in a readable way:
`sprintf(outputPointer, "0x%04hX,",rawbuf[cnt]);`

```
Here below the output that I logged.
         1      2      3      4      5      6      7      8      9      10     11     12     13     14     15     16     17     18     19     20     21     22     23     24     25     26     27     28     29     30     31     32     33     34     35     36     37     38     39     40     41     42     43     44     45     46     47     48     49     50     51     52     53     54     55     56     57     58     59     60     61     62     63     64     65     66     67     68     69     70     71     72     73    
15:38:34 0x0001,0x0788,0x02BD,0x0116,0x025F,0x0116,0x025F,0x0115,0x00FE,0x0109,0x00FE,0x0109,0x00FF,0x0107,0x0260,0x0115,0x00FE,0x0109,0x00FE,0x0109,0x0260,0x0115,0x025F,0x0115,0x00FE,0x0109,0x025F,0x0116,0x00FD,0x0109,0x00FE,0x0109,0x0260,0x0115,0x025F,0x0117,0x00FD,0x0109,0x025F,0x0116,0x025F,0x0116,0x00FD,0x0109,0x00FE,0x0109,0x025F,0x0115,0x00FE,0x0109,0x00FE,0x0109,0x025F,0x0116,0x00FD,0x0109,0x00F3,0x0114,0x00FE,0x0109,0x00FE,0x0109,0x00FD,0x0109,0x00FE,0x0109,0x00FE,0x0109,0x00FD,0x010B,0x00FC,0x0109,0x00FE,
15:38:39 0x0001,0x077E,0x02C7,0x010B,0x026A,0x010A,0x026B,0x010A,0x00FD,0x010A,0x00FD,0x010A,0x00FD,0x0109,0x026B,0x0116,0x00F1,0x010A,0x00FE,0x010B,0x0269,0x010A,0x026B,0x010A,0x00FD,0x010A,0x026A,0x010B,0x00FC,0x010A,0x00FE,0x0108,0x026B,0x010B,0x026A,0x010A,0x00FC,0x010A,0x026B,0x010A,0x026B,0x0109,0x00FD,0x010A,0x00FD,0x010A,0x026A,0x010A,0x00FD,0x0109,0x00FD,0x010A,0x026B,0x0109,0x00FE,0x010A,0x00FC,0x010A,0x00FC,0x0116,0x00F1,0x0116,0x00F1,0x010A,0x00FD,0x0115,0x00F1,0x010B,0x00FC,0x010A,0x00FD,0x0109,0x00FD,
15:38:46 0x0001,0x0789,0x02BC,0x0115,0x0260,0x0108,0x026C,0x0109,0x00FE,0x0114,0x00F3,0x0117,0x00F0,0x0109,0x026B,0x010C,0x00FD,0x0113,0x00F2,0x0108,0x026C,0x0109,0x026B,0x0115,0x00F3,0x0115,0x0260,0x0114,0x00F2,0x0109,0x00FE,0x0109,0x026B,0x0109,0x026C,0x0114,0x00F3,0x0109,0x026C,0x0108,0x026C,0x0114,0x00F2,0x0115,0x00F2,0x0109,0x026B,0x0115,0x00F2,0x010A,0x00FD,0x0115,0x0260,0x0109,0x00FE,0x0115,0x00F2,0x0109,0x00FE,0x0115,0x00F1,0x0109,0x00FD,0x0115,0x00F2,0x0115,0x00F1,0x0115,0x00F1,0x0115,0x00F2,0x0109,0x00FE,
15:38:48 0x0001,0x078C,0x02BB,0x010A,0x026A,0x010A,0x026A,0x0109,0x010C,0x00FB,0x010B,0x00FC,0x0101,0x0106,0x026D,0x0106,0x010C,0x00FB,0x010C,0x00FB,0x026C,0x0109,0x026C,0x0108,0x010C,0x00FB,0x026C,0x0109,0x010B,0x00FB,0x010B,0x00FB,0x026C,0x0109,0x026B,0x0115,0x0100,0x0106,0x0262,0x0112,0x0263,0x0113,0x0100,0x0107,0x0100,0x0107,0x0261,0x0114,0x0100,0x0106,0x0100,0x0107,0x0260,0x0115,0x00FF,0x0107,0x0100,0x0111,0x00F5,0x0108,0x00FF,0x0107,0x0100,0x0106,0x0100,0x0107,0x0100,0x0107,0x00FF,0x0107,0x0100,0x0107,0x0100,
15:38:50 0x0001,0x078B,0x02BB,0x0116,0x025F,0x010A,0x026C,0x0108,0x00FD,0x0114,0x00F2,0x0115,0x00F2,0x0115,0x025F,0x010A,0x00FD,0x0109,0x00FD,0x010A,0x026A,0x010C,0x0269,0x010A,0x00FD,0x010A,0x026B,0x010A,0x00FD,0x010A,0x00FD,0x010A,0x026B,0x010A,0x026B,0x010A,0x00FD,0x0109,0x026B,0x010B,0x026A,0x010A,0x00FD,0x010A,0x00FD,0x0109,0x026B,0x0109,0x00FE,0x010A,0x00FC,0x0109,0x026C,0x0109,0x00FE,0x0109,0x00FE,0x0109,0x00FE,0x0109,0x00FD,0x0109,0x00FF,0x0108,0x00FE,0x0109,0x00FD,0x0109,0x00FE,0x0109,0x00FE,0x0109,0x00FD,
15:38:52 0x0001,0x078D,0x02B9,0x010A,0x026B,0x0109,0x026B,0x0109,0x010B,0x0107,0x0102,0x0104,0x0100,0x0106,0x0260,0x0116,0x00FF,0x0107,0x00FF,0x0108,0x025F,0x0116,0x025F,0x0120,0x00F3,0x0108,0x0260,0x0116,0x00FE,0x0108,0x00FE,0x0109,0x025F,0x0115,0x0260,0x0116,0x00FE,0x0109,0x025F,0x0116,0x025F,0x0115,0x00FE,0x0109,0x00FE,0x0114,0x0254,0x0115,0x00FE,0x0108,0x00FF,0x0109,0x025E,0x0116,0x00FE,0x0109,0x00FE,0x0108,0x00FF,0x0109,0x00FE,0x010A,0x00FD,0x0108,0x00FE,0x0109,0x00FE,0x0108,0x00FF,0x0108,0x00FF,0x0108,0x00FE,
15:38:54 0x0001,0x077D,0x02C9,0x0115,0x025F,0x0116,0x025F,0x0115,0x00FF,0x0108,0x00FF,0x0107,0x00FF,0x0108,0x025F,0x0115,0x00FF,0x0108,0x00FF,0x0108,0x025F,0x0115,0x025F,0x0116,0x00FE,0x0108,0x025F,0x0115,0x00FE,0x0108,0x00FF,0x0115,0x0252,0x0115,0x025F,0x0116,0x00FE,0x0109,0x025F,0x0115,0x025F,0x0115,0x00FE,0x0108,0x00FF,0x0108,0x025F,0x0117,0x00FD,0x0109,0x00FE,0x0114,0x0253,0x0115,0x00FF,0x0108,0x00FE,0x0108,0x00FE,0x0108,0x00FE,0x0108,0x00FF,0x0108,0x00FF,0x010A,0x00FC,0x0108,0x0100,0x0113,0x00F2,0x0108,0x00FE,
15:38:57 0x0001,0x077D,0x02C9,0x0115,0x0260,0x0115,0x0260,0x0109,0x010B,0x0108,0x00FF,0x0108,0x00FF,0x0108,0x025F,0x0116,0x00FF,0x0108,0x00FF,0x0108,0x025F,0x0115,0x0260,0x0108,0x010B,0x00FE,0x0269,0x0114,0x0100,0x0108,0x00FF,0x0108,0x025F,0x0115,0x025F,0x0116,0x00FF,0x0108,0x0260,0x0121,0x0253,0x0116,0x00FF,0x0107,0x00FF,0x0108,0x0260,0x0115,0x00FF,0x0108,0x00FF,0x0109,0x0260,0x0115,0x00FF,0x0108,0x00FF,0x0108,0x00FF,0x0108,0x00FF,0x0108,0x00FF,0x0108,0x00FE,0x0108,0x00FF,0x0108,0x00FE,0x0108,0x00FF,0x0108,0x00FE,

15:38:34 IRR: RawLen 228, Bits 114, Value 3FFF64D4, Decode -39821610
15:38:39 IRR: RawLen 228, Bits 114, Value 3FFF64D4, Decode -1674830522
15:38:46 IRR: RawLen 228, Bits 114, Value 3FFF64D4, Decode 2085124178
15:38:48 IRR: RawLen 228, Bits 114, Value 3FFF64D4, Decode -541808092
15:38:50 IRR: RawLen 228, Bits 114, Value 3FFF64D4, Decode 1212834680
15:38:52 IRR: RawLen 228, Bits 114, Value 3FFF64D4, Decode 2038024676
15:38:54 IRR: RawLen 228, Bits 114, Value 3FFF64D4, Decode -743135188
15:38:57 IRR: RawLen 228, Bits 114, Value 3FFF64D4, Decode -401392496
```

I am not an expert of IR protocols, can someone help me understand the codes above?

ToniA, 2018-09-17:
Do you have any way to run my decoder sketch on an Arduino? I have a decoder for the Fuego protocol, perhaps this is really close: https://github.com/ToniA/Raw-IR-decoder-for-Arduino/blob/master/Fuego.cpp

This IRRemote output is really hard to read...

grebar, 2018-09-18:
Thanks to my friend I executed the decoder sketch on Arduino.
Please find attached the results.
[ir_telefunken_decoded.txt](https://github.com/ToniA/arduino-heatpumpir/files/2393167/ir_telefunken_decoded.txt)

tolma488, 2018-09-18:
Cancel

ToniA, 2018-09-18:
OK, this protocol looks very much the same as the 'Fuego' protocol. The number of bits matches, the checksum calculation matches, and on a quick glance the most important functions seemed to match as well.

Maybe there's some 'unit code' or something, so could you compare the Fuego implementation against your samples? The code template is on this code line: https://github.com/ToniA/arduino-heatpumpir/blob/master/FuegoHeatpumpIR.cpp#L101, the last byte is the checksum (which is already OK), so it must be something about the rest of the bytes.

Another approach is to send the Fuego code from ESPEasy, and record it on this recorder sketch, and then compare against the real remote, to see if there are any obvious differences.

grebar, 2018-09-19:
I have a simple question: if I take the decoded array from the Arduino decoder (e.g. POWER ON 23,CB,26,01,00,24,07,07,20,00,00,01,04,6C) and I replace it in the Fuego implementation https://github.com/ToniA/arduino-heatpumpir/blob/master/FuegoHeatpumpIR.cpp#L101
then I jump directly to the code below the comment "// 38 kHz PWM frequency". 
I suppose the HVAC must power on, isn't true?

ToniA, 2018-09-19:
Yes, you should be able to play back recorded codes that way.

I actually have an example about sending raw codes (with Panasonic timing), take a look at this: https://github.com/ToniA/arduino-heatpumpir/blob/master/examples/PanasonicRelayControl/PanasonicRelayControl.ino

grebar, 2018-09-20:
Thank you very much @ToniA for your help!
I just tried your raw sending code sample adapting it for my Telefunken, and magically the HVAC turns ON.

Then I tried to modify the array with presets values, changing some initialization codes
https://github.com/ToniA/arduino-heatpumpir/blob/master/FuegoHeatpumpIR.cpp#L101
`
  uint8_t FuegoTemplate[] = { 0x23, 0xCB, 0x26, 0x01, 0x00, 0x20, 0x00, 0x00, 0x40, 0x00, 0x00, 0x01, 0x04, 0x00 };

//uint8_t FuegoTemplate[] = { 0x23, 0xCB, 0x26, 0x01, 0x80, 0x20, 0x00, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00 };
// c                            0     1     2     3     4     5     6     7     8     9     10    11    12    13
`
Basically I changed the bytes:
`
#4  from 0x80 to 0x00
#11 from 0x00 to 0x01
#12 from 0x00 to 0x04
`
The code starts to work, I tried POWER ON and POWER OFF. After that, I start to analyse the bytes from your decoder, I have a doubt on the byte #12 it appears to be the identifier of pressed button, but I am not sure and it is not clear if is mandatory or not.

`
0  1  2  3  4  5  6  7  8  9  10 11 12 13 - # byte
23,CB,26,01,00,24,07,07,20,00,00,01,04,6C - POWER ON
23,CB,26,01,00,20,07,07,20,00,00,01,04,68 - POWER OFF
23,CB,26,01,00,24,01,01,20,00,00,01,10,6C - TEMP -1 from 31 to 30
23,CB,26,01,00,24,01,00,20,00,00,01,14,6F - TEMP +1 from 30 to 31
23,CB,26,01,00,24,01,00,38,00,00,01,18,8B - Vertical air SWING
23,CB,26,01,00,24,01,00,28,00,00,01,4C,AF - Vertical air down press 1
23,CB,26,01,00,24,01,00,08,00,00,01,4C,8F - Vertical air down press 2 (became UP)
23,CB,26,01,00,24,01,00,10,00,00,01,4C,97 - Vertical air down press 3 (became MIDDLE UP)
23,CB,26,01,00,24,01,00,18,00,00,01,4C,9F - Vertical air down press 4 (became MIDDLE)
23,CB,26,01,00,24,01,00,20,00,00,01,4C,A7 - Vertical air down press 5 (became MIDDLE DOWN)
23,CB,26,01,00,24,01,00,28,00,00,01,4C,AF - Vertical air down press 5 (turn to DOWN )
23,CB,26,01,00,24,08,07,28,00,00,01,08,79 - Mode Press 1 (became AUTO)
23,CB,26,01,00,24,03,00,28,00,00,01,08,6D - Mode Press 2 (became COOL
23,CB,26,01,00,24,02,07,28,00,00,01,08,73 - Mode press 3 (became DRY)
23,CB,26,01,00,24,07,07,28,00,00,01,08,78 - Mode press 4 (became FAN)
23,CB,26,01,00,24,01,00,28,00,00,01,08,6B - Mode press 5 (became HEAT)
23,CB,26,01,00,24,03,00,2F,00,00,01,2C,98 - TURBO PRESS 1
23,CB,26,01,00,24,03,00,2A,00,00,01,2C,93 - TURBO PRESS 2
23,CB,26,01,00,24,03,00,2F,00,00,01,2C,98 - TURBO PRESS 3
23,CB,26,01,00,24,03,00,2A,00,00,01,2C,93 - TURBO PRESS 4
23,CB,26,01,00,24,03,00,28,00,00,01,0C,71 - FAN Speed press 1 (AUTO)
23,CB,26,01,00,24,03,00,2A,00,00,01,0C,73 - FAN SPEED Press 2 (1)
23,CB,26,01,00,24,03,00,2B,00,00,01,0C,74 - FAN SPEED press 3 (2)
23,CB,26,01,00,24,03,00,2D,00,00,01,0C,76 - FAN SPEED press 4 (3)
23,CB,26,01,00,24,03,00,28,00,00,01,0C,71 - FAN SPEED press 5 (turn back to AUTO)
23,CB,26,01,00,24,03,00,28,00,00,01,50,B5 - CANCEL PRESS
23,CB,26,01,00,24,83,00,28,00,00,01,28,0D - DISPLAY button PRESS 1
23,CB,26,01,00,24,83,00,28,00,00,01,28,0D - DISPLAY BUTTON Press 2
`
Thanks!

grebar, 2018-09-22:
Dear @ToniA, i would like to suggest to @arendst (the [Sonoff-Tasmota](https://github.com/arendst/Sonoff-Tasmota) creator) the integration of your library in its firmware. I already tested the integration in my environment and all appears to works properly. 

I have to deal with two issues:
1) both, your [library](https://github.com/ToniA/arduino-heatpumpir/blob/master/HeatpumpIR.h#L15) and a [library](https://github.com/arendst/Sonoff-Tasmota/blob/development/lib/Joba_Tsl2561-2.0.7/src/Tsl2561.h#L65) already used by [Sonoff-Tasmota](https://github.com/arendst/Sonoff-Tasmota), define the identifiers  POWER_ON and POWER_OFF. Without a rename of one of two definitions the sources does not compile. 
Have you any suggestion to resolve the conflict?
2) for my HVAC, as already sayed in previous posts, there is only a difference with your Fuego implementation (the values of the array FuegoTemplate). In a simple way, I can implement a new class extending the Fuego one. Is there other ways? Do you think it is possible to introduce in the base HeatPumpIR class a method to manage little variants? For example a Configuration method that in my case allows to set the remote control type to be used by the Fuego (GZ-1002B-E3 the default one or GZ01-BEJO-000 the mine).

Gregorio

Blend3rman, 2022-01-06:
I would like to thank you @grebar for your work on making sense of the decoded message, I was working with the same remote and this issue proved to be a great source of info to base my work on.
@ToniA, I would like to contribute to the [IRremoteESP8266 library](https://github.com/crankyoldgit/IRremoteESP8266), extending the `IRac` class to support sending of "Fuego" AC commands. Requesting permission to use your code in the [FuegoHeatpumpIR.cpp](https://github.com/ToniA/arduino-heatpumpir/blob/master/FuegoHeatpumpIR.cpp)/[FuegoHeatpumpIR.h](https://github.com/ToniA/arduino-heatpumpir/blob/master/FuegoHeatpumpIR.h) files as a basis for this. 
([Link to my discussion post in the IRremoteESP8266 repo](https://github.com/crankyoldgit/IRremoteESP8266/discussions/1728))

TIA.