ToniA/arduino-heatpumpir #65: Senville AURA HF/Z Remote

kpishere, 2018-10-21:
Great work! Id started to add support for this heat pump but well, my solution is using interrupts and would work better on ESP8266. It is soo different too and the CRC still isn't right so frankly it isnt ready for this project.  If your looking for this unit too check https://github.com/kpishere/GetRawIRData

kpishere, 2018-11-22:
I did manage to figure out the CRC calculation for this heat pump.  Feel free to incorporate etc.  It is simple but also at the same time very specific.

trev347, 2018-11-29:
I have a senville heat pump do a way to put this in espeasy

kpishere, 2018-11-29:
Ah, didn't know about ESPEasy. Thanks. Code is there as a working reference so feel free.  It is not where I'm going with this however.