ToniA/arduino-heatpumpir #66: Powerful and quiet mode seems have no action on the ESPeasy fork

advfr, 2018-11-18:
Hello @ToniA 
Big thanks for that great project. I was able to "burn" your ESPeasy fork on an ESP01 (ESP8266 with 1M rom), as I've never been able to compile ESPeasy with your project, I had each time error durring the compilation (I'm a beginner).

I owned a TKE model, and get good result with DKE commands but in your ESPeasy firmware the byte 21 didn't change when I try to swicth powerful or quiet mode by a web command like this:

http://192.168.1.xxx/control?cmd=heatpumpir,panasonic_dke,1,2,0,24,2,0,0,1
or
http://192.168.1.xxx/control?cmd=heatpumpir,panasonic_dke,1,2,0,24,2,0,1,0

My test plateform is ESP01 with your fork in front of your Raw-IR-decoder-for-Arduino project.

Is there a bug or something I'm doing wrong ?
Thank you for your help.

ToniA, 2018-11-19:
I only have an old Panasonic DKE unit myself, and on those ones the quiet/powerful setting is a separate message, and you can only toggle the state (i.e. for example the 'quiet' message is always the same, and it just switches between quiet and non-quiet).

advfr, 2018-11-19:
OK Tony may be I was unclear. The intention of the message was to let you know that the web command doesn't take car of the 8th and 9th argument (0,0 0,1 or 1,0, cannot be 11), which seems (after I took a look at the code) normal... as I see only 7 parameters are read:

`if (tmpString.equalsIgnoreCase("HEATPUMPIR") && Plugin_115_irSender != NULL)
        {
          if (GetArgv(command, TmpStr1, 2)) heatpumpModel = TmpStr1;
          if (GetArgv(command, TmpStr1, 3)) powerMode = str2int(TmpStr1);
          if (GetArgv(command, TmpStr1, 4)) operatingMode = str2int(TmpStr1);
          if (GetArgv(command, TmpStr1, 5)) fanSpeed = str2int(TmpStr1);
          if (GetArgv(command, TmpStr1, 6)) temperature = str2int(TmpStr1);
          if (GetArgv(command, TmpStr1, 7)) vDir = str2int(TmpStr1);
          if (GetArgv(command, TmpStr1, 8)) hDir = str2int(TmpStr1);
`

is there a way to modify the code to take care about the 8th and 9th arg, and "link" them to power modes ?

Thank you :)
Antoine from France.

ToniA, 2018-11-19:
Yes, I wanted to make a simple interface which is the same for every model... But yes, it would not be that difficult to add some extra code into https://github.com/ToniA/ESPEasy/blob/HeatpumpIR/src/_P115_HeatpumpIR.ino#L213, to check if the model is a Panasonic model, create a new Panasonic-specific instance of the HeatpumpIR class, and call that instead of the generic 'send' method.