ToniA/arduino-heatpumpir #67: Compile error with ESP8266 (ESP-01)

gazialankus, 2018-12-22:
I'm trying to make the DaiseikaiTest.ino sketch on a ESP-01 flavor of ESP8266 with no luck. I made it work on an Arduino Nano earlier, but after switching the board to "Generic ESP8266 Module" it won't compile anymore. 

I first installed HeatpumpIR via the Library Manager, which was v1.0.8. Later I moved to the version in GitHub. I had the error below in both cases: 

Here's the error. Why might this be happening?

```
In file included from C:\Users\gazihan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,

                 from C:\Users\gazihan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,

                 from C:\Users\gazihan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Arduino.h:32,

                 from C:\Users\gazihan\Documents\Arduino\libraries\HeatpumpIR/HeatpumpIR.h:10,

                 from C:\Users\gazihan\Documents\Arduino\libraries\HeatpumpIR/AIRWAYHeatpumpIR.h:8,

                 from C:\Users\gazihan\Documents\Arduino\libraries\HeatpumpIR\AIRWAYHeatpumpIR.cpp:1:

C:\Users\gazihan\Documents\Arduino\libraries\HeatpumpIR\AIRWAYHeatpumpIR.cpp: In constructor 'AIRWAYHeatpumpIR::AIRWAYHeatpumpIR()':

C:\Users\gazihan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: section of 'model' conflicts with previous declaration

   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))

                                                                                                                                  ^

C:\Users\gazihan\Documents\Arduino\libraries\HeatpumpIR\AIRWAYHeatpumpIR.cpp:6:37: note: in expansion of macro 'PROGMEM'

   static const char PROGMEM model[] PROGMEM = "AIRWAY";

                                     ^

C:\Users\gazihan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0-beta2/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:25:130: error: section of 'info' conflicts with previous declaration

   #define PROGMEM __attribute__((section( "\".irom.text." __FILE__ "." __STRINGIZE(__LINE__) "."  __STRINGIZE(__COUNTER__) "\"")))

                                                                                                                                  ^

C:\Users\gazihan\Documents\Arduino\libraries\HeatpumpIR\AIRWAYHeatpumpIR.cpp:7:37: note: in expansion of macro 'PROGMEM'

   static const char PROGMEM info[]  PROGMEM = "{\"mdl\":\"AIRWAY\",\"dn\":\"AIRWAY\",\"mT\":18,\"xT\":31,\"fs\":3}";

                                     ^

Using library HeatpumpIR at version 1.0.8 in folder: C:\Users\gazihan\Documents\Arduino\libraries\HeatpumpIR 
Using library IRremoteESP8266 at version 2.5.3 in folder: C:\Users\gazihan\Documents\Arduino\libraries\IRremoteESP8266 
exit status 1
Error compiling for board Generic ESP8266 Module.

```

gazialankus, 2018-12-22:
And here is the beginning of the sketch: 

```
#include <Arduino.h>
#include <ToshibaDaiseikaiHeatpumpIR.h>

// worked fine with arduino nano
//IRSenderPWM irSender(9);     // IR led on Duemilanove digital pin 3, using Arduino PWM
IRSenderBitBang irSender(0);

ToshibaDaiseikaiHeatpumpIR *heatpumpIR;
```

gazialankus, 2018-12-22:
I just replaced `static const char PROGMEM` with `static const char` throughout the library and I was able to upload it fine. It also ran fine, turned on and off my A/C unit. 

Looking at the [docs here](https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html) made me think one PROGMEM after the variable should be enough.

I wonder why nobody else ran into this. Could it be the Arduino IDE ESP8266 board library version?

gazialankus, 2018-12-22:
Yeah they are a pain to work with but I like the tiny form factor:)

Is there a reason for having two `PROGMEM`s, though?

```
   static const char PROGMEM model[] PROGMEM = "AIRWAY";
```

I haven't seen anything like that in docs here: https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html

If modifying them to look like this below does not break anything, maybe it's worth changing them for the sake of ESP-01 compatibility. I can confirm that it works this way in ESP-01.

```
   static const char model[] PROGMEM = "AIRWAY";
```

Thank you so much for this very useful library, btw.

N-Storm, 2019-03-28:
It doesn't related to ESP-01 model, in fact I could confirm this issue related to 2.5.0 version of ESP8266 toolchain. With 2.4.2 it works fine, but should be fixed for 2.5.0 (including betas). Will look into this further tomorrow.

N-Storm, 2019-04-02:
Fixed in https://github.com/ToniA/arduino-heatpumpir/pull/73

N-Storm, 2019-06-17:
@ToniA could confirm this was fixed & working in hardware. You may close this issue. :)

closed 2019-06-17 by ToniA