ToniA/arduino-heatpumpir #71: AUXHeatPumpIR (YKR-P/002E) Power On/Off Command issue

nanoGigaTech, 2019-02-25:
Hi,
I was searching for IR protocol for my A/C with remote controller no. YKR-P/002E and got your library. I tried to verify your code using analysIR. Except the following minor bugs everything works just fine:
1. Byte[9] which is used for Power On/Off is 0x30 for Power On, but your library says it's 0x20.
2. Byte[11] is set as 0x08 in your code but my decoder says it's 0x05. 

Though I haven't tested the code directly on my A/C as my A/C Unit is in maintenance stage. Can anyone confirm if it's ToniA's library or my decoder which is wrong?

kind regards,

ToniA, 2019-02-25:
The library might very well have errors. This protocol was contributed, so I haven't had any chance to check it out myself. Pull requests are welcome :)

whatsupskip, 2019-02-25:
Was the protocol for the exact same model of air conditioner, not just the same brand?

Different models in the same brand frequently have different protocols.

N-Storm, 2019-03-29:
I've tested this with Hyundai H-AR21-07H (remote control P/N YKR-P/002E) and it worked for power on and mode setting.
As I don't have direct access to this A/C unit now I can't test it in nearby time, but try some time later. But it might take months.

N-Storm, 2019-04-16:
I've re-tested current library from trunk and it works fine with the A/C unit I've mentioned above.

abdulmannan, 2019-05-21:
Byte[9] should have 0x20 for Power ON/OFF which is correct in the library.
Byte[11] is actually key code which will different for each key. Library is setting it to 0x08 which is key code for up/down swing button. However setting it to 0 must suffice.

I have tested it on actual AUX unit using different remotes (YKR-H/009E, YKR-L/102E).

rasyid7, 2019-06-29:
I tested it on my AUX AC. It could powered ON but not OFF. When I debugged the code, OFF command sent as mode=Heat, trying to changed the mode but it doesn't work. My AC unit doesn't support Heat so that's why it doesn't work. 
The issue is, the mode always says heat when OFF is sent.