ToniA/arduino-heatpumpir #76: NOT an ISSUE! Mitsubishi MSZ-FD25VA List of IR code available?

lumanga, 2019-06-17:
Hi All,
I have **Mitsubishi MSZ-FD25VA**.

To capture IR code sent from original remote (code KM09D 0314438) I'm using this windows software  https://sourceforge.net/projects/broadlink-manager/ with BROADLINK MINI 3.

Capturing a lot of times, the IR code only for POWER ON, 24°C, COOL, fan auto etc.... and all the settings always the same, **I obtain always different codes!!!**

So Im'm asking you if **exist a library/file.JSON or list of IR code  for my Mitsubishi MSZ-FD25VA** ??.
(I'm afraid to don't send all the code needed... including PLASMA,CLEAN,AREA,WideVane... that my AC has and are not standard options for an AC!)

What happen if I don't send all the command?? 
If I don't send the special feature above (PLASMA, AREA etc.) what heppen???

Many thanks in advance!

N-Storm, 2019-06-17:
This is really not the right place to ask. You have to check forums dedicated to your product. 

The AC units are different in IR protocol to all other devices. You don't have "IR codes" for them, like you have "IR codes" for TV remote where every button have own code. AC units remote sends **all settings** (power, temperature, mode, etc) in single large packet.

Try checking this link: https://community.home-assistant.io/t/smartir-control-your-climate-tv-and-fan-devices-via-ir-rf-controllers/100798 but again, this library doesn't have anything to do with Broadlink at all.

ToniA, 2019-06-17:
This is the system I'm using to decode IR signals: https://github.com/ToniA/Raw-IR-decoder-for-Arduino

closed 2019-06-17 by ToniA