ToniA/arduino-heatpumpir #77: samsung windfree AC

skorc99, 2019-07-02:
Hello,

is plugin working with this type of AC?
I tryed to add plugin to espeasy but no success..

Is *.bin file with plugin available?

ToniA, 2019-07-05:
I have no idea if this particular model would work, but you could try it out. I have an ESPEasy binary with the plugin for ESP8266-based boards (like Wemos D1 Mini) here: https://github.com/ToniA/ESPEasy/releases

The library has support for some samsung models, see this: https://github.com/ToniA/arduino-heatpumpir/blob/master/SamsungHeatpumpIR.h

skorc99, 2019-09-11:
Hello,

i tested and it is working with windfree Ac -AR12 model. It is the same as "fjm".

Is it possible that you update firmware to the latest mega?

U

ToniA, 2019-09-11:
This is now integrated into the official ESP Easy releases: https://github.com/letscontrolit/ESPEasy/releases

HeatpumpIR is included into the 'minimal IRext' image.

whatsupskip, 2019-09-11:
Great to see.

skorc99, 2019-09-11:
Hello,

yes it is latest but no support for temp sensors in this FW.. 
I need for dht11 or dht22...

U

ToniA, 2019-09-12:
The problem was that with HeatpumpIR, the binary size was so large that OTA was not possible any more. The 'minimal IRext' image is a minimal image with most of the plugins removed.

I think we should make a change request to ESPEasy, to include some basic sensors into the 'minimal IRext'. DHT11 and DS18x20 at least.

skorc99, 2019-09-12:
I agree.   IRplugin without temperature is not ok.

At least DHT11 and DS18x20.

skorc99, 2019-09-15:
Hello,
with version **mega-20180809 with HeatpumpIR P115 w/ Domoticz MQTT**
Plugin si working correctly but after 1h uptime i get error-MQTT.

3836263: DHT  : Temperature: 24.00
3836263: DHT  : Humidity: 56.00
3836268: Read settings: ControllerSettings index: 0
3836287:  Domoticz: Sensortype: 2 idx: 259 values: 22.00;63.00;3
3836287: MQTT : {'idx':259,'RSSI':10,'nvalue':0,'svalue':'22.00;63.00;3'}
**3836289: MQTT : publish failed**
3836526: sendcontent free: 13936 chunk size:400
3836532: sendcontent free: 13264 chunk size:400
3836536: sendcontent free: 12808 chunk size:145
3836539: sendcontent free: 13480 chunk size:0
3837714: sendcontent free: 14152 chunk size:360
3837716: sendcontent free: 13480 chunk size:0
3841685: Memtrace^0: lowest: 10104  sendContentBlocking-&gt; 10800 sendContentBlocking-&gt; 11472 sendContentBlocking-&gt; 10800 sendContentBlock
3841685: 1: lowest: 10128  PluginCall_s (46)-&gt; 14912 parseTemplate-&gt; 14936 parseTemplate3-&gt; 14872 PluginCall_s (4)-&gt; 14912 PluginCall_s 
3841686: 2: lowest: 10128  PluginCall_s (4)-&gt; 14912 PluginCall_s (46)-&gt; 14912 handle_advanced-&gt; 12848 sendHeaderBlocking-&gt; 12832 sendWeb
3841686: WD   : Uptime 64 ConnectFailures 0 FreeMem 15048
3841807: sendcontent free: 13936 chunk size:400
3842579: PluginStats P_47_Heatpump IR transmitter TEN_PER_SECOND       Count: 283 Avg/min/max 4.04/2/27 usec
3842579: PluginStats P_47_Heatpump IR transmitter EVENT_OUT            Count: 1 Avg/min/max 31.00/31/31 usec
3842580: PluginStats P_47_Heatpump IR transmitter FIFTY_PER_SECOND     Count: 1405 Avg/min/max 10.11/2/35 usec
3842580: Load File stats: Count: 2 Avg/min/max 9918.00/2033/17803 usec
3842580: Plugin call 50 p/s   stats: Count: 1405 Avg/min/max 962.11/838/1430 usec
3842581: Plugin call 10 p/s   stats: Count: 283 Avg/min/max 875.33/838/1115 usec
3842581: Plugin call 10 p/s U stats: Count: 283 Avg/min/max 3343.12/2930/4036 usec
3842581: Plugin call  1 p/s   stats: Count: 30 Avg/min/max 1508.10/1373/1888 usec
3842582: SensorSendTask()     stats: Count: 1 Avg/min/max 54146.00/54146/54146 usec
3842582: sendData()           stats: Count: 1 Avg/min/max 24246.00/24246/24246 usec
3842582: Compute formula      stats: Count: 1 Avg/min/max 2625.00/2625/2625 usec
3842583: setNewTimerAt()      stats: Count: 2063 Avg/min/max 136.04/116/237 usec
3842583: timeDiff()           stats: Count: 1314892 - CPU cycles per call: 8.92
3842583: Scheduler stats: (called/tasks/max_length/idle%) 626419/1835/8/89.40

ToniA, 2019-09-17:
I made PR https://github.com/letscontrolit/ESPEasy/pull/2620 to ESPEasy, this would include basic environmental sensors and switch input into the minimal_IRext_ESP8266_1M image. This way I hope you could use the latest ESPEasy images, and I hope the stability would be better as well.

skorc99, 2019-09-21:
hello,

release date for this update?

u

ToniA, 2019-09-21:
It depends on the ESPEasy guys. But you could just clone my repo, checkout the master branch and compile on PlatformIO:

platformio run -e minimal_IRext_ESP8266_4M1M -t upload

skorc99, 2019-09-22:
I cloned compiled but i still get only your plugin - no temp plugin..

Plugins:⋄ | 6 [Normal] [Minimal, IR with AC]
-- | --
Build Time:⋄ | Sep 22 2019 07:55:39
**Binary Filename:⋄ | Self built!**

ToniA, 2019-09-22:
Are you sure you built from this: https://github.com/ToniA/ESPEasy/tree/master, commit 6414bda
?

I just double-checked by uploading first the normal image, and then the irext one, and I have 11 plugins, including plugin 5 'Environment - DHT11/12/22 SONOFF2301/7021':

platformio run -e normal_ESP8266_4M1M -t upload
platformio run -e minimal_IRext_ESP8266_4M1M -t upload

Plugin 34 'Environment - DHT12 (I2C)' is not included, would you need that one?

skorc99, 2019-09-22:
All ok now.

I will test now.

skorc99, 2019-09-23:
all ok.
AC working
Mqtt to domoticz working
dht22 working

so far no crashes..

ToniA, 2019-09-23:
Good to hear :) I'll close this issue now.

closed 2019-09-23 by ToniA