ToniA/arduino-heatpumpir #80: Zenith AC brand

kilbamoo, 2019-07-29:
I have a zenith ac unit. HeatpumpIR does not show any compatible devices. So I put the RAW IR decoder for arduino on my uno and connected a ir receiver.
I was able to capture something, however I dont know what exactly to do with it.
This is the serial output under option 4 (Hyundai). Option 5, Samsung, will output similar data. The output shown is when I press the power button 'on' from the remote. How do I create a decoder module?

`18:58:18.883 -> Enter choice: 4
18:58:27.753 -> 
18:58:27.753 -> Ready to decode IR for choice '4'
18:58:27.786 -> 
18:58:27.786 -> 
18:58:33.663 -> 
18:58:33.663 -> Number of symbols: 31
18:58:33.663 -> Symbols:
18:58:33.698 -> Hh1000100000000000001101000111
18:58:33.732 -> Bytes:
18:58:33.732 -> 00:  1000|1000 | 11 | 00010001 
18:58:33.767 -> 01:  0000|0000 | 00 | 00000000 
18:58:33.801 -> 02:  0011|0100 | 2C | 00101100 
18:58:33.837 -> 03:  0111|11,00,2C
18:58:33.837 -> Timings (in us): 
18:58:33.871 -> PAUSE SPACE:  0
18:58:33.871 -> HEADER MARK:  8580
18:58:33.904 -> HEADER SPACE: 4320
18:58:33.938 -> BIT MARK:     462
18:58:33.938 -> ZERO SPACE:   516
18:58:33.973 -> ONE SPACE:    1603
18:58:33.973 -> Decoding known protocols...
18:58:34.008 -> Unknown protocol
18:58:34.042 -> Bytecount: 3
`

ToniA, 2019-07-30:
The raw decoding looks fine. The next step is to try to understand the structure of the message, i.e. what is the meaning of each bit in the message. I just systematically go through one change at a time, and try to figure out what change corresponds to a change in the bit pattern.

There is also a possibility to just record raw samples and play them back: https://github.com/ToniA/arduino-heatpumpir/blob/master/examples/rawsender/rawsender.ino

N-Storm, 2019-08-02:
I don't think this a good decoding. Only 3 bytes. Seems like the timings was wrong or carrier issues.
@kilbamoo try a few more decoding to see if bytecount remains the same. Also try other modes of sketch operation.