ToniA/arduino-heatpumpir #84: Samsung ac - OFF command not working

niccodemi, 2019-08-18:
I've got Samsung a/c. I am using library via mysensors sketch. Everything works fine (power on, fan, modes and temp change) but I cannot power the ac off. Below is decoded info - there are differences in 2 bytes (C2-D2 and 1B-49). Also header space seems to be different.
I sent off command through myscontroller by typing 0090121B

Actual remote power OFF
Number of symbols: 177
Symbols:
Hh01000000010011011111000000000000000000000000000000000011hHh10000000010010111111000000000000000000000000000000000000hHh10000000010000110111111110001110000011011101100000000011
Bytes:
02,B2,0F,00,00,00,C0,01,D2,0F,00,00,00,00,01,C2,FE,71,B0,1B,C0
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  3100
HEADER SPACE: 6488
BIT MARK:     540
ZERO SPACE:   380
ONE SPACE:    1417
Decoding known protocols...
Looks like a Samsung protocol
POWER OFF

MYSENSORS command 0090121B
Number of symbols: 177
Symbols:
Hh010000000100110111110000000000000000000000000000000000111Hh100000000100101111110000000000000000000000000000000000001Hh10000000010010110111111110001110000011011001001000000011
Bytes:
02,B2,0F,00,00,00,C0,01,D2,0F,00,00,00,00,01,D2,FE,71,B0,49,C0
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  3200
HEADER SPACE: 9286
BIT MARK:     478
ZERO SPACE:   413
ONE SPACE:    1478
Decoding known protocols...
Looks like a Samsung protocol
POWER OFF

ToniA, 2019-08-19:
Do you have a cooling only (no heating) device? It could be that it does not understand 'auto - power off' combination, but only 'cool - power off'. In '0090121B', the fifth number is '1', i.e. auto mode. Try send '0090321B', i.e. 'cool - power off' instead.

niccodemi, 2019-08-19:
yes, device is cooling only. I tried with '0090321B' but the result is the same - aircon doesn't react.

ToniA, 2019-08-20:
How about sending '0090331B', this should be cool + fan3. Can you record a sample?

If it is about the message data, I think it's the second last byte (0x1B = cool + fan3, 0x49 = heat + fan2). The byte in position 15 is the checksum, so this obviously also changes if the message changes.

The other possibility is the header space length, but if it works on all other commands, it would seem weird if the timings would be different for power off message.

niccodemi, 2019-08-21:
Hi,

this are decoded signals:

sent 0090321B

Number of symbols: 177
Symbols:
Hh010000000100110111110000000000000000000000000000000000111Hh100000000100101111110000000000000000000000000000000000001Hh10000000010010110111111110001110000011011001001000000011
Bytes:
02,B2,0F,00,00,00,C0,01,D2,0F,00,00,00,00,01,D2,FE,71,B0,49,C0
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  3200
HEADER SPACE: 9273
BIT MARK:     505
ZERO SPACE:   400
ONE SPACE:    1468
Decoding known protocols...
Looks like a Samsung protocol
POWER OFF

sent 0090331B

Number of symbols: 177
Symbols:
Hh010000000100110111110000000000000000000000000000000000111Hh100000000100101111110000000000000000000000000000000000001Hh10000000010000110111111110001110000011011101001000000011
Bytes:
02,B2,0F,00,00,00,C0,01,D2,0F,00,00,00,00,01,C2,FE,71,B0,4B,C0
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  3213
HEADER SPACE: 9280
BIT MARK:     505
ZERO SPACE:   400
ONE SPACE:    1466
Decoding known protocols...
Looks like a Samsung protocol
POWER OFF

I recorded the OFF signal with AnalysisIR sketch and then sent it as raw (with iRemote library) and it works. It was decoded with same bytes as actual remote.
I am using your library with two other aircons (not Samsung) and it works flawlessly. I guess something is odd with this model.

ToniA, 2019-08-21:
Ah, now I get it... I have hardwired it to always use 'heat' mode when powering off. I need to make a code change.

ToniA, 2019-08-21:
I created a new version of the library, 1.0.10. I this this commit should fix it: https://github.com/ToniA/arduino-heatpumpir/commit/bed6c9e2e1eb0bea3cdfd79f3ead597e8a4eecad

You need to use 'cool' mode in the 'off' message. Let me know if this works.

niccodemi, 2019-08-21:
I cannot compile the code with new library. Before I had earlier version of heatpumpir library (not latest). I get several errors like below. Using Mysensors sketch from heatpumpir github website.

libraries\heatpumpir/CarrierHeatpumpIR.h:63:5: error: 'CarrierHeatpumpIR::CarrierHeatpumpIR()' is protected

     CarrierHeatpumpIR();

     ^

sketch_aug21a:72:52: error: within this context

                              new CarrierHeatpumpIR(),      // 4

ToniA, 2019-08-21:
Oh yes, I have splitted up some models into derived classes, and that would need to be taken into account in the Mysensors sketch as well... I haven't updated that one for ages.

What if you just use the library version you had, and just apply this change into the code: https://github.com/ToniA/arduino-heatpumpir/commit/bed6c9e2e1eb0bea3cdfd79f3ead597e8a4eecad

niccodemi, 2019-08-22:
Off command works as well after changing the code. 

Thank you, this issue can be closed.

closed 2019-08-22 by ToniA