ToniA/arduino-heatpumpir #86: Gree AC I Feel function

Machaeon, 2019-08-27:
Is it posible to implement the function, when remote sends to AC temperature value from its own sensor?
Thank you.

ToniA, 2019-08-27:
Sure, we'd just need to know what the message is.

Machaeon, 2019-08-28:
The only reversing of this protocol that I found is [here](http://courses.cs.tau.ac.il/embedded/projects/fall2010/i-feel/project.htm)
I have AC with this function. How can I help you in implementation?

ToniA, 2019-08-29:
I took a look at that, but I don't think it's enough. I would really need the signals from a real remote controller.

Machaeon, 2019-08-31:
I can try to make a dump. Can you tell me which soft I have to use?

ToniA, 2019-09-02:
This is the setup I have been using: https://github.com/ToniA/Raw-IR-decoder-for-Arduino

duyuelryel, 2020-11-01:
Greetings!
If you still working on this, then 
i started to listen to my remote's maybe IFEEL datas with my tasmotized wemos d1 mini. i get this message:

"{"Time":"2020-11-01T17:48:53","IrReceived":{"Protocol":"UNKNOWN","Bits":18,"Hash":1593411057,"RawData":[6044,2952,692,510,696,510,694,516,688,1612,696,1612,694,512,694,514,718,488,694,1610,696,512,694,1612,694,514,692,512,692,1612,696,514,690,1612,694],"RawDataInfo":[35,35,0]}}"

If you need more, i'll save more data.
Actually i need it for an other project to be able to send specific temperature data with other irsend device.
(https://github.com/inwaar/node-red-contrib-gree-hvac)

cmroche, 2020-11-12:
I have some IR code for this, "I Feel" is associated with lines 16 and 17 which are not present at all when the feature is turned off:

```
Number of symbols: 161
Symbols:
Hh00110010111000000000011000001010010W10000000001000100000000000001000WHh00110010111000000000011000001110010W00000000000000000000000000001011hHh0101100010100101
Bytes:
00:  0011|0010 | 4C | 01001100 
01:  1110|0000 | 07 | 00000111 
02:  0000|0110 | 60 | 01100000 
03:  0000|1010 | 50 | 01010000 
04:  01004:  1000|0000 | 01 | 00000001 
05:  0010|0010 | 44 | 01000100 
06:  0000|0000 | 00 | 00000000 
07:  0000|1000 | 10 | 00010000 
08:  0011|0010 | 4C | 01001100 
09:  1110|0000 | 07 | 00000111 
10:  0000|0110 | 60 | 01100000 
11:  0000|1110 | 70 | 01110000 
12:  01012:  0000|0000 | 00 | 00000000 
13:  0000|0000 | 00 | 00000000 
14:  0000|0000 | 00 | 00000000 
15:  0000|1011 | D0 | 11010000 
16:  0101|1000 | 1A | 00011010 
17:  1010|0101 | A5 | 10100101 
4C,07,60,50,01,44,00,10,4C,07,60,70,00,00,00,D0,1A,A5
Timings (in us): 
PAUSE SPACE:  27293
HEADER MARK:  8213
HEADER SPACE: 5088
BIT MARK:     667
ZERO SPACE:   420
ONE SPACE:    1570
Decoding known protocols...
Unknown protocol
Bytecount: 18
```

cmroche, 2021-02-09:
Digging deeper, I noticed that the decoder is very forgiving. If you look at the raw timings, for all intents and purposes bytes 16 & 17 on the decoded messages are really just a different message. Basically the short message with temperature updates.

Byte 5 bit 3 (LSB) is set to 1 when I-Feel is on, 0 otherwise.

[image]

closed 2021-02-10 by ToniA

edisonbar, 2024-03-21:
Any update regarding I feel?

PR #116 (2021-02-10): Add support for GreeYAC and I-Feel
A few changes coming in around Gree heatpumps.

1. Classes for GreeYAN, GreeYAA didn't actually implement their send() functions properly, that is fixed.
2. Added GreeYAC class and send with support for Horizontal Swing, I-Feel, and current temperature
3. Added example for testing GreeYAC class

I've tested this on a breadboard for the moment, with and IR receiver and it does appear to work. It'll need more testing which I'll carry out in the coming weeks, but I think this is good to merge right now.

CLOSES: #86