ToniA/arduino-heatpumpir #87: Toshiba RAS-10EKV not all functions work WH-UA01NE

geniusalexey, 2019-08-29:
Hello, i have Toshiba RAS-10EKV heatpump with remote WH-UA01NE, some functions dont work.

<details>
<summary>
<i>
Checksum in some commands does not match real 
</i>
</summary>
<p>
Select model to decode (this affects the IR signal timings detection):
* '1' for Panasonic DKE>, Mitsubishi Electric, Fujitsu etc. codes
* '2' for Panasonic CKP, Midea etc. codes
* '3' for Mitsubishi Heavy etc. codes
* '4' for Hyundai etc. codes
* '5' for Samsung etc. codes
* '9' for entering the bit sequence on the serial monitor (instead of the IR receiver)

Enter choice: 4

Ready to decode IR for choice '4'

Number of symbols: 150
Symbols:
Hh111100100000110100000011111111000000000110100000110000000000000001100001hHh111100100000110100000011111111000000000110100000110000000000000001100001
Bytes:
00:  1111|0010 | 4F | 01001111 
01:  0000|1101 | B0 | 10110000 
02:  0000|0011 | C0 | 11000000 
03:  1111|1100 | 3F | 00111111 
04:  0000|0001 | 80 | 10000000 
05:  1010|0000 | 05 | 00000101 
06:  1100|0000 | 03 | 00000011 
07:  0000|0000 | 00 | 00000000 
08:  0110|0001 | 86 | 10000110 
09:  1111|0010 | 4F | 01001111 
10:  0000|1101 | B0 | 10110000 
11:  0000|0011 | C0 | 11000000 
12:  1111|1100 | 3F | 00111111 
13:  0000|0001 | 80 | 10000000 
14:  1010|0000 | 05 | 00000101 
15:  1100|0000 | 03 | 00000011 
16:  0000|0000 | 00 | 00000000 
17:  0110|0001 | 86 | 10000110 
4F,B0,C0,3F,80,05,03,00,86,4F,B0,C0,3F,80,05,03,00,86
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  4450
HEADER SPACE: 5280
BIT MARK:     447
ZERO SPACE:   520
ONE SPACE:    1642
Decoding known protocols...
Looks like a Carrier protocol #1
FAN: 5
MODE: AUTO
Temperature: 27
ModeFan  00000011
Checksum 10000111 does not match real
checksum 10000110

Number of symbols: 150
Symbols:
Hh111100100000110100000011111111000000000101110000110000000000000010110001hHh111100100000110100000011111111000000000101110000110000000000000010110001
Bytes:
00:  1111|0010 | 4F | 01001111 
01:  0000|1101 | B0 | 10110000 
02:  0000|0011 | C0 | 11000000 
03:  1111|1100 | 3F | 00111111 
04:  0000|0001 | 80 | 10000000 
05:  0111|0000 | 0E | 00001110 
[... 2193 lines cut ...]
14:  0111|0000 | 0E | 00001110 
15:  1100|0011 | C3 | 11000011 
16:  0000|0000 | 00 | 00000000 
17:  1011|0010 | 4D | 01001101 
4F,B0,C0,3F,80,0E,C3,00,4D,4F,B0,C0,3F,80,0E,C3,00,4D
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  4540
HEADER SPACE: 5193
BIT MARK:     572
ZERO SPACE:   420
ONE SPACE:    1540
Decoding known protocols...
Looks like a Carrier protocol #1
FAN: 5
MODE: HEAT
Temperature: 24
ModeFan  11000011
Checksum 01001100 does not match real
checksum 01001101

</p>
</details>

what other information can I provide about this remote?

ToniA, 2019-08-30:
I originally wrote this code for my brother, I loaned his remote controller for some time. No matter how hard I tried, I just couldn't figure out the algorithm for the checksum. Take a look at this, to see what I mean: https://github.com/ToniA/arduino-heatpumpir/blob/33f32d13d9a1a20a25dd23572c4a3956b0f4bb58/CarrierHeatpumpIR.cpp#L139

nelgi, 2019-09-30:
Attached is my own personal driver to drive my Toshiba Daiseik-Ai - Checksum, Hi-power and ECO modes (additionally also Frostguard +8 celsius) work on my device. I actually took the original Toshiba driver, and modified it (basically simply changed that the code is sent twice), the codes i think i verified with the original remote, but do not remember exactly as i did this 2 years ago.

[HeatpumpIR.zip](https://github.com/ToniA/arduino-heatpumpir/files/3673368/HeatpumpIR.zip)

mattiarossi, 2020-02-10:
Hi,
the code for the toshiba CRC can be found in this post:
http://blog.mrossi.com/2016/05/toshiba-air-conditioner-ir-signal.html
It is a bit addition over all the other bytes of the data buffer
```
  // Byte 9 - CRC
  data[8] = 0;
  for (i = 0; i < HVAC_TOSHIBA_DATALEN - 1; i++) {
    data[HVAC_TOSHIBA_DATALEN-1] = (byte) data[i] ^ data[HVAC_TOSHIBA_DATALEN -1];  // CRC is a simple bits addition
  }
```