ToniA/arduino-heatpumpir #88: Doesn't compile for Arduino Leonardo (ATMega 32u4)

augustozanellato, 2019-08-30:
The library doesn't compile if the selected board is Leonardo or any board based on 32u4 because timer registers names are different

augustozanellato, 2019-08-30:
Compilation errors:
```
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp: In member function 'virtual void IRSenderPWM::setFrequency(int)':
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:127:7: error: 'TCCR2A' was not declared in this scope
       TCCR2A = _BV(WGM20);
       ^
In file included from /home/augusto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/avr/include/avr/io.h:99:0,
                 from /home/augusto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/avr/include/avr/pgmspace.h:90,
                 from /home/augusto/.arduino15/packages/arduino/hardware/avr/1.6.23/cores/arduino/Arduino.h:28,
                 from /home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:1:
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:127:20: error: 'WGM20' was not declared in this scope
       TCCR2A = _BV(WGM20);
                    ^
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:128:7: error: 'TCCR2B' was not declared in this scope
       TCCR2B = _BV(WGM22) | _BV(CS20);
       ^
In file included from /home/augusto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/avr/include/avr/io.h:99:0,
                 from /home/augusto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/avr/include/avr/pgmspace.h:90,
                 from /home/augusto/.arduino15/packages/arduino/hardware/avr/1.6.23/cores/arduino/Arduino.h:28,
                 from /home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:1:
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:128:20: error: 'WGM22' was not declared in this scope
       TCCR2B = _BV(WGM22) | _BV(CS20);
                    ^
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:128:33: error: 'CS20' was not declared in this scope
       TCCR2B = _BV(WGM22) | _BV(CS20);
                                 ^
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:129:7: error: 'OCR2A' was not declared in this scope
       OCR2A = pwmval8;
       ^
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:130:7: error: 'OCR2B' was not declared in this scope
       OCR2B = pwmval8 / 3;
       ^
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp: In member function 'virtual void IRSenderPWM::mark(int)':
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:184:8: error: 'TCCR2A' was not declared in this scope
       (TCCR2A |= _BV(COM2B1)); // Enable pin 3 PWM output
        ^
In file included from /home/augusto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/avr/include/avr/io.h:99:0,
                 from /home/augusto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/avr/include/avr/pgmspace.h:90,
                 from /home/augusto/.arduino15/packages/arduino/hardware/avr/1.6.23/cores/arduino/Arduino.h:28,
                 from /home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:1:
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:184:22: error: 'COM2B1' was not declared in this scope
       (TCCR2A |= _BV(COM2B1)); // Enable pin 3 PWM output
                      ^
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:195:22: error: 'COM2A1' was not declared in this scope
       (TCCR2A |= _BV(COM2A1)); // Enable pin 11 PWM output
                      ^
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp: In member function 'virtual void IRSenderPWM::space(int)':
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:239:8: error: 'TCCR2A' was not declared in this scope
       (TCCR2A &= ~(_BV(COM2B1))); // Disable pin 3 PWM output
        ^
In file included from /home/augusto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/avr/include/avr/io.h:99:0,
                 from /home/augusto/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/avr/include/avr/pgmspace.h:90,
                 from /home/augusto/.arduino15/packages/arduino/hardware/avr/1.6.23/cores/arduino/Arduino.h:28,
                 from /home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:1:
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:239:24: error: 'COM2B1' was not declared in this scope
       (TCCR2A &= ~(_BV(COM2B1))); // Disable pin 3 PWM output
                        ^
/home/augusto/Arduino/libraries/arduino-heatpumpir/IRSenderPWM.cpp:250:24: error: 'COM2A1' was not declared in this scope
       (TCCR2A &= ~(_BV(COM2A1))); // Disable pin 11 PWM output
```

ToniA, 2019-09-02:
I think it would need another implementation of the https://github.com/ToniA/arduino-heatpumpir/blob/master/IRSender.cpp class, for example by implementing one based on the IRLib2 library (https://github.com/cyborg5/IRLib2).

The IRSenderPWM implementation of the IRSender works on older Arduino's, like the Duemilanove and Mega. Nowadays I'm using ESP8266-based microcontrollers and the IRSenderESP8266 sender class.

Try if the IRSenderBitBang implementation would work for you.

augustozanellato, 2019-09-02:
I tried using that, but it seems like that IRSenderPWM is always included in the build, so it always causes errors.
So I tried your library with an Arduino uno just to confirm that it was compatible with my Samsung ac, but it seems like the protocol is slightly different so it doesn’t work.
In the end I tried to use an ESP8266 with [this library](https://github.com/crankyoldgit/IRremoteESP8266) which seems to work fine with my ac.

But yeah, nowadays I also use almost only esp8266/esp32 and stm32 boards, I was trying this project on a Leonardo with yun shield but I switched to an esp8266 because of the protocol incompatibilities