ToniA/arduino-heatpumpir #89: How can I know if the parameters are correctly setup?

Luisiado, 2019-09-11:
Hello, thanks for this library its amazing.
I am sending parameters to my fujitsu cassette unit, model ARYG24LMLA.
I can switch ON and switch OFF.
But about the other instruction I dont know if are working or not, cause I can not read the actual configuration of the system.
At least I test FANspeed and swings dont work. 
Some modes works.
About temperature I dont know how to check if it is working or not.

I also tried with the special parameters ( Fujitsu codes).
Also there are some special functions like sendFujitsuHiPower(IRSender& IR).

ToniA, 2019-09-11:
There's really no easy way to check, as the IR communication is one-way only, there's no feedback from the A/C unit, other than the 'beep' sound. I think the best you can do is to use this (https://github.com/ToniA/Raw-IR-decoder-for-Arduino) on Arduino with an IR receiver, and check if the messages are the same as they are with the real remote controller.

Luisiado, 2019-09-12:
Thank you! it was much easier than I had expected