ToniA/arduino-heatpumpir #91: Midea > celsius vs fahrenheit

ameoba32, 2019-10-04:
Great library!! It works finally!!!  Thank you!

Minor thing: command control?cmd=heatpumpir,midea,1,2,0,26,6,0 sets AC to display 80F at the device, not to 26C.   

I think some bit is not set up on the IR packet.  How can I debug it using my remote?

Thanks,

ToniA, 2019-10-04:
You're welcome :) For debugging, I have used this (https://github.com/ToniA/Raw-IR-decoder-for-Arduino) to read the IR remote commands. So you'd need to check out what are the differences between the signal the library sends, and the signal from the real remote.

ameoba32, 2019-10-04:
Thanks!  It looks "Raw-IR-decoder-for-Arduino" does not work on NodeMcu.. I have device in reboot loop.  Tried to fix it but not sure why.

But I found another program, called "IRrecvDumpV2" from IRremoteESP8266 library and it works and outputs raw data.

But here I am bit confused :)   On AC remote it outputs "Encoding  : MIDEA".  BUT when I use  "heatpumpir,midea,1,2,0,26,6,0" it outputs "Encoding  : COOLIX"  Weird.

```
# Real AC remote
Timestamp : 000076.541
Library   : v2.6.6
Encoding  : MIDEA
Code      : A1A346FFFF6A (48 bits)
Mesg Desc.: Power: On, Mode: 3 (Heat), Celsius: On, Temp: 23C/73F, Fan: 4 (UNKNOWN), Sleep: Off, Swing(V) Toggle: Off
Raw Timing[199]:
   +  4420, -  4362,    +   568, -  1584,    +   566, -   508,    +   594, -  1556, 
   +   566, -   508,    +   566, -   508,    +   566, -   508,    +   568, -   506, 
   +   568, -  1582,    +   568, -  1582,    +   568, -   506,    +   568, -  1582, 
   +   568, -   508,    +   562, -   512,    +   566, -   508,    +   568, -  1582, 
   +   564, -  1584,    +   566, -   508,    +   568, -  1582,    +   564, -   510, 
   +   566, -   508,    +   566, -   508,    +   564, -  1584,    +   566, -  1584, 
   +   566, -   508,    +   568, -  1582,    +   566, -  1582,    +   566, -  1584, 
   +   566, -  1584,    +   566, -  1584,    +   568, -  1582,    +   568, -  1582, 
   +   568, -  1582,    +   562, -  1588,    +   564, -  1584,    +   570, -  1582, 
   +   566, -  1584,    +   568, -  1582,    +   568, -  1582,    +   570, -  1580, 
   +   564, -  1584,    +   566, -   510,    +   566, -  1582,    +   566, -  1586, 
   +   566, -   508,    +   568, -  1582,    +   566, -   508,    +   566, -  1584, 
   +   566, -   508,    +   566, -  5178,    +  4420, -  4364,    +   568, -   508, 
   +   568, -  1580,    +   566, -   508,    +   566, -  1582,    +   568, -  1582, 
   +   568, -  1584,    +   568, -  1582,    +   564, -   508,    +   568, -   506, 
   +   564, -  1572,    +   582, -   506,    +   568, -  1580,    +   566, -  1584, 
   +   566, -  1584,    +   568, -   506,    +   570, -   504,    +   570, -  1580, 
   +   568, -   506,    +   568, -  1582,    +   568, -  1582,    +   570, -  1580, 
   +   566, -   508,    +   564, -   510,    +   564, -  1584,    +   568, -   508, 
   +   564, -   510,    +   568, -   506,    +   564, -   512,    +   562, -   510, 
   +   566, -   508,    +   568, -   506,    +   564, -   510,    +   568, -   506, 
   +   564, -   508,    +   566, -   508,    +   568, -   506,    +   568, -   508, 
   +   564, -   510,    +   566, -   510,    +   564, -   508,    +   564, -  1584, 
   +   568, -   508,    +   568, -   506,    +   568, -  1582,    +   566, -   508, 
   +   568, -  1580,    +   568, -   508,    +   568, -  1584,    +   538

uint16_t rawData[199] = {4420, 4362,  568, 1584,  566, 508,  594, 1556,  566, 508,  566, 508,  566, 508,  568, 506,  568, 1582,  568, 1582,  568, 506,  568, 1582,  568, 508,  562, 512,  566, 508,  568, 1582,  564, 1584,  566, 508,  568, 1582,  564, 510,  566, 508,  566, 508,  564, 1584,  566, 1584,  566, 508,  568, 1582,  566, 1582,  566, 1584,  566, 1584,  566, 1584,  568, 1582,  568, 1582,  568, 1582,  562, 1588,  564, 1584,  570, 1582,  566, 1584,  568, 1582,  568, 1582,  570, 1580,  564, 1584,  566, 510,  566, 1582,  566, 1586,  566, 508,  568, 1582,  566, 508,  566, 1584,  566, 508,  566, 5178,  4420, 4364,  568, 508,  568, 1580,  566, 508,  566, 1582,  568, 1582,  568, 1584,  568, 1582,  564, 508,  568, 506,  564, 1572,  582, 506,  568, 1580,  566, 1584,  566, 1584,  568, 506,  570, 504,  570, 1580,  568, 506,  568, 1582,  568, 1582,  570, 1580,  566, 508,  564, 510,  564, 1584,  568, 508,  564, 510,  568, 506,  564, 512,  562, 510,  566, 508,  568, 506,  564, 510,  568, 506,  564, 508,  566, 508,  568, 506,  568, 508,  564, 510,  566, 510,  564, 508,  564, 1584,  568, 508,  568, 506,  568, 1582,  566, 508,  568, 1580,  568, 508,  568, 1584,  538};  // MIDEA A1A346FFFF6A
uint64_t data = 0xA1A346FFFF6A;

# heatpumpir,midea,1,2,0,26,6,0"
Timestamp : 000174.370
Library   : v2.6.6
Encoding  : COOLIX
Code      : B2BF5C (24 bits)
Mesg Desc.: Power: On, Mode: 3 (Heat), Fan: 5 (Auto), Temp: 23C, Zone Follow: Off, Sensor Temp: Ignored
Raw Timing[199]:
   +  4390, -  4304,    +   644, -  1552,    +   644, -   492,    +   618, -  1546, 
   +   666, -  1522,    +   644, -   490,    +   618, -   490,    +   618, -  1548, 
   +   690, -   422,    +   640, -   470,    +   640, -  1548,    +   642, -   466, 
   +   644, -   490,    +   642, -  1522,    +   644, -  1546,    +   640, -   464, 
   +   642, -  1550,    +   638, -  1548,    +   642, -   490,    +   618, -  1546, 
   +   646, -  1648,    +   522, -  1562,    +   644, -  1568,    +   618, -  1568, 
   +   618, -  1546,    +   644, -   492,    +   616, -  1546,    +   644, -   516, 
   +   590, -   494,    +   616, -   492,    +   616, -   490,    +   614, -   470, 
   +   642, -   468,    +   666, -   466,    +   618, -  1570,    +   620, -   464, 
   +   668, -  1520,    +   642, -  1546,    +   642, -  1546,    +   642, -   492, 
   +   618, -   490,    +   614, -  1548,    +   658, -   452,    +   640, -  1546, 
   +   642, -   468,    +   642, -   466,    +   644, -   464,    +   640, -  1546, 
   +   644, -  1544,    +   646, -  5088,    +  4440, -  4316,    +   616, -  1548, 
   +   644, -   488,    +   616, -  1546,    +   642, -  1572,    +   616, -   492, 
[... cut ...] , 1546,  644, 464,  642, 466,  640, 1548,  642, 1548,  642, 466,  642, 1570,  616, 1548,  646, 460,  648, 1568,  618, 1570,  616, 1548,  668, 1522,  644, 1570,  618, 1572,  618, 466,  640, 1546,  642, 466,  644, 464,  644, 490,  644, 464,  618, 490,  644, 442,  640, 468,  644, 1542,  668, 468,  614, 1548,  642, 1548,  638, 1548,  642, 490,  620, 466,  642, 1548,  642, 490,  642, 1520,  646, 466,  638, 466,  644, 492,  640, 1524,  636, 1550,  644};  // COOLIX B2BF5C
uint64_t data = 0xB2BF5C;

```

ToniA, 2019-10-05:
The plugin P035 also seems to support these devices, what if you try to use that one instead ("Communication - IR Transmit")?

One example of the command is this:

IRSENDAC, { 'protocol': 'WHIRLPOOL_AC', 'model': 2, 'power', 'he', 'mode' 'cool', 'Temp': 22, 'swingv' 'off', 'swingh' : 'off', 'light', 'on'}

ameoba32, 2019-10-05:
I found that IRremoteESP8266 library also supports a lot of AC.  I tried MIDEA protocol and it works, shows correct temp in Celsius.   So guess I'll switch to it and do wifi part myself, instead of easyspa.

Here is example. 
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/CommonAcControl/CommonAcControl.ino

Thank you for help!