ToniA/arduino-heatpumpir #100: Mitsubishi MSY Heat not working

RubyBailey, 2020-02-13:
First time using GitHub, please let me know if following is too wordy or should be posted differently next time.
I’ll post notes, questions and comments first because the following text is fairly wordy with explanations:
1.	See code snippet below MITSUBISHI_FA seems not to be executed.
2.	MITSUBISHI_MSY code does not have HEAT_MODE
3.	MITSUBISHI_MSY COOL same as DRY
4.	How can I help with the coding and protocol for the Mitsubishi MSY / MSZ units?
5.	I have never used GitHub so the learning curve may be a bit steep.
6.	Any suggestions welcome.
7.	Thanks

We have MSZ units, they should use the same protocol as the MSY’s

System: NodeMCU ESP12E
Git Build: mega-20191208
Binary File: ESPEasy build ESP_Easy_mega-20191208_normal_IRext_no_rx_ESP8266_4M2M.bin

Confirmed Heatpump IR transmitter working through IP command and MQTT.
Sending command: heatpumpir,mitsubishi_msy,1,2,0,20,0,0
Which per https://espeasy.readthedocs.io/en/latest/Plugin/P088.html should be Power ON, Heat, Auto Fan Speed, 20C, No SwngV, No SwingH

Heat pump started but didn’t seem to be heating.

Downloaded and installed https://github.com/ToniA/Raw-IR-decoder-for-Arduino and decoded IR signal from ESPEasy.

Serial Monitor returned:
Decoding known protocols...
Looks like a Mitsubishi FD / FE / MSY series protocol
Checksum matches
POWER ON
MODE COOL
I-See: OFF
Plasma: OFF
Temperature: 20
FAN AUTO
VANE: AUTO1?
WIDE VANE: SWING 

Changed the Modes from 1 through 6, these reported:
Mode 1: unknown
Mode 2: Cool
Mode3: Cool
Mode 4: Cool
Mode 5: Fan
Mode 6: Cool

When I look at the code in MitsubishiHeatpumpIR.cpp, from line 96

if (_mitsubishiModel !=  MITSUBISHI_MSY)
  {
.
(if it’s not a MSY execute this code, which I think would include the FA type unit)
.
        break;
    }
  }
  else if (_mitsubishiModel == MITSUBISHI_FA) // set operating model for FA
  {
.
(I don’t think this code would ever get executed)
.
  }
  else
  {
There’s no HEAT MODE in the code

	  operatingMode = MITSUBISHI_AIRCON2_MODE_COOL 
	  switch (operatingModeCmd)
	  {
	  case MODE_AUTO:
		  operatingMode = MITSUBISHI_AIRCON2_MODE_IFEEL;
		  break;
	  case MODE_COOL:
		  operatingMode = MITSUBISHI_AIRCON2_MODE_COOL;
		  break;
	  case MODE_DRY:
		  operatingMode = MITSUBISHI_AIRCON2_MODE_DRY;
		  break;
	  case MODE_FAN:
		  operatingMode = MITSUBISHI_AIRCON2_MODE_FAN;
		  break;
	  }  
  }

Also noted in MitsubishiHeatpumpIR.h at line 28  and 31 that cool and dry mode are the same 0x18
#define MITSUBISHI_AIRCON2_MODE_COOL  0x18 // MSY cool mode

#define MITSUBISHI_AIRCON2_MODE_DRY   0x18 // MSY DRY mode

From our Mitsubishi remote I get the following  from the Arduino

Using Mitsubishi remote, Power ON, Mode Heat, Fan Auto, 20C, Vanes: Position 1, Wide Vanes: middle
Heat is 0x08

23,CB,26,01,00,20,08,04,30,50,65,00,00,00,00,00,00,26,23,CB,26,01,00,20,08,04,30,50,65,00,00,00,00,00,00,26
Timings (in us): 
PAUSE SPACE:  17520
HEADER MARK:  3580
HEADER SPACE: 1710
BIT MARK:     338
ZERO SPACE:   380
ONE SPACE:    1287
Decoding known protocols...
Looks like a Mitsubishi FD / FE / MSY series protocol
Checksum matches
POWER ON
MODE HEAT
I-See: OFF
Plasma: OFF
Temperature: 20
FAN AUTO
VANE: UP-1
WIDE VANE: MIDDLE 

Using Mitsubishi remote, Power ON, Mode Auto, Fan Auto, 20C, Vanes: Position 1, Wide Vanes: middle
Auto is 0x04

23,CB,26,01,00,20,20,04,36,40,65,00,00,00,00,00,00,34,23,CB,26,01,00,20,20,04,36,40,65,00,00,00,00,00,00,34
Timings (in us): 
PAUSE SPACE:  17580
HEADER MARK:  3560
HEADER SPACE: 1730
BIT MARK:     332
ZERO SPACE:   373
ONE SPACE:    1296
Decoding known protocols...
Looks like a Mitsubishi FD / FE / MSY series protocol
Checksum matches
POWER ON
MODE AUTO
I-See: OFF
Plasma: OFF
Temperature: 20
FAN AUTO
VANE: AUTO1?
WIDE VANE: MIDDLE 
Using Mitsubishi remote, Power ON, Mode Cool, Fan Auto, 20C, Vanes: Position 1, Wide Vanes: middle
Cool is 0x18

23,CB,26,01,00,20,18,04,36,50,65,00,00,00,00,00,00,3C,23,CB,26,01,00,20,18,04,36,50,65,00,00,00,00,00,00,3C
Timings (in us): 
PAUSE SPACE:  17740
HEADER MARK:  3540
HEADER SPACE: 1740
BIT MARK:     328
ZERO SPACE:   388
ONE SPACE:    1294
Decoding known protocols...
Looks like a Mitsubishi FD / FE / MSY series protocol
Checksum matches
POWER ON
MODE COOL
I-See: OFF
Plasma: OFF
Temperature: 20
FAN AUTO
VANE: UP-1
WIDE VANE: MIDDLE 
Using Mitsubishi remote, Power ON, Mode Dry, Fan Auto, 20C, Vanes: Position 1, Wide Vanes: middle
Dry is 0x10

23,CB,26,01,00,20,10,08,32,50,66,00,00,00,00,00,00,35,23,CB,26,01,00,20,10,08,32,50,66,00,00,00,00,00,00,35
Timings (in us): 
PAUSE SPACE:  17600
HEADER MARK:  3550
HEADER SPACE: 1730
BIT MARK:     340
ZERO SPACE:   360
ONE SPACE:    1279
Decoding known protocols...
Looks like a Mitsubishi FD / FE / MSY series protocol
Checksum matches
POWER ON
MODE DRY
I-See: OFF
Plasma: OFF
Temperature: 24
FAN AUTO
VANE: UP-1
WIDE VANE: MIDDLE

ToniA, 2020-02-13:
Thanks, this is really good stuff, I'll take a look at this over the weekend.

RubyBailey, 2020-02-13:
Thanks, that's great. 
I just noticed a mistake in my message Auto is 0x20, not 0x04 from our remote. It's correct in the arduino recorded 0x string in the message, I just copied it wrong when I wrote it in my text.

ToniA, 2020-02-16:
Would you try out this firmware? It's a minimal_IRext_ESP8266_4M1M build from this: https://github.com/ToniA/ESPEasy/tree/mitsu_msy

https://1drv.ms/u/s!AsOVS72cZynsif0o6Ig2EKEtFbDZrg?e=Rcrmom

RubyBailey, 2020-02-17:
Thanks. I should be able to try it out in the next day or two.

RubyBailey, 2020-02-18:
Heat now works on our units.  Thanks.

I had some challenges getting your minimal_IRext_ESP8266_4M1M build to load.  Loading it over an existing  ESP_Easy_mega-20191208_normal_IRext_no_rx_ESP8266_4M2M.bin build resulted in the ModeMCU not communicating at all. Reloading the  ESP_Easy_mega-20191208_normal_IRext_no_rx_ESP8266_4M2M.bin build same results. Had to load the blank_4MB.bin before loading the minimal_IRext_ESP8266_4M1M build. Works now.

A couple of questions:
1)	Why are Cool and Dry the same 0x18? On our remote it was 0x18 and 0x10. I would expect them to be different.
2)	Regarding the build you compiled and the one I initially used; is there a list of which plugin is included in each of these builds? I have VSC and Platformio, worked my way through the various platformio*.ini files back to define_plugin_sets.h, yet finding it challenging to determine which plug-ins are included in each of the builds.

RubyBailey, 2020-02-18:
Just noticed that the build you provided does not include the log views. Would you please recompile so those are included that we can verify the IR commands are being received and interpreted correctly. Thanks.

ToniA, 2020-02-18:
I fixed the DRY code, and published it in Github, on my branch mentioned above. As you have VS Code and PlatformIO, you could try to build this yourself. Just type this into the VS Code Terminal window, on the root directory of the Git repo (you can obviously change the image type if you like, but only the IRext image includes the HeatpumpIR plugin):

`platformio run -e minimal_IRext_ESP8266_4M1M --target upload`

Build flags are defined in environment variables, in VS Code Powershell you can for example set this before building the firmware:

`$env:PLATFORMIO_BUILD_FLAGS = '-DIR_DEBUG_PACKET'`

ESPEasy defines the plugin sets in the file src\define_plugin_sets.h

RubyBailey, 2020-02-27:
Noticed that the update hasn't been pushed to EspEasy. Still code with no heat.

ToniA, 2020-02-27:
I asked above if you could build this branch yourself. Let me know if it works.

RubyBailey, 2020-02-27:
Thanks. My misunderstanding. I will work on it and let you know.

RubyBailey, 2020-03-03:
I have downloaded, compiled and successfully uploaded the modifications. They work. At least for heating mode. 
But, having looked at the code as well as decoded the signals from the Mitsubishi remotes. (Thanks for the decoding program. It worked extremely well) I`m not completely happy with the results of the heatpumpir output for the MSZ units.
Decoded file: [https://drive.google.com/open?id=1l0w-XOn3esRBHpRagOv-N2E5HAjH774X] 
The MSZ-A24, D36, GE18 and KA18 our from me. The MSZ-GF60 and FD25 is from Vincent (r45635), see link on that page. 
The issue is that the MSZ`s are different enough form the MSY units that it may justify including them as a separate option in the code. For example they do not have an IFEEL mode, they have AUTO. In the MSY section where the operating mode is set selecting AUTO sets the Mode to IFEEL.
A couple of other items:
MitsubishiHeatpumpIR.h there are #define MITSUBISHI_AIRCON1_VS_AUTO2 and MITSUBISHI_AIRCON1_VS_AUTO3 with questions marks. I think these are from byte 9. When either the Fan or Vertical Vanes are switched to Auto the 8th bit become 1, 7th 0. Next time any button is pressed the 8th goes back to 0, 7th to 1. See "SWITCH TO AUTO" decode document. I don't think the bit does anything, because the heat pump recognizes FAN Auto and VERT VANE Auto by the other bits in that byte. 
MitsubishiHeatpumpIR.cpp MitsubishiTemplate[8] may need some work because for all the remotes I decoded that byte includes both MODE and WIDE VANE settings, whereas the code doesn't aggregate both settings into the byte. (I think it would look similar to MitsubishiTemplate[9] which has FAN SPEED and VERTICAL VANES in the same byte)
I tried to take a look at the code, but my knowledge currently is very limited. So I don't have a good idea of how to approach this challenge. Suggestions welcome.
One other thing: the https://espeasy.readthedocs.io/en/latest/Plugin/P088.html description says that vertical and horizontal swing position (not supported on most models), yet in the case of these Mitsubishi's they are. It would be helpful to at least have a list of the options and settings on that page so in case they are supported that people don't need to find them in the code. 
Thanks again, like I said suggestions would be welcome on how to move forward here and fully support these Mitsubishi units.

ToniA, 2020-03-03:
The MSY code fix is now in pull request phase: https://github.com/letscontrolit/ESPEasy/pull/2916

I think I should separate out the MSY code from the generic Mitsubishi code, as a separate class. If I do this base work, could you continue from there?

RubyBailey, 2020-03-03:
Since I don't have any MSY heat pump I would be unable to thoroughly test the code to ensure that I haven't broken anything in the process. Also since mine appear to have some differences compared to the MSY another option would be to create a class or classes for the MSZ-A, D, GE and MFZ-KA that I have. I would be able to work on and test those. What do you think?

RubyBailey, 2020-03-04:
Just something else in MitsubishiHeatPump.cpp
  if ( temperature != 10 && temperatureCmd > 16 && temperatureCmd < 32)
  {
    temperature = temperatureCmd;
  }
The above should be >= and <=. If the temperature setting is equal to 16 then the above if statement isn't true and temperature stays at the default 23. Same if the setting is 32.