ToniA/arduino-heatpumpir #102: daikin and daikin_arc417 modules don't work

KameSense, 2020-04-07:
Hello,

I flashed NodeMCU 12E / ESP8266 with ESP_Easy_mega-20200328_minimal_IRext_ESP8266_4M1M.bin

When I browse to http://<IP>/control?cmd=heatpumpir,daikin,1,2,1,26,0,0 the LED doesn't blink at all and I have a "Unknown or restricted command!".

Same with daikin_arc417, but not with a few other manufacturers modules (tried ivt, gree, toshiba).

ToniA, 2020-04-08:
You're right, I have totally forgotten Daikin. I need to make a PR for ESPEasy about this.

The model supported by recent ESPEasy builds can be found from this file: https://github.com/ToniA/arduino-heatpumpir/blob/master/HeatpumpIRFactory.cpp

ToniA, 2020-04-13:
DaikinHeatpumpARC480A14 didn't implement the HeatpumpIR interface. HeatpumpIRFactory didn't include Daikin models, so they didn't work in ESPEasy. I will make a PR to ESPEasy (HeatpumpIR library update).

closed 2020-04-13 by ToniA

ToniA, 2020-04-13:
The PR got merged into ESPEasy, so this should work on the next release (after the current 20200410).