ToniA/arduino-heatpumpir #103: Support for specific Samsung AQV12MSAN airco model

ericvb, 2020-05-16:
Having a problem to power-off the airco, I debugged the existing code for Samsung airco's to get it working with my specific airco model.
If you want, I can do a pull request.
But first, I want to know if this is the way to go when coding for a specific Samsung airco model.

In the SamsungHeatpumpIR.h file I created a new define
`#define MODEL_AQV12_MSAN            1`
```
class SamsungHeatpumpIR : public HeatpumpIR
{
  protected:
    SamsungHeatpumpIR();
    uint8_t _samsungAQVModel;	
  
  public:
    virtual void send(IRSender& IR, uint8_t powerModeCmd, uint8_t operatingModeCmd, uint8_t fanSpeedCmd, uint8_t temperatureCmd, uint8_t swingVCmd, uint8_t swingHCmd);

  private:
    void sendSamsung(IRSender& IR, uint8_t powerMode, uint8_t operatingMode, uint8_t fanSpeed, uint8_t temperature, uint8_t swingV);
};
```
and a new class
```
class SamsungAQV12MSANHeatpumpIR : public SamsungAQVHeatpumpIR
{
  public:
    SamsungAQV12MSANHeatpumpIR();
};
```
In the SamsungHeatpumpIR.cpp file, I added the missing specific code.
```
SamsungAQV12MSANHeatpumpIR::SamsungAQV12MSANHeatpumpIR() : SamsungAQVHeatpumpIR()
{
  static const char model[] PROGMEM = "samsung_aqv12msan";
  static const char info[]  PROGMEM = "{\"mdl\":\"samsung_aqv12msan\",\"dn\":\"Samsung AQV12MSAN\",\"mT\":16,\"xT\":27,\"fs\":4}";

  _model = model;
  _info = info;
  _samsungAQVModel = MODEL_AQV12_MSAN;
}
```
I changed the methods 
`void SamsungAQVHeatpumpIR::send(IRSender& IR, uint8_t powerModeCmd, uint8_t operatingModeCmd, uint8_t fanSpeedCmd, uint8_t temperatureCmd, uint8_t swingVCmd, uint8_t swingHCmd)`
and
`void SamsungAQVHeatpumpIR::sendSamsung(IRSender& IR, uint8_t powerMode, uint8_t operatingMode, uint8_t fanSpeed, uint8_t temperature, uint8_t swingV)`
to include the specific bytes commands from the reverse engineering.

I saw that you already included a specific change for an only cooling airco when powering off, 
so I took the poweroff code to the end of the method to be able to overwrite the settings with specific bytes for my model.
I my case the operatingMode must be the same as the operatingMode when powered on, so the switch operatingModeCmd must be always executed.

The checksumbyte is also different when doing a powering off from heat and cool, but not from auto...   strange things

For the other changes, please see the attached files.
I include also a reverse engineered IR file from my remote ARH-1362

If you want more reverse engineered IR codes, please don't hesitate to request them.

[HeatPumpIR debug aqv12msan.zip](https://github.com/ToniA/arduino-heatpumpir/files/4638381/HeatPumpIR.debug.aqv12msan.zip)

ToniA, 2020-05-17:
I think your code looks good. Would you make a pull request from these changes?

ericvb, 2020-05-18:
@ToniA , please find the pull request in the corresponsing pull request tab #104

closed 2020-05-20 by ToniA