ToniA/arduino-heatpumpir #106: Feature: Provide a generic way to specify additional/model specific settings

arion-p, 2020-06-16:
It would very useful to be able to specify additional model specific parameters on each call to `HeatPumpIR::send()`. In some cases this may not be just desirable but actually necessary as I found out (the hard way). For example in my case my Mitsubishi FD50VF has a "Clean" setting. Last summer I used ESPEasy and this library to control it remotely. Since "Clean" is not one of the abstracted settings it was not activated and due to the high humidity I came back home to find the AC unit full of mildew. 

I can think of two ways to implement this:
1. Pass an (optional) array of tag->value pairs, one pair for each setting. If different makes/models support the same setting/functionality, only named differently, it should be abstracted. Any unsupported tags can be ignored by the driver.
2. Pass a JSON string with the necessary key->value pairs (same as #1 above). I don't like this solution very much because it uses more memory and add dependency to ArduinoJSON library. But it may be easier to implement and maintain

I can help with coding once the architecture is decided

ToniA, 2020-06-17:
Yes, I've been thinking about this since beginning, but didn't really get anywhere with the device-specific settings. I think the JSON string format would probably be the way to go. What if there would be another send() method which would take all input data as JSON, i.e. just one string as the parameter?

I'm also using ESPEasy myself.

arion-p, 2020-06-17:
As far as I can tell this is what they are doing in ESPEasy plugin 35 and IRSENDAC using IRremoteESP8266. Not sure if the average arduino board has enough memory to handle it and still have enough memory left to do something useful.

In any case I think JSON parsing should be done in a generic class and the device specific classes only encoding the parameters as necessary. Perhaps it would be better to have two layers:
- one low level where all parameters are passed as an array of tag->value pairs
- and one upper layer that translates a JSON object an array and call the lower layer

whatsupskip, 2020-06-17:
Does the Mitsubishi FD50VF specifically require the clean mode to be run after each session, or is it run every so often?

arion-p, 2020-06-18:
When setting is active, clean operation will run after every power off. All it does is run the fan for some time after power off to dry out the interior parts of the internal unit thus preventing mold. Whether this is necessary depends on the environment in general (humidity, installation location, ventilation etc). The unit will not check if conditions that require clean mode exist. As long as the setting is activated it will just run clean mode.

ToniA, 2020-06-18:
One workaround for this is to run the unit in FAN mode for a while before finally turning it off. I.e. 'COOL' -> 'FAN' for 30 minutes or so -> 'POWER OFF'.

arion-p, 2020-06-18:
Yep, that was my initial thought. But I may as well do a custom build of ESPEasy and hard-code the "clean" bit in the command template for now.