ToniA/arduino-heatpumpir #108: Feature: Y512F2 "Light button" state emulation

nistvan86, 2020-07-19:
I have a Simbio 12C2 A/C (some kind of Gree rebranding, I've bought it with my apartment) which has a Y512F2 remote.
Controlling the A/C works via the HyundaiHeatpumpIR module, but as soon as I send the command with this library, the display on the unit is turned off, so I don't see what is happening. There's a Light button on the remote to turn the display on/off, I guess the bit responsible in the protocol for this state is not implemented.
Unfortunatelly I don't have an IR receiver to decode the raw IR data from the remote, so I cannot manually add the functionality to this library. I'm going to order an IR sensor, but maybe somebody already knows which bit is responsible for this.

[image]

ToniA, 2020-07-20:
I don't have access to this remote, so I also can't decode the code needed for this :(

nistvan86, 2020-09-02:
Receiver arrived finally. It's working with the Hyundai mode of the Raw-IR-decoder-for-Arduino:

`-- POWER ON, LIGHT OFF

Number of symbols: 38
Symbols:
Hh10010000010100000000000000001010010
Bytes:
00:  1001|0000 | 09 | 00001001 
01:  0101|0000 | 0A | 00001010 
02:  0000|0000 | 00 | 00000000 
03:  0000|1010 | 50 | 01010000 
04:  01009,0A,00,50
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  9140
HEADER SPACE: 4560
BIT MARK:     616
ZERO SPACE:   496
ONE SPACE:    1587
Decoding known protocols...
Looks like a Hyundai protocol
POWER ON
MODE COOL
Temperature: 26
FAN: AUTO
SLEEP: OFF
SWING: OFF

-- POWER ON, LIGHT ON

Number of symbols: 38
Symbols:
Hh10010000010100000000010000001010010
Bytes:
00:  1001|0000 | 09 | 00001001 
01:  0101|0000 | 0A | 00001010 
02:  0000|0100 | 20 | 00100000 
03:  0000|1010 | 50 | 01010000 
04:  01009,0A,20,50
Timings (in us): 
PAUSE SPACE:  0
HEADER MARK:  9140
HEADER SPACE: 4560
BIT MARK:     614
ZERO SPACE:   494
ONE SPACE:    1591
Decoding known protocols...
Looks like a Hyundai protocol
POWER ON
MODE COOL
Temperature: 26
FAN: AUTO
SLEEP: OFF
SWING: OFF`

"Light" button is represented in the second byte (0x20 if on, 0x00 if it isn't).
I'll give it a try later to add it to the library.