ToniA/arduino-heatpumpir #126: How to add Hisense A-series codes

pr0phe, 2021-11-02:
Hi there.
Id like to add Hisense support to yr lib to use it in future project in esphome.
Ive sniff turn on code as follows:

OnRawData : 2600600100012495123811381213121411141114121411381213123812381114121312141114121312131213123812131213121411141214113812131213121312381238123812141113121312131214111412131214121312131213121312131213121411141213120001061114111411141114111412131214123811141138121312141114111412141114121311141114111412131214111412131213121312131213121411141214111411141113121411141114121312141213121312131213121312131214111412141113121312131214111412131214111412371238123812131238123812381238110001061213121312131214111412131214121312381114111411141213121411141213121312131213121312141114121411141114111411141114121312141114121312131213121312381213121411141214111312131213121411141213121411141238111312141138121411141214111314000D050000000000000000
OnRawData Base64 : JgBgAQABJJUSOBE4EhMSFBEUERQSFBE4EhMSOBI4ERQSExIUERQSExITEhMSOBITEhMSFBEUEhQROBITEhMSExI4EjgSOBIUERMSExITEhQRFBITEhQSExITEhMSExITEhMSFBEUEhMSAAEGERQRFBEUERQRFBITEhQSOBEUETgSExIUERQRFBIUERQSExEUERQRFBITEhQRFBITEhMSExITEhMSFBEUEhQRFBEUERMSFBEUERQSExIUEhMSExITEhMSExITEhQRFBIUERMSExITEhQRFBITEhQRFBI3EjgSOBITEjgSOBI4EjgRAAEGEhMSExITEhQRFBITEhQSExI4ERQRFBEUEhMSFBEUEhMSExITEhMSExIUERQSFBEUERQRFBEUERQSExIUERQSExITEhMSExI4EhMSFBEUEhQRExITEhMSFBEUEhMSFBEUEjgRExIUETgSFBEUEhQRExQADQUAAAAAAAAAAA==

Is it usable in that formats?
If yes i will sniff all other actions for that remote.

N-Storm, 2021-11-03:
Please refer to a project README. There is a section at the end "How to contribute" it describes how to properly capture IR data.

pr0phe, 2021-11-18:
Im trying to follow faq:
1. buy m5stak
2. IR reciever
3. install vs code+platformio

and im trying to comile yr code, but every time ive recieved:

Error: Invalid 'C:\Users\User\Documents\PlatformIO\Projects\IR Reciever\platformio.ini' (project configuration file): 'File contains no section headers.
file: 'C:\\Users\\User\\Documents\\PlatformIO\\Projects\\IR Reciever\\platformio.ini', line: 1
'<!DOCTYPE html>\n''

could you pls help?
i doesnt copy code from repo, ive dowloaded it as file.

mickeyva, 2024-07-25:
@pr0phe did you manage to control dg11L1 eventually?