ToniA/arduino-heatpumpir #131: Old Panasonic AC - I can read the remote, now what?

mrgrumpy83, 2022-01-03:
I’ve got an old Panasonic AC which isn’t currently supported. I’m hoping to add it.

I’ve built the circuit and can read the codes from the remote… but I’m stuck here. It’s showing ‘Unknown Protocol’ but I appear to get sensible output (ie. Symbols/headers aren’t random or repeating) when running for Panasonic CKP.

How do I move from this point to a situation where the information is useable and able to be written in as a working module?

ToniA, 2022-01-04:
Look at the different decoders of this schema you are now running. The first step could be to decode the temperature request, i.e. go through the full range of temperatures, and try to write a piece of code which correctly decodes this. Then the fan speed etc.

At least the old Panasonic CKP does not have separate ON and OFF signals, it just has a toggle, i.e. the same signal is used, but it just switches the device on or off.

mrgrumpy83, 2022-01-04:
Thanks for the response!

I'm absolutely no good at coding. I can look at it and kind of figure out what's going on, but I cannot write code unfortunately. I'm happy to work it out manually, though, I just need a little more help understanding what I need to look for.

I can see a clear change in the Hexadecimal output when pressing buttons. Looking at PanasonicCKPHeatpumpIR.h it looks as though the 'CKP codes' are also defined using Hex.

Based on this, is the process a matter of working out - 
1. What the hex codes for each function/mode/output/command are?
2. Which bytes contain the code for the respective function/mode/output/command?

One other thing I can't seem to determine, is where the codes for temperature are defined?

mrgrumpy83, 2022-01-09:
OK, I spent a stack of time on this and answered the questions I had above for myself.
Then when I looked at the appropriate Panasonic code, it turned out it was all exactly the same as Panasonic CKP.
Turns out the receiver is extremely picky about directionality and I may not have quite been aiming the transmitter correctly. Some cursory testing suggests that it's all ok with a CKP.

N-Storm, 2022-01-10:
Which IRSender* class do you use? These are listed on README.md and some offers better carrier frequency accuracy than older IRSenderBitBang. Accuracy affects range & direction.