ToniA/arduino-heatpumpir #132: Gree greeya - change "light" to "off"

TrylobitBroccoli, 2022-01-15:
I have Gree AC's at my place. Remote model number is YAN1F6. "greeya" library works, but it keeps on turning on display on my units, which I don't want. How can I change the library, so the byte responsible for light/display is off?

ToniA, 2022-01-18:
You would somehow need to figure out which bit turns the light on. One way is to build a receiver, and compare the signals from remote against signals from library. See https://github.com/ToniA/Raw-IR-decoder-for-Arduino

I wonder if this could give a hint on which bit controls it: https://github.com/tomikaa87/gree-remote

TrylobitBroccoli, 2022-01-18:
Thank you for your answer. I have a receiver. My previous controller was working with Blynk and I set it up in not a very sophisticated way: I've jus saved raw timings for all the settings...

I've just tested the library that seems to work, but it produces different code and number of bits is changing each time I send the command, while my remote is always sending 64 bits.

Example of light off:
`

Timestamp : 000498.531
Protocol  : GREE
Code      : 0x1C0E405000400080 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 30C, Fan: 1 (Low), Turbo: Off, Econo: Off, IFeel: Off, WiFi: On, XFan: Off, Light: Off, Sleep: Off, Swing(V) Mode: Manual, Swing(V): 0 (Last), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Raw Timing[139]:
   +  9078, -  4426,    +   742, -   464,    +   742, -   462,    +   714, -  1592, 
   +   714, -  1594,    +   742, -  1566,    +   742, -   464,    +   714, -   490, 
   +   714, -   492,    +   714, -   492,    +   738, -  1568,    +   744, -  1562, 
   +   742, -  1566,    +   714, -   492,    +   740, -   464,    +   742, -   464, 
   +   742, -   462,    +   716, -   492,    +   712, -   492,    +   714, -   492, 
   +   712, -   494,    +   714, -   490,    +   740, -   464,    +   714, -  1592, 
   +   714, -   494,    +   714, -   492,    +   742, -   464,    +   770, -   436, 
   +   740, -   464,    +   714, -  1592,    +   714, -   492,    +   740, -  1566, 
   +   742, -   464,    +   740, -   464,    +   714, -  1594,    +   738, -   468, 
   +   742, - 19872,    +   740, -   490,    +   740, -   442,    +   766, -   462, 
   +   742, -   462,    +   744, -   462,    +   716, -   490,    +   714, -   490, 
   +   716, -   490,    +   716, -   490,    +   742, -   464,    +   746, -   458, 
   +   746, -   458,    +   714, -   492,    +   716, -   490,    +   740, -  1564, 
   +   770, -   436,    +   772, -   432,    +   744, -   462,    +   714, -   492, 
   +   740, -   464,    +   742, -   464,    +   744, -   460,    +   772, -   434, 
   +   744, -   460,    +   714, -   490,    +   714, -   490,    +   716, -   490, 
   +   742, -   464,    +   742, -   464,    +   742, -   462,    +   716, -   490, 
   +   714, -  1592,    +   740

uint16_t rawData[139] = {9078, 4426,  742, 464,  742, 462,  714, 1592,  714, 1594,  742, 1566,  742, 464,  714, 490,  714, 492,  714, 492,  738, 1568,  744, 1562,  742, 1566,  714, 492,  740, 464,  742, 464,  742, 462,  716, 492,  712, 492,  714, 492,  712, 494,  714, 490,  740, 464,  714, 1592,  714, 494,  714, 492,  742, 464,  770, 436,  740, 464,  714, 1592,  714, 492,  740, 1566,  742, 464,  740, 464,  714, 1594,  738, 468,  742, 19872,  740, 490,  740, 442,  766, 462,  742, 462,  744, 462,  716, 490,  714, 490,  716, 490,  716, 490,  742, 464,  746, 458,  746, 458,  714, 492,  716, 490,  740, 1564,  770, 436,  772, 432,  744, 462,  714, 492,  740, 464,  742, 464,  744, 460,  772, 434,  744, 460,  714, 490,  714, 490,  716, 490,  742, 464,  742, 464,  742, 462,  716, 490,  714, 1592,  740};  // GREE
uint8_t state[8] = {0x1C, 0x0E, 0x40, 0x50, 0x00, 0x40, 0x00, 0x80};`

Example of light on:
`Timestamp : 000178.816
Protocol  : GREE
Code      : 0x1C0E605000400080 (64 Bits)
Mesg Desc.: Model: 1 (YAW1F), Power: On, Mode: 4 (Heat), Temp: 30C, Fan: 1 (Low), Turbo: Off, Econo: Off, IFeel: Off, WiFi: On, XFan: Off, Light: On, Sleep: Off, Swing(V) Mode: Manual, Swing(V): 0 (Last), Swing(H): 0 (Off), Timer: Off, Display Temp: 0 (Off)
Library   : v_IRREMOTEESP8266_VERSION_MAJOR._IRREMOTEESP8266_VERSION_MINOR._IRREMOTEESP8266_VERSION_PATCH

Raw Timing[139]:
   +  9136, -  4394,    +   718, -   488,    +   744, -   462,    +   744, -  1564, 
   +   770, -  1536,    +   744, -  1564,    +   718, -   488,    +   746, -   460, 
   +   748, -   458,    +   774, -   430,    +   776, -  1532,    +   716, -  1592, 
   +   744, -  1562,    +   774, -   430,    +   778, -   428,    +   746, -   460, 
   +   742, -   464,    +   744, -   438,    +   770, -   458,    +   774, -   430, 
   +   748, -   458,    +   716, -   488,    +   720, -  1588,    +   744, -  1562, 
   +   746, -   458,    +   718, -   488,    +   744, -   462,    +   718, -   488, 
   +   718, -   488,    +   774, -  1532,    +   748, -   456,    +   718, -  1588, 
   +   718, -   490,    +   772, -   432,    +   774, -  1532,    +   748, -   458, 
   +   718, - 19900,    +   798, -   432,    +   776, -   428,    +   720, -   488, 
   +   744, -   462,    +   744, -   460,    +   774, -   432,    +   774, -   432, 
   +   774, -   432,    +   718, -   488,    +   716, -   490,    +   742, -   460, 
   +   744, -   460,    +   800, -   406,    +   746, -   460,    +   718, -  1590, 
   +   744, -   462,    +   744, -   460,    +   774, -   430,    +   746, -   458, 
   +   718, -   488,    +   744, -   464,    +   718, -   488,    +   746, -   460, 
   +   774, -   432,    +   774, -   430,    +   774, -   432,    +   718, -   488, 
   +   742, -   464,    +   718, -   488,    +   770, -   434,    +   776, -   428, 
   +   776, -  1532,    +   718

uint16_t rawData[139] = {9136, 4394,  718, 488,  744, 462,  744, 1564,  770, 1536,  744, 1564,  718, 488,  746, 460,  748, 458,  774, 430,  776, 1532,  716, 1592,  744, 1562,  774, 430,  778, 428,  746, 460,  742, 464,  744, 438,  770, 458,  774, 430,  748, 458,  716, 488,  720, 1588,  744, 1562,  746, 45 [... cut ...]
It looks like this is the bit controlling light:
uint8_t state[8] = {0x1C, 0x0E, **0x40**, 0x50, 0x00, 0x40, 0x00, 0x80}
uint8_t state[8] = {0x1C, 0x0E, **0x60**, 0x50, 0x00, 0x40, 0x00, 0x80}
0x1C0E**40**5000400080
0x1C0E**60**5000400080

I'd love to decode the whole remote and it to the library, but home automation is my only my hobby and my coding skills are limited, so I'm not sure where to start...
Maybe I can get somewhere a template library or example library?

Thank you in advance for your help!

ToniA, 2022-01-18:
You would probably need to change this like of code: https://github.com/ToniA/arduino-heatpumpir/blob/279ae403af3fee9d1478a4a2f6632fa1b653c4bb/GreeHeatpumpIR.cpp#L242

Try setting it to 0x40 or 0x60. The upper 4 bits seem to control features TURBO,LIGHT,HEALTH,X-FAN.

TrylobitBroccoli, 2022-01-18:
OK, I'd know if you can help me with that, but how can I make ESPHome use my modified file?

```
climate:
  - platform: heatpumpir
    protocol: greeya
    horizontal_default: auto
    vertical_default: middle
    max_temperature: 30
    min_temperature: 16
    sensor: temperature
    name: "Office AC"
```

ToniA, 2022-01-18:
That's a build detail of the ESPHome. I'm sorry, but I don't have any experience on it.