ToniA/arduino-heatpumpir #148: Gree YAG-remote suppport

samuliHeikkinen, 2023-05-12:
Hey!

I've been trying to add support for the Gree Amber/Nordic with the YAG remote.
So far I've been able to decode most of the ir signals and they are pretty close to YAC model. 
With some light mods the YAC decoder is able to detect everything else than I-FEEL temperature from the signals.

There is one big difference though. The signal sent by YAG remote is always 24 bytes (213 pulses). There is also some advanced features which send which 32 bytes. I tried implementing the sending the signals but what I can see the YAG needs a send()-function which is pretty much different from the others as the structure of the payload is a lot different.

I've attached a sample of the decoding.

Any tips on where to go from here?

[gree_yag_decoded.txt](https://github.com/ToniA/arduino-heatpumpir/files/11459065/gree_yag_decoded.txt)

thoutenbos, 2023-05-23:
Some time ago I made some changes to support the Gree YAP model.  It is not really ready to merge with back with the mainline yet but it does look pretty much similar to your output for the YAG

Perhaps you can give it a go...
https://github.com/thoutenbos/arduino-heatpumpir/commit/4addf28e57526b4dbcdddc560c3002420e68cd6c

vampywiz17, 2023-11-09:
@samuliHeikkinen @thoutenbos 

I also have a YAG remote and i would like to use ifeel function. Any idea how to solve this thing?

I will able to help data or similar if you need.

thoutenbos, 2023-11-09:
@vampywiz17 you could try my branch to see if it works for you on YAG

arnerek, 2025-02-26:
I see that the YAP support has been merged, but it does not work for me (YAG1FB3 remote). I am able to change temperature, but I doubt the I-feel sensor is working.

Has anyone progressed on this?