ToniA/arduino-heatpumpir #156: Support for SRK25ZC-S

sergioperez, 2023-08-19:
Hello,

I have a Mitsubishi Daiya SRK25ZC-S, which does not seem to work correctly with any of the current implementations (tested via https://esphome.io/components/climate/climate_ir.html#heatpumpir)

At the moment, I have recorded the following input from the remote (address 0x4A75, aeha protocol)

25 cold eco on   0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x6E,0x91
25 cold eco off  0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x7E,0x81
25 cold auto on  0xC3,0x64,0x9B,0xFF,0x00,0xFF,0x00,0x6E,0x91
25 cold auto off 0xC3,0x64,0x9B,0xFF,0x00,0xFF,0x00,0x7E,0x81
25 cold high   0xC3,0x64,0x9B,0xFF,0x00,0xFE,0x01,0x6E,0x91
25 cold med    0xC3,0x64,0x9B,0xFF,0x00,0xF9,0x06,0x6E,0x91
25 cold low    0xC3,0x64,0x9B,0xFF,0x00,0xFD,0x02,0x6E,0x91
25 cold hi-pow 0xC3,0x64,0x9B,0xFF,0x00,0xFC,0x03,0x6E,0x91

24 cold eco on  0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x61,0x9E
24 cold eco off 0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x71,0x8E
24 cold auto    0xC3,0x64,0x9B,0xFF,0x00,0xFF,0x00,0x61,0x9E
24 cold high    0xC3,0x64,0x9B,0xFF,0x00,0xFE,0x01,0x61,0x9E
24 cold med     0xC3,0x64,0x9B,0xFF,0x00,0xF9,0x06,0x61,0x9E
24 cold low     0xC3,0x64,0x9B,0xFF,0x00,0xFD,0x02,0x61,0x9E
24 cold hi-pow  0xC3,0x64,0x9B,0xFF,0x00,0xFC,0x03,0x61,0x9E

25 humidty eco 0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xAE,0x51

25 hot  eco    0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xCE,0x31
25 hot eco off 0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xDE,0x21
25 hot  auto   0xC3,0x64,0x9B,0xFF,0x00,0xFF,0x00,0xCE,0x31
25 hot  high   0xC3,0x64,0x9B,0xFF,0x00,0xFE,0x01,0xCE,0x31
25 hot  med    0xC3,0x64,0x9B,0xFF,0x00,0xF9,0x06,0xCE,0x31
25 hot  low    0xC3,0x64,0x9B,0xFF,0x00,0xFD,0x02,0xCE,0x31
25 hot  hi-pow 0xC3,0x64,0x9B,0xFF,0x00,0xFC,0x03,0xCE,0x31

24 hot eco     0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xC1,0x3E
24 hot auto    0xC3,0x64,0x9B,0xFF,0x00,0xFF,0x00,0xC1,0x3E
24 hot high    0xC3,0x64,0x9B,0xFF,0x00,0xFE,0x01,0xC1,0x3E
24 hot med     0xC3,0x64,0x9B,0xFF,0x00,0xF9,0x06,0xC1,0x3E
24 hot low     0xC3,0x64,0x9B,0xFF,0x00,0xFD,0x02,0xC1,0x3E
24 hot hi-pow  0xC3,0x64,0x9B,0xFF,0x00,0xFC,0x03,0xC1,0x3E

24 cold eco (airflow nomove) 0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x61,0x9E
24 cold eco (airflow move)   0xC3,0x64,0x9B,0xFF,0x00,0xE8,0x17,0x61,0x9E
24 cold eco (airflow down)   0xC3,0x64,0x9B,0xFF,0x00,0xF0,0x0F,0x61,0x9E

The pattern that I see at the moment, it's:
bytes:
[0]: 0xC3
[1]: 0x64
[2]: 0x9B
[3]: 0xFF
[4]: 0x00
[5]: [Fan mode / Power mode]
[6]: opposite of 5
[7]: [Mode(Cool/Warm/Humidity) / Temperature]
[8]: opposite of 7

If I manage to get the time, I will try to implement a PR to control these AC units. Any comments or hints are more than welcome, as I have not too much idea about what I am trying to do :)

nickw444, 2024-07-28:
Hi @sergioperez, did you end up making any progress on implementing IR for this unit? I have a very old `SRK56HAA` which I assume is similar in protocol. The remote part number is RKS502A503D.

sergioperez, 2024-07-28:
Hello @nickw444! Not at all, I just implemented this (far from ideal) workaround in ESPHome to test it, but I did not end up using it:

```
tremote_transmitter:
  pin: D7
  carrier_duty_percent: 50%

button:
  - platform: template
    name: "Frío 26 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x66,0x99]
  - platform: template
    name: "Frío 26 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x76,0x89]
  - platform: template
    name: "Frío 25 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x6E,0x91]
  - platform: template
    name: "Frío 25 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x7E,0x81]
  - platform: template
    name: "Frío 24 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x61,0x9E]
  - platform: template
    name: "Frío 24 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x71,0x8E]
  - platform: template
    name: "Frío 23 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x69,0x96]
  - platform: template
    name: "Frío 23 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0x79,0x86]
  - platform: template
    name: "Calor 26 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xC6,0x39]
  - platform: template
    name: "Calor 26 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xD6,0x29]
  - platform: template
    name: "Calor 25 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xCE,0x31]
  - platform: template
    name: "Calor 25 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xDE,0x21]
  - platform: template
    name: "Calor 24 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xC1,0x3E]
  - platform: template
    name: "Calor 24 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xD1,0x2E]
  - platform: template
    name: "Calor 23 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xC9,0x36]
  - platform: template
    name: "Calor 23 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xD9,0x26]
  - platform: template
    name: "Calor 22 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xC5,0x3A]
  - platform: template
    name: "Calor 22 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xD5,0x2A]
  - platform: template
    name: "Calor 21 - ON"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xCD,0x32]
  - platform: template
    name: "Calor 21 - OFF"
    on_press:
    - remote_transmitter.transmit_aeha:
        address: 0x4A75
        data: [0xC3,0x64,0x9B,0xFF,0x00,0xF8,0x07,0xDD,0x22]
 ```

nickw444, 2024-07-29:
Thanks, I am thinking I will do something similar, the work involved in decoding the protocol seems too high when I could just do what you have done and repeat captured states that I am interested in. Thanks for sharing.