ToniA/arduino-heatpumpir #160: "I-Feel" support for "greeya" protocol

ronnieSVK, 2023-10-10:
Dear developer !

Firstly, thank you very much for your time and effort invested in this project !

I would like to kindly ask you to provide the upper mentioned "I-Feel" support for "greeya" protocol.
Sadly the "greeyac" commands are not registered by our AirCondition, however the factory remote controller is equipped with this function, which is proofed-tested, since we are using our device only with this option.

If i can provide any help in solving this request, just let me know !

Thank you
Best wishes

ronnieSVK, 2024-05-19:
Dear @ToniA ,

is here any chance please to cooperate on this request please ? I have to find a solution because of dimensions of our apartment and presence of a child i have to use our climate device only via i-feel function. The "greeyac" commands are not accepted by coolexpert and "greeya" protocol misses this function.
As i wrote upper, just let me know if i can help/contribute in this !

Thank you Toni !

Greetings !

ToniA, 2024-05-20:
Hi, sorry for not responding earlier. So this 'I-Feel' is a special command (temperature measurement) sent periodically by the remote controller? This would need a protocol decoding from a real remote controller, this would tell us the data that should be sent. See https://github.com/ToniA/Raw-IR-decoder-for-Arduino for instructions.

I can't help much on decoding, as I don't have access to any Gree device.

ronnieSVK, 2024-05-20:
Hey Toni ! 
Thank you very much for your response !
Exactly, the i-feel function is what you described. Sadly i have to use it for proper and safe function. Couple of ESP32S are prepared including remote receiver/transmitter, so i can do whatever is needed/whatever im able to do. I think couple of users facing this issue will be glad for this function. The only question is, how should i perform the signal capture ? I mean not technically (ill check your provided link for that), but from the perspective that the remote measures the ambient temp, so if we would like to capture the values for different temperatures-temperature range, i have to simulate those ambient temperatures and also save them, to know which Ir code belongs to which ambient temperature value. It is like this please ? Do you know maybe, how this capture/decoding was done for the "greeyac" protocol please ?

Thank you !
Greetings
Ronnie

Nateonas, 2024-05-20:
I don't have an I-feel AC, but for my setup I attached a DHT22-sensor to the ESP. I installed ESPeasy, so both heatpumpIR and the DHT-sensor are supported. The room-temperature is sent to my Domoticz controller. With this setup it is possible to control the AC based on the temperature in the room from Domoticz (or Home Assistant of course).
You probably will need to find the I-feel IR-codes sent by the remote, but perhaps you can use the temperature from a DHT-sensor to include the temperature.

ronnieSVK, 2024-05-20:
Hi @Nateonas !
Thank you very much for your contribution !

This is exactly what i would like to achieve. Im running HomeAssistant, so my goal is to build Ir remote on ESP32 with ESPHome and link my room temperature sensor to it. But for this, i have to activate/use the "I-Feel" option, which will measure the ambient temperature from AC remote controllers sensors (in this case from sensor linked to ESP32) otherwise the AC unit will use its internal ambient temp sensor. So the "I-Feel" is the way to tell AC to use the different sensor, and also to send/receive/behave regarding to measured values. Even if the device is equipped with WiFi module, you can activate "I-Feel" only via remote (i think it is self explaining...) and the remote should be placed in distance/position from where it can transmit every 2minutes the measured values to the AC.

Nateonas, 2024-05-21:
Would it be possible at all to implement functions for a specific model or brand? It might be easier to develop a separate ESP just to transmit the temperature readings to the AC. What setup did you have in mind?

Machaeon, 2024-06-03:
Hi @ronnieSVK ! I am using ESPHome with **heatpumpir** platform to send temperature data to my AC. I have specified the **greeyac** protocol, which actually does not match the protocol of my AC. It doesn't allow me to control the AC, but it transmits temperature data properly. But there is one problem - when I control the AC through the integration in my Home Assistant, it disables the iFeel function and I have to reactivate it using the remote.

ronnieSVK, 2024-06-03:
Hi @Machaeon !

Thank you very much for you reply !
Yes, the behaviour that "i-Feel" turns off everytime the AC is controlled differently as with "i-Feel" enabled Ir Remote Controller (or ESP) is known. I think the reason is, that Remote Controller is transmitting this information (so about i-Feel On/Off status) each time you control your device.
Im glad, that at least the "i-Feel" function works. Maybe some workaround in HA can be made, where after each "command sent" to AC will be "i-Feel ON" command transmitted. Maybe i can check it. 
Could you be please so nice and share your full ESPhome code ? I would like to test out exactly the same config as yours and find some way to make it work (hopefully only till i-Feel will be implemented to other protocols). If you wish, we can stay in touch if i achieve my goal in HA about sending the mentioned command.

Thank you !

Machaeon, 2024-06-03:
@ronnieSVK, Here is my yaml:
```
esphome:
  name: gree_ir
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Gree IR AP"
    password: !secret ap_password 

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key 

web_server:
  port: 80

ota:
  password: !secret ota_password 

logger:

remote_transmitter:
  id: ir_transmit
  pin: D3
  carrier_duty_percent: 50%  

sensor:
  - platform: homeassistant
    id: ha_temperature
    entity_id: sensor.current_temperature

  - platform: template
    id: current_temperature
    lambda: |-
      return (id(ha_temperature).state - 0.5);
    update_interval: 120s  

climate:
  - platform: heatpumpir 
    protocol: greeyac
    name: "AC"
    horizontal_default: middle
    vertical_default: auto
    max_temperature: 30
    min_temperature: 16
    sensor: current_temperature
```
Here's something else that might be important: my AC accepts commands if I specify the **greeyan** protocol but it disables its WiFi adapter when it receives an IR command and I had to reconnect the AC to the network using the mobile app.

ronnieSVK, 2024-06-03:
Thank you @Machaeon once again !

However (if i good remeber) my code is quite the same, ill check it out in short and provide feedback. At least, we are now "two" with approx. same equipment and issue. If @ToniA provides reply for my questions about capturing Ir codes, maybe we can both cooperate to move things bit quicker.

ronnieSVK, 2024-06-03:
@Machaeon , feedback from me:
As i expected, my code was the same. 
What i have done : 

- Turned ON AC via factory Remote Controller with i-Feel option ON and put it "out of sight", to avoid communication/temperature transmitting to the unit.

- Turned ON the AC via ESPhome (this was of course just software stuff, since as we know, the AC doesnt capture the "greeya" command) and pointed the Ir Transmitter to face to AC.

The result is, that ESPhome shows the temperature from HA tenperature sensor, but this value isnt transmitted to AC... Now im confused where is the issue, since my Ir Transmitter has LED diode on the board, which light up if command is transmitted. The LED behaves like this if i push command from ESPHome/HA to operate the AC (what again, doesnt work woth greeya for us). BUT, since the temperatures should be transmitted to AC unit every 2minutes, im expecting to "see" this via LED on Ir Transmitter. But this doesnt happen in my case. So im unsure now, if my unit didnt accept anything or the transmitter doesnt send the temp datas. May i please ask you to confirm/deny if your Transmitter has confirmation diode, if you see some activity in 2minute cycles ?

Thank you !

Machaeon, 2024-06-03:
> * Turned ON the AC via ESPhome (this was of course just software stuff, since as we know, the AC doesnt capture the "**greeya**" command) and pointed the Ir Transmitter to face to AC.

If you used the **greeya** protocol - this may be the source of the problem. In my config I am using **greeyac** and I see a blinking LED every two minutes

ronnieSVK, 2024-06-03:
@Machaeon sorry mate, this was just a typo in description. Im using the "greeyac" protocol of course

Machaeon, 2024-06-03:
That's very strange. Can you show your config, maybe we can find a non-obvious problem?

ronnieSVK, 2024-06-03:
@Machaeon , yeah, im also confused :/

```
esphome:
  name: esphome-web-c66308
  friendly_name: Kitchen Air Condition Ir

esp32:
  board: esp32dev
  framework:
    type: arduino
    
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none

  manual_ip:
    static_ip: 192.168.1.65
    gateway: 192.168.1.1
    subnet: 255.255.255.0

remote_transmitter:
  pin: GPIO25
  carrier_duty_percent: 50%

climate:
  - platform: heatpumpir
    protocol: greeyac
    horizontal_default: auto
    vertical_default: up
    max_temperature: 30
    min_temperature: 16
    sensor : kitchen_air_condition_temperature_sensor
    name: "Kitchen Air Condition Ir"

sensor:
  - platform: homeassistant
    name: "Kitchen Air Condition Temperature"
    entity_id: sensor.living_room_th_sensor_temperature
    id: kitchen_air_condition_temperature_sensor

  - platform: template
    id: kitchen_air_condition_current_temeprature
    lambda: |-
      return (id(kitchen_air_condition_temperature_sensor).state);
    update_interval: 120s
```

Machaeon, 2024-06-03:
@ronnieSVK 
`sensor : kitchen_air_condition_temperature_sensor`
Try to change to
`sensor : kitchen_air_condition_current_temeprature`

ronnieSVK, 2024-06-03:
@Machaeon ,

Thank you for suggestion. The issue was on my side... i was too quick to put expansion board under my ESP for easy wiring, and one of the connections was not stable. So now the issue with "2min interval" is solved, the LED lights up in this interval according to the logs. However my AC didnt capture those readings and temperatures. I can make tomorrow longer test, but i think those 10minutes today were enough. But i repeat it.

In the meantime i have idea about reactivating "i-feel" option. I have Ir Receiver somewhere... If i would be able to capture the "I-Feel ON" command from the remote, then it would be possible to make automation in HA, where after each WiFi operated command would be with some delay the mentioned command sent.

Machaeon, 2024-06-03:
@ronnieSVK, after fixing the connection problems, it is very strange that the config works without the fixes I mentioned. In your config you created template sensor **kitchen_air_condition_current_temeprature** but did not use it because **kitchen_air_condition_temperature_sensor** was specified in the integration climate

ronnieSVK, 2024-06-03:
@Machaeon Everything looks fine, even the logs. Sadly what is missing is the temperature update in AC unit. It looks like my device didnt captures even the temps from "i-Feel" (my AC work normally with greeya, your device with greeyan, this makes for sure difference). Anyways, i try tomorrow to capture the function switch from Remote, at least it helps you to operate (somehow) the device. It is really very unlucky, that only one from the protocols is till now equipped with this feature

guntiss, 2024-06-04:
Hello!

Looks like I have bought similar AC. My AC remote's model on the back says YAC1FB9, but none of the provided protocols (generic, yan, yaa, yac) seems to work.

I was able to decode signal from remote using this code -
https://github.com/ToniA/Raw-IR-decoder-for-Arduino/commit/6239a28decd0a4b89ac4987695caf677fd1e2f87

So It seems that it might actually be a new variant - YAP?

This is the output I got from decoder - I-FEEL value seems to be updating accordingly and other values are decoded correctly.
```
################# Start

Number of symbols: 232
Symbols:
Hh10010001000100000000001000001010010W00000000001000110000000000001110WHh10010001000100000000001000001110010W00000010000000000000000000001111WHh00000000000000000000000000000101010W00000000000000000000000000000101hHh0101100010100101
Bytes:
00:  1001|0001 | 89 | 10001001 
01:  0001|0000 | 08 | 00001000 
02:  0000|0010 | 40 | 01000000 
03:  0000|1010 | 50 | 01010000 
04:  01004:  0000|0000 | 00 | 00000000 
05:  0010|0011 | C4 | 11000100 
06:  0000|0000 | 00 | 00000000 
07:  0000|1110 | 70 | 01110000 
08:  1001|0001 | 89 | 10001001 
09:  0001|0000 | 08 | 00001000 
10:  0000|0010 | 40 | 01000000 
11:  0000|1110 | 70 | 01110000 
12:  01012:  0000|0010 | 40 | 01000000 
13:  0000|0000 | 00 | 00000000 
14:  0000|0000 | 00 | 00000000 
15:  0000|1111 | F0 | 11110000 
16:  0000|0000 | 00 | 00000000 
17:  0000|0000 | 00 | 00000000 
18:  0000|0000 | 00 | 00000000 
19:  0000|0101 | A0 | 10100000 
20:  01020:  0000|0000 | 00 | 00000000 
21:  0000|0000 | 00 | 00000000 
22:  0000|0000 | 00 | 00000000 
23:  0000|0101 | A0 | 10100000 
24:  0101|1000 | 1A | 00011010 
25:  1010|0101 | A5 | 10100101 
89,08,40,50,00,C4,00,70,89,08,40,70,40,00,00,F0,00,00,00,A0,00,00,00,A0,1A,A5
Timings (in us): 
PAUSE SPACE:  28039
HEADER MARK:  8211
HEADER SPACE: 3897
BIT MARK:     584
ZERO SPACE:   567
ONE SPACE:    1692
Decoding known protocols...
Looks like a Gree YAC or YAP protocol
Checksum matches
POWER ON
ECO OFF
TURBO OFF
MODE COOL
Temperature: 24
FAN: AUTO
SLEEP: ON 1
SWING: OFF
VERT. VANE STOP
HORIZ. VANE STOP
DISPLAY: SET TEMP
I-FEEL: ON
WiFi: ON
StHt: OFF
I-FEEL TEMPERATURE: 26
################# End 
```

I See that there is a recent commit in this repo regarding YAP protocol - https://github.com/ToniA/arduino-heatpumpir/commit/fe4789d9cb7091778f79b961b443674bf409ad06
Perhaps this Is the one that will work for this AC.

It seems that ESPHome has not yet implemented this code, as it's not allowing to define "greeyap" protocol.

ronnieSVK, 2024-06-04:
Hi @guntiss !

According to the remote controller model, it should be "yac", but yeah... who knows, as log wrote (looks like YAC or YAP protocol). My remote is marked "YAN" , but only command from protocol "greeya" are accepted. I try to decode the signal today, i have to check the documentations

ronnieSVK, 2024-06-04:
Hi @Machaeon !
Mate, i can confirm that the temperatures are transferred for me also to the AC unit ! The issue yesterday was the "Ir Transmitter -> AC unit" distance, and the time lag between value update from "Room Temp Sensor -> MQTT -> ESP 2min interval -> AC unit". I dont know if the 2min can be shortened, ill test it out. So now im on "stage" as you are and till this issue will be solved in all protocols, the "i-Feel" function has to be turned ON after each WiFi command. As i mentioned previously, my goal would be to capture the "i-Feel ON" command from remote and send it automatically after every WiFi command. Im just unsure if this will work, since our devices are operating under different protocols

Machaeon, 2024-06-04:
Hi @ronnieSVK! I don't think it is possible to send I-Feel enable as a separate command. It is part of the data package.
I checked the version of my remote. It's YAA1FB.
@ToniA, apparently, the I-Feel temperature data format of the YAC protocol is the same as the one in the YAA protocol. Could you add I-Feel to the YAA protocol?

Machaeon, 2024-06-04:
@ronnieSVK, although control via Home Assistant disables the I-Feel function, I found a compromise solution for controlling the AC temperature. I created a virtual sensor and input_number. The virtual sensor takes the room temperature value plus the input_number value (which can be positive or negative) and transmits them in I-Feel data format to the AC. So I don't change the target temperature via Home Assistant, I change the temperature that the AC sees via I-Feel and thus can influence the AC's output.

ronnieSVK, 2024-06-05:
@Machaeon , lets say it is a "solution" but it seems to me really "dirty". I think, at this point, capturing the "i-Feel" command separately could help us a lot. Sadly it is as you wrote, it is always part of command package from remote, since the button combination doesnt work for me with remote turned off. I have the YAN1F6 remote, where the "UP+Mode" button combination triggers the function. YAN1F1 version has dedicated "i-Feel" button, i dont know how this remote behaves

Machaeon, 2024-06-05:
@ronnieSVK, so far all we have is this, but I keep digging )
My method has a lot of flaws, but it has one good feature - it's quiet. This way of controlling the temperature of the AC doesn't make it beep )

ronnieSVK, 2024-06-05:
@Machaeon Yeah, i thought that :) However, i turn on AC unit once, and touch it once or twice after that. My issue only is because of size of our room, where i have to use deflector ahead of AC. This makes little turbulences and causes temperature misreadings of internal temp sensor. Anyways, it is better to measure ambient temp from my external room sensor, since it is located exactly "where it should be", but otherwise i had no issue with built-in sensor or beeps :) I dont want to disassemble the unit, otherwise i was thinking of replacing the internal temp sensor with ESP32 and send the temp readings from room sensor as we (would like to) do now...

Machaeon, 2024-06-05:
@ronnieSVK, I was forced to deal with this problem by the exact same situation with the deflector and I also thought about replacing the internal A/C sensor, until I found out that ESPHome can send data in iFeel format. 
Now I can specify which external temperature sensor readings to send to the A/C and correct them by adding a delta, which I can specify manually if necessary.
[image]

Machaeon, 2024-06-06:
@ronnieSVK, well, I did it. Added iFeel and WiFi enable for the greyaa protocol to the heatpumpir platform for ESPHome. 
I will try to add pull request, but for now I compiled with file swapping and it works. I have created an automation that duplicates the AC settings via IR when they are changed in Home Assistant. It adds two extra beeps, but that's something I'm willing to put up with )

ronnieSVK, 2024-06-06:
hi @Machaeon ! Very nice result ! So it looks like, the solution for some protocols should be not too hard as expected. I dont think ToniA will do anything here, so the only could be using it as external component and make some tests. The duplicated commands in HA was exactly what i was thinking about at the beginning :) Great results mate !

ronnieSVK, 2024-06-29:
Hi @Machaeon ,
i hope youre doing well !

as i predicted, nothing changed in this topic sadly :( I would like to touch two topics, if you have time and possiblity, your answer would be really appreciated.

- i made a decent enclosure for Ir sensor and ESP, and im using the AC with "external sensor" since 3 weeks. My experience is, that the whole concept with "greeya/greeyac/iFeel" works (of course i cannot control the AC via Ir), but i have sometimes difference in temperatures.
I mean, right now : 
The AC is set to COOL 26C
My room Temp Sensor reports CURRENTLY 25.0C -> 
this value is successfully transferred to ESP = the ESP Climate reports also 25C and this should be forwarded via iFeel to AC
But the AC (via iFeel from ESP) shows CURRENT 26C.

It works, but i have to care about this, since (as you know) 1C difference with AC is really noticeable. This difference is not constant, i mean 27 from sensor is 27 from iFeel too. But this 25/26C range is reported not correctly. Do you have also such issues ?

- the second thing is, that you mentioned, that you had successfully merged the two protocols, and you are now able to control your AC via Ir. Do you have please the opportunity to do the same with "iFeel from greeyac and control from greeya" and share it as a external component via link please ?

Thank you very much !
Greetings !

Machaeon, 2024-07-01:
Hi @ronnieSVK! I also encountered a temperature difference. This problem was solved by correcting the temperature in the sensor template by subtracting 0.5 degrees:
```
  - platform: template
    id: current_temperature
    lambda: |-
      return (id(ha_temperature).state - 0.5);
    update_interval: 120s  
```
I don't know how to make an external component, but I can provide a link to an archive with the library code that can be used to replace the one downloaded when compiling the yaml:
https://drive.google.com/file/d/1ebamESalSEToWIyC4SiiKZ3YVpI7xtZh/view?usp=sharing

ronnieSVK, 2024-07-03:
Thank you @Machaeon ! Ill check it out and try to make external component available

ronnieSVK, 2024-07-18:
Hi @Machaeon ,
since you are the only one (i know) using "same workaround for same reason", i would like to ask you, if you encountered some issues after ESPHome 2024.7 upgrade ?

There was some changes made in "heatpumpir" itself and included to new ESPHome release
https://github.com/esphome/esphome/pull/6996

After update is AC capturing always "24C" temperature from "iFeel", no matter what the real (room sensor) temperature is. Also i found out, that the unit is now responding to some Ir commands, but not all (if you remember, till now it was only iFeel function/temp reporting that worked with greeya for me). For instance i can change temperature up (but not down) and i can turn off the device. It is strange, but the worse part is the corrupted iFeel function and additionally i have a feeling, that it messes also the WiFi connection. I think WiFi reset command is sent via Ir to the device.

Did you encountered some of those issues also ?

Thank you !

Machaeon, 2024-07-18:
Hi @ronnieSVK! Haven't tried the update yet, but the latest version of heatpumpir had the timings changed, causing it to lose performance with my AC. In my modification of the library, I have reverted the timings back to those specified in the earlier version.

`After update is AC capturing always "24C" temperature from "iFeel"`
It looks like in your case the AC is not accepting iFeel data and is showing the temperature from the inbuilt sensor. Talking about WiFi disconnection - this was also in earlier versions and I fixed it in my modification.

ToniA, 2024-07-18:
I rely entirely on PR's on the Gree protocol, I have no way to test them myself. If there's a bad change, could you please submit a PR to fix it?

ronnieSVK, 2024-07-21:
Hi @Machaeon !
Thank you for your reply. At least i know im not alone with those issues. The temperature readings wasnt from internal sensor (because of air deflector it should show much lower value), but the readings was somehow faulty, as the whole controller. I try to revert to you library, which you posted here.

Thanks once again !

namezys, 2025-06-14:
Just in case, there is my config for remote YAC1FB9.
Setting sensor in climate section does not work because it doesn't send anything.

Also, I spend to much time to realize that cheap IR diode is too week and I have to choose correct direction.

```
# iFeel for Gree indoor units (based on remote YAC1FB9)

substitutions:
  ir_pin: uknown
  temp_entity_id: some.sensor

script:
  - id: send_ifeel
    mode: restart
    then:
      - remote_transmitter.transmit_raw:
          repeat:
            times: 1
          carrier_frequency: 38000
          code: !lambda |-
            // Timing and codes was reconstracted by reading line to IR diod directly.
            // Actually: freq 38kHz, duty 40%, prefix is shorter than for standard commands

            // get temperature from external sensor
            const auto input_temp = round(id(ifeel_temperature).state);
            ESP_LOGI("iFeel", "iFeel temperature: %u", input_temp);
            // valid temperature range [8, 59], out of this range unit uses internal sensor
            const uint8_t ifeel_temp = (input_temp < 8 || input_temp > 59) ? 0 : input_temp;
            ESP_LOGI("iFeel", "Send iFeel temperature (zero force to use internal sensor): %u", ifeel_temp);
            // this value was valid for remote YAC1F9B
            const auto I = 650;
            const auto S = 550;
            const auto L = 1700;
            const auto v = [=](auto bit) { return -((ifeel_temp & (1 << bit)) ? L : S); };  
            return {
              // header
              6000, -3000,
              // temperature, LBF
              I, v(0), I, v(1), I, v(2), I, v(3), I, v(4), I, v(5), I, v(6), I, v(7),
              // constant marker
              I,   -L, I,   -S, I,   -L, I,   -S, I,   -S, I,   -L, I,   -S, I, -L, 
              // end
              I
            };

sensor:
  - platform: homeassistant
    id: ifeel_temperature
    entity_id: ${temp_entity_id}
    on_value:
      then:
        - script.execute:
            id: send_ifeel

remote_transmitter:
  pin: ${ir_pin}
  carrier_duty_percent: 40%

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
      - seconds: 0
        minutes: /1
        then:
          - script.execute:
              id: send_ifeel

```

Machaeon, 2025-06-22:
@namezys
Nice workaround! Sadly, there is still the problem of iFeel deactivating when sending WiFi commands to the air conditioner.

ronnieSVK, 2025-06-22:
Hi @namezys , hi guys !

Im unsure if i understand your post correctly, but you mentioned, that "sensor" doesnt send anything. This is topic exactly about the "i-feel" part/function. Can you explain bit more what your post is about ? Im sorry for that, but i dont understand.

Thanks !

Machaeon, 2025-06-22:
Hi @ronnieSVK !
This config sends temperature data from the Home Assistant sensor to the AC via the iFeel protocol.

ronnieSVK, 2025-06-22:
Hey @Machaeon . Thanks for your reply. Yes, im using this device since we had the ci versations in the past. I was just curious about namezys post and words...