ToniA/arduino-heatpumpir #165: Correct swing commands for Hitachi models

octal-ip, 2024-01-13:
There are a few minor mistakes in HitachiHeatpumpIR.cpp that prevent the vertical and horizontal swing functions from operating. The corrected code block is below with changes in bold:

```
  switch (**swingVcmd**)
  {
    case **VDIR_AUTO**:
      swingV = HITACHI_AIRCON1_VDIR_AUTO;
      break;
    case **VDIR_SWING**:
      swingV = HITACHI_AIRCON1_VDIR_SWING;
      break;
  }

  switch (**swingHcmd**)
  {
    case HDIR_AUTO:
      swingH = HITACHI_AIRCON1_HDIR_AUTO;
      break;
    case HDIR_SWING:
      swingH = HITACHI_AIRCON1_HDIR_SWING;
      break;
  }
```

ToniA, 2024-01-13:
Hi, would you create a pull request? I will then merge and release it. Thanks!

octal-ip, 2024-04-03:
Pull request created.