ToniA/arduino-heatpumpir #177: Inverted PWM Signal?

tswonke, 2024-07-27:
Sorry for a maybe dumb question, but I don't understand the generated signal.
When I start the ESP32 the signal on my chosen GPIO is LOW. With the first sent signal it flips to HIGH (s. attached picture) and stays there for ever (until the next signal):

[image]
[image]

Is this on purpose?
From my understanding this would mean that an attached IR-LED would constantly be glowing. Or do I have to invert the signal with a transistor?

My code is very simplified for testing:

```cpp
#include <Arduino.h>
#include <FujitsuHeatpumpIR.h>

IRSenderESP32 irSender(33, 0);
FujitsuHeatpumpIR *heatpumpIR;

void setup()
{
  Serial.begin(115200);
  delay(500);
  heatpumpIR = new FujitsuHeatpumpIR();
}

void loop()
{
  heatpumpIR->send(irSender, POWER_ON, MODE_FAN, FAN_3, 18, VDIR_AUTO, HDIR_AUTO);
  delay(2000);
}

tswonke, 2024-07-27:
After changing `IRSenderESP32.cpp` by moving `pinMode(_pin, OUTPUT);` from Line 46 to 51

[image]

the signal seems to be right:

[image]

... at least in my eyes. Maybe I'm wrong?
Sender is working, Fujitsu AC reacts.

ToniA, 2024-08-10:
You're probably right on this one... Would you create a pull request with the fix (and please verify that it's also working correctly on non-inverted case)?

tswonke, 2024-08-20:
Ok, I will probably come back to this this week or next week