ToniA/arduino-heatpumpir #185: Panasonic A75C3208 Remote

barchasse38, 2024-11-04:
Hello,

Thank you for the project and the work accomplished.

I'm trying to control a Panasonic Inverter, the remote part provided in the manual is CWA75C3208.

I tried all the Panasonic protocol provided in the project and none of them was entirely satisfactory.

DKE / LKE / JKE  / NKE works for the mode selection (OFF / HEATCOOL / HEAT / COOL / DRY) but the temperature selection doe not ;
-Selecting the minimum temp send the max temp to the inverter
-Selecting the maximum temp send the min temp to the inverter
-Selecting other temp does nothing.

Could the problem be with the hardware?
Or it just means that the remote use another protocol ?

If so I want to try to decode this new protocol and share it on the project.

Thank you for your help

germin8, 2025-02-22:
This is a bitwise reversal of the input value doubled.

ToniA, 2025-02-24:
Nice! Was it so that on newer Panasonic models you can set the temperature on half degree increments?